MichaIng/DietPi #2498: Post update v6.14 to v6.20 yields error /DietPi/dietpi/login line 168 error token

kevkha, 2019-02-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version` 6.14
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version` jessie
- Kernel version | `uname -a`  Linux DietPi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3) RPi 3 Model B (armv7l)
- Power supply used | (EG: 5V 1A RAVpower) 5V 2.5A
- SDcard used | (EG: SanDisk ultra) SanDisk

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) Nexcloud
- Was the software title installed freshly or updated/migrated? updated
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

The update completed but upon reboot it still at the 6.14 version
 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 dietpi-update    = Run now to update DietPi (from v6.14 to v6.20).
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@DietPi:~# dietpi-update
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

...

[  OK  ] DietPi-Patchfile | Desired setting in /DietPi/dietpi/.installed was already set: aSOFTWARE_INSTALL_STATE[116]=0
[  OK  ] DietPi-Patchfile | APT installation for: procps, please wait...

[  OK  ] DietPi-Patchfile | G_AGI: procps
[  OK  ] DietPi-Update | Patching to v6.20.6 completed

[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com
[ SUB2 ] DietPi-Update > Completed
[ INFO ] DietPi-Update | Current version : v6.20.6
[ INFO ] DietPi-Update | Latest version  : v6.20.6
Connection closing...Socket close.

Connection closed by foreign host.

Disconnected from remote host(Nextcloud) at 15:55:04.

Type `help' to learn how to use Xshell prompt.
[C:\~]$ 

Connecting to 192.168.1.75:22...
Connection established.
To escape to local shell, press 'Ctrl+Alt+]'.

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
/DietPi/dietpi/login: line 42: /DietPi/dietpi/.dietpi-autostart_index: No such file or directory

 ───────────────────────────────────────
 DietPi     | Update available 
 ───────────────────────────────────────
 v6.14      | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
 eth0       | 192.168.1.75
 ───────────────────────────────────────
 DietPi Core Team   : Daniel Knight (founder), MichaIng, K-Plan
 Web                : https://DietPi.com | https://twitter.com/dietpi_
 Donate             : https://goo.gl/pzISt9
 DietPi web hosting : Powered by https://MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 dietpi-update    = Run now to update DietPi (from v6.14 to v6.20).
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

/DietPi/dietpi/login: line 168: ((: > 0 : syntax error: operand expected (error token is "> 0 ")

- ...

#### Extra details

- ...

MichaIng, 2019-02-05:
@kevkha 
Thanks for your report.

The reboot is not done automatically, right? (It should not, if this is no fresh install)
If asked to restart, cancel/exit instead.

Please try:
```
dietpi-update 1
systemctl restart dietpi-ramdisk
reboot
```

MichaIng, 2019-02-06:
PR for fix is up: https://github.com/Fourdee/DietPi/pull/2509
Release will be very soon.

Fourdee, 2019-02-07:
Completed, v6.21 released.

closed 2019-02-07 by Fourdee

kevkha, 2019-02-11:
Sorry for a delayed response, I was out for lunar new year. Thanks for the fix! All sent well.