MichaIng/DietPi #2499: Dietpi fails to update due to 404 error

Rainee4563, 2019-02-05:
#### Details:
- Date           | Mon  4 Feb 20:57:16 PST 2019
- Bug report     | N/A
- DietPi version | v6.19.7 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. Attempt to update to v6.20.6
2. ...

#### Expected behaviour:

- Update should proceed smoothly with no errors

#### Actual behaviour:

- Gives a 404 error for the raspbian stretch Release... 

#### Extra details:

- not sure if its a issue on my end but I did check the link in the browser and it seems to obviously not exist anymore

#### Additional logs:
```
Log file contents:
Hit:1 http://apt.sonarr.tv develop InRelease
Hit:2 https://dtcooper.github.io/raspotify jessie InRelease
Ign:3 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:4 https://download.docker.com/linux/raspbian stretch InRelease
Err:5 http://raspbian.raspberrypi.org/raspbian stretch Release
  404  Not Found
Hit:6 https://download.mono-project.com/repo/debian raspbianstretch InRelease
Hit:7 http://dev2day.de/pms stretch InRelease
Hit:8 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
E: The repository 'http://raspbian.raspberrypi.org/raspbian stretch Release' does no longer have a Release file.
```

MichaIng, 2019-02-05:
@Rainee4563
Thanks for your report.

This should be only a temporary issue. The release file exists now: http://archive.raspberrypi.org/debian/dists/stretch/Release

Please retry: `G_AGUP`

FWeissenb, 2019-02-05:
Dont't know if its the same issue. But i get

 DietPi-Software: Connection test: http://raspbian.raspberrypi.org/raspbian
                                                                         │
- Exit code: 8
                                                                         │
- DietPi version: v6.19.7 (Fourdee/master) | HW_MODEL:3 | HW_ARCH:2 |
DISTRO:4
                                                                         │
- Image creator: n/a
                                                                         │
- Pre-image: n/a
                                                                         │
                                                                         │
Log file contents:
                                                                         │
Spider mode enabled. Check if remote file exists.
                                                                         │
--2019-02-05 08:32:22--  http://raspbian.raspberrypi.org/raspbian
                                                                         │
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)...
5.153.225.207,
                                                                         │
2001:41c9:1:3ce::1:11
                                                                         │
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org
)|5.153.225.207|:80...
                                                                         │
connected.
                                                                         │
HTTP request sent, awaiting response... 404 Not Found
                                                                         │
Remote file does not exist -- broken link!!!

Edit network settings and choose another apt mirror works. But if i come
back to the connection test it fails with the old url.

MichaIng, 2019-02-05:
Changing the APT mirror only affects the Debian repo, the Raspbian repo is only available via this one mirror.

Hmm strange it does not work, at least yesterday night it did on my RPi.

Will retest later.

FWeissenb, 2019-02-05:
Now it works. Sorry. Seems to be a  temporariness failure. :/

frank.weissenborn@gmail.com>:

fyrfli, 2019-02-05:
So I noticed a difference between what the DietPi link and the link you specified:

i.e. DietPi - http://raspbian.raspberrypi.org/raspbian
      Your link - http://archive.raspberrypi.org/debian/dists/stretch/Release

And I know my knowledge of how repositories work might be limited, but can you say whether that difference matters? I am still getting the error:

"W: The repository 'http://raspbian.raspberrypi.org/raspbian stretch Release' does not                                   have a Release file."

vertti, 2019-02-05:
Doesn't seem to work. I get 

```
Reading package lists...
W: The repository 'http://raspbian.raspberrypi.org/raspbian stretch Release' does not have a Release file.
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/stretch/main/binary-armhf/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

everytime I run `G_AGUP`

MichaIng, 2019-02-05:
Ah yeah did not think about the other URL. Both (should) however point to exactly the same repo. In DietPi we moved to the new URL since during one APT update of `raspberrypi-sys-mods` they patched it on the sources file in a way that broke the source, e.g. when https used or a mirror director. raspbian.raspberrypi.org does not support https, while archive.raspberrypi.org does.

However I see the Release file on both repos when opening on browser :thinking:.

Perhaps you guys can try to switch the URL from between the mentioned ones inside /etc/apt/sources.list.d/raspi.list

fyrfli, 2019-02-05:
Original repo works now. I had just switched to one of the others in the list inside dietpi-config and all seems well. 

Thanks!

MichaIng, 2019-02-05:
Hmm just tested here http://raspbian.raspberrypi.org/raspbian in sources and it worked well, although I run Buster here, not Stretch.

Fourdee, 2019-02-06:
Will mark as closed, unable to replicate currently, appears to be a temp remote server issue.

closed 2019-02-06 by Fourdee