MichaIng/DietPi #2501: Impossible to stop the installer! ( + Error 404 on the connection test)

tim54000, 2019-02-05:
#### Details:
- Date           | Tue  5 Feb 17:07:33 CET 2019
- Bug report     | N/A
- DietPi version | v6.20.6 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | Connection test: http://deb.debian.org/debian
- Exit code      | 8
- Software title | DietPi-Software

#### Steps to reproduce:
1. Update from 6.14 to 6.20
2. Finish the installation until restart
3. The installer now wants to change the passwords (root and dietpi)
4. Cancel the password change for both accounts

#### Expected behaviour:
The installer is supposed to finish after the Connection test

#### Actual behaviour:
The connection test fails (404 error). Try to pass it with "Cancel" and the installer will resume at point 3.

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-02-05 17:07:29--  http://deb.debian.org/debian
Resolving deb.debian.org (deb.debian.org)... 130.89.148.14, 5.153.231.4, 149.20.4.15, ...
Connecting to deb.debian.org (deb.debian.org)|130.89.148.14|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://cdn-fastly.deb.debian.org/debian [following]
Spider mode enabled. Check if remote file exists.
--2019-02-05 17:07:29--  http://cdn-fastly.deb.debian.org/debian
Resolving cdn-fastly.deb.debian.org (cdn-fastly.deb.debian.org)... 151.101.120.204, 2a04:4e42:1d::204
Connecting to cdn-fastly.deb.debian.org (cdn-fastly.deb.debian.org)|151.101.120.204|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

#### Extra logs:

*After a failed connection test and the user wants to pass "Exit", it loops*
```
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.
[  OK  ] DietPi-Software | Root access verified.
[  OK  ] DietPi-Software | RootFS R/W access verified.

[  OK  ] DietPi-Software | Initialized database
[  OK  ] DietPi-Software | Reading database
[ SUB1 ] DietPi-Set_software > passwords ()
[  OK  ] passwords  | Completed
[FAILED] DietPi-Software | Connection test: http://deb.debian.org/debian

```

Fourdee, 2019-02-06:
Merged into this ticket (same issue):
https://github.com/Fourdee/DietPi/issues/2501

closed 2019-02-06 by Fourdee

tim54000, 2019-02-12:
After updating in 6.21.1, the assistant continues to loop endlessly.

Fourdee, 2019-02-12:
@tim54000 

Hi Tim,

Please try the following:
- Login to the system
- Press CTRL+C straight away
- Use the following command ```echo 2 > /DietPi/dietpi/.install_stage```
- Restart dietpi-ramdisk ```systemctl restart dietpi-ramdisk```
- Reboot system, device will now be in completed setup stage.

tim54000, 2019-02-12:
Thank you for your help! Finally I decided to add to the `/etc/apt/sources.list` the Raspbian repository (on the first line) that now works during the connection test.

MichaIng, 2019-02-12:
@tim54000 
The Raspbian repo was not already inside /etc/apt/sources.list??

tim54000, 2019-02-12:
@MichaIng 
Raspbian repositories are not directly in `/etc/apt/sources.list` (and certainly not at the first line) but in `/etc/apt/sources.list.d/raspi.list`.
So it is the Debian repository that is chosen during the connection test and is apparently down (Error 404).

MichaIng, 2019-02-12:
@tim54000 
Check my comments above. `/etc/apt/sources.list.d/raspi.list` does not contain the Raspbian repo, but the Raspberry Pi specific repo with kernel/firmware packages. This stays untouched during first run update.

tim54000, 2019-02-12:
Right now, my `sources.list` contains this (without comments): 
```
deb https://archive.raspberrypi.org/debian/ stretch main ui
deb http://deb.debian.org/debian stretch main contrib non-free
deb-src http://deb.debian.org/debian stretch main contrib non-free
deb http://security.debian.org/ stretch/updates main contrib non-free
deb-src http://security.debian.org/ stretch/updates main contrib non-free
deb http://deb.debian.org/debian stretch-updates main contrib non-free
deb-src http://deb.debian.org/debian stretch-updates main contrib non-free
```
And I deleted my `raspi.list`
What are the repos I need to add?

MichaIng, 2019-02-12:
@tim54000 
Ah do not delete `raspi.list`, this is required for firmware updates!

To fix everything at once please do the following:
`/DietPi/dietpi/func/dietpi-set_software apt-mirror default`

tim54000, 2019-02-12:
Okay, I did that and I was running `apt-get update`.
I hope that will be enough and I thank you for your precious help

MichaIng, 2019-02-12:
@tim54000 
Jep that should be it. Just checked the other boot scripts for possibly wrong applied first run setup, but everything else is fine.

You could run G_AGUP to now update APT packages, but that should have been done already during prior update steps, so might just reply "everything up to date".