MichaIng/DietPi #2503: [FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.

kuerious, 2019-02-06:
--------------------------------------------------------------------
- DietPi has encountered an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Tue  5 Feb 17:45:51 MST 2019
- Bug report     | N/A
- DietPi version | v6.20.0 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | NULL (index=21)
- Kernel version | #1 SMP Debian 4.9.130-2 (2018-10-27)
- Distro         | stretch (index=4)
- Command        | chmod -R +x /DietPi/dietpi /var/lib/dietpi/services /etc/cron.daily/dietpi /etc/cron.hourly/dietpi
- Exit code      | 1
- Software title | DietPi-PREP

#### Steps to reproduce:

1. Install via Debian netinstall
2. Install pre-requisite using script in the link:
`apt-get update; apt-get install -y systemd-sysv ca-certificates sudo wget locales --reinstall`
3. Run the provided script:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
./PREP_SYSTEM_FOR_DIETPI.sh
```

#### Expected behaviour:

- The minimal Debian install should be converted into a DietPi install

#### Actual behaviour:

- 1. First an error message:
`https://i.imgur.com/o5SMzQu.png`
- 2. The script fails:
```
--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------
```

#### Extra details:

- Installing, updating, and running _locate_ finds this under the `/boot/dietpi` menu:
```
/boot/dietpi
/boot/dietpi-CHANGELOG.txt
/boot/dietpi-README.md
/boot/dietpi.txt
/boot/dietpi/.install_stage
/boot/dietpi/boot
/boot/dietpi/dietpi-autostart
...
```

#### Additional logs:
```
Log file contents:
chmod: cannot access '/DietPi/dietpi': No such file or directory
```

--------------------------------------------------------------------

kuerious, 2019-02-06:
So I plugged in the following:
```
mkdir /DietPi/dietpi/
ln -s /boot/dietpi/ /DietPi/dietpi
```

Attempted the script again, this time I get the following:
```
systemctl start dietpi-ramdisk Exit code 1
Job for dietpi-ramdisk.service failed because the control process exited with error code.
See "systemctl status dietpi-ramdisk.service" and "journalctl -xe" for details.
```

So, yeah ... I'm unsure about what to do now.

MichaIng, 2019-02-06:
@kuerious 
Thanks for your report. Jep indeed a wrong chmod target, I fixed it and merged into master: https://github.com/Fourdee/DietPi/pull/2506
Just redo with the steps, which will download the fixed DietPi-PREP version.

kuerious, 2019-02-06:
This bug #2503 points to bug #2506 ... which points back here, to 2503.

Where's the fix?

Fourdee, 2019-02-06:
@kuerious 

Use the dev branch for PREP script, then select master during GUI:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/dev/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
./PREP_SYSTEM_FOR_DIETPI.sh
```

MichaIng, 2019-02-06:
@kuerious @Fourdee 
I already merged it into master. Okay since it does not change any remaining DietPi code.
#2506 is a pull request, selecting "Files changed" shows the changed/fixed code line(s).

At the top you see "master from beta", which means that I merged the fixed code from beta branch (where I added the commit first) into the master branch (release branch).

Fourdee, 2019-02-06:
@MichaIng 

Ah nice one, thanks 👍 

Been unwell last few days, trying to catch up 😃

speeedturtle, 2019-03-31:
I have the same problem installing the latest version to my ARM device. 

DietPi-PREP: G_AGUP                                                                      
                     │  - Exit code: 100                                                                        
                     │  - DietPi version: v6.22.3 (MichaIng/master) | HW_MODEL:22 | HW_ARCH:3 | DISTRO:4        
                     │  - Image creator: n/a                                                                    
                     │  - Pre-image: n/a

MichaIng, 2019-03-31:
@speeedturtle 
This is not the same error 😉. Please paste the full output or rerun the plain APT command and paste all output of it:
`apt update`