MichaIng/DietPi #2505: kernel ip_forward disabled after reboot

maartenlangeveld, 2019-02-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | v6.20.6

#### Additional Information (if applicable)
- Software intalled: Pi-hole, OpenVPN

#### Steps to reproduce

Enable kernel IP Forwarding
1. echo 'net.ipv4.ip_forward=1' >> /etc/sysctl.conf
2. sysctl -p /etc/sysctl.conf
3. sysctl net.ipv4.ip_forward --> output : "net.ipv4.ip_forward = 1"
4. cat /proc/sys/net/ipv4/ip_forward --> output : "1"
5. reboot
6. grep 'net.ipv4.ip_forward' /etc/sysctl.conf --> output : "net.ipv4.ip_forward=1"
7. sysctl net.ipv4.ip_forward --> output : "net.ipv4.ip_forward = 0"
8. cat /proc/sys/net/ipv4/ip_forward --> output : "0"

#### Expected behaviour

- After reboot I expect ip forwarding still enabled since /etc/sysctl.conf  includes 'net.ipv4.ip_forward=1'

#### Actual behaviour

- After reboot ip forwarding is disabled despite /etc/sysctl.conf  includes 'net.ipv4.ip_forward=1'

#### Extra details

-

MichaIng, 2019-02-06:
@maartenlangeveld 
Thanks for your report. Indeed a very strange thing.

Which SBC/device do you use? Obviously something overrides the setting, after sysctl applied it.
Possibly there is a `/etc/sysctl.d/*.conf` drop-in config file, please check:
`grep 'net.ipv4.ip_forward' /etc/sysctl.d/*.conf`

maartenlangeveld, 2019-02-06:
Hi Micha,

SBC: NanoPi NEO2
grep 'net.ipv4.ip_forward' /etc/sysctl.d/*.conf gives no result...

Thanks, M

maartenlangeveld, 2019-02-06:
I have found a solution...

When I add a dediacted 01-ip_forward.conf file in /etc/sysctl.d/ with line "net.ipv4.ip_forward=1" the ip forward keeps enabled after reboot.
 
I have very very little knowledge of UNIX/Linux but probably /etc/sysctl.conf is not parsed as long as there are .conf files in /etc/sysctl.d/ ?

MichaIng, 2019-02-06:
@maartenlangeveld 
That is strange as well. Actually `/etc/sysctl.conf` itself is implemented via `/etc/sysctl.d` as symlink:
```
root@VM-Stretch:~# l /etc/sysctl.d/
total 12
lrwxrwxrwx 1 root root  14 Jan 15 10:59 99-sysctl.conf -> ../sysctl.conf
-rw-r--r-- 1 root root  16 Feb  6 12:32 dietpi.conf
-rw-r--r-- 1 root root  70 Oct  3 16:16 dietpi-disable_ipv6.conf
-rw-r--r-- 1 root root 639 Oct  3 16:16 README.sysctl
```

But anyway it is generally advised to use a separate `/etc/sysctl.d/*.conf` file for manual additions, so they can easily be reverted and `/etc/sysctl.conf` stays untouched, so can be automatically updated via APT, e.g. to apply new defaults or simply add/change comments about the settings.

maartenlangeveld, 2019-02-06:
@MichaIng,
Thanks! At mine's 99-sysctl.conf was not a symlink but a separate file. I have manually made the symlink now.

MichaIng, 2019-02-06:
@maartenlangeveld 
Ahhh, why ever this was the case. So then I think changes to `/etc/sysctl.conf` were simply ignored, since only `/etc/sysctl.d/*.conf` are parsed and the symlink exists as a backwards compatibility.

In your case the setting was applied only, since you actively forced the config file with `sysctl -p /etc/sysctl.conf`

Ref: https://www.freedesktop.org/software/systemd/man/systemd-sysctl.service.html
> When invoked with no arguments, /lib/systemd/systemd-sysctl applies all directives from configuration files listed in sysctl.d(5).

Ref: https://manpages.debian.org/stretch/systemd/sysctl.d.5.en.html
> /etc/sysctl.d/*.conf
> /run/sysctl.d/*.conf
> /usr/lib/sysctl.d/*.conf

No `/etc/sysctl.conf` 😉.

closed 2019-02-06 by MichaIng

coolduke666, 2019-04-29:
same issue here... installed dietpi last week and wondered why openvpn is not working... 
now made a symlink /etc/sysctl.conf to /etc/sysctl.d/99-sysctl.conf

is that correct? thx

MichaIng, 2019-04-29:
Jep this symlink should exist since /etc/sysctl.conf is not loaded anymore directly by systemd. I wonder why this link does not yet exist or is added by the related APT update. Seems to be relatively new. I add a patch to v6.23 to recreate the symlink if it does not exist.

OpenVPN should use it's dedicated /etc/sysctl.d/dietpi-openvpn.conf to add the redirects. I verify this is the case, otherwise patch it for new installs.

MichaIng, 2019-04-29:
@coolduke666
Hmm actually already with DietPi v6.22 the redirects are added via /etc/sysctl.d/dietpi-openvpn.conf (on fresh installs), so the symlink should not matter, although it should exist anyway. Can you verify this file exists?

coolduke666, 2019-04-29:
here my output: 

`ls -la /etc/sysctl.d/
total 24
drwxr-xr-x  2 root root 4096 Apr 29 12:33 .
drwxr-xr-x 84 root root 4096 Apr 29 12:29 ..
-rw-r--r--  1 root root   51 Apr 22 10:56 98-rpi.conf
lrwxrwxrwx  1 root root   14 Apr 29 12:32 99-sysctl.conf -> ../sysctl.conf
-rw-r--r--  1 root root   16 Apr 22 10:56 dietpi.conf
-rw-r--r--  1 root root   70 Apr 22 10:56 dietpi-disable_ipv6.conf
-rw-r--r--  1 root root  639 Apr 22 10:56 README.sysctl
`

coolduke666, 2019-04-29:
ls -la /etc/sysctl.d/
total 24
drwxr-xr-x  2 root root 4096 Apr 29 12:33 .
drwxr-xr-x 84 root root 4096 Apr 29 12:29 ..
-rw-r--r--  1 root root   51 Apr 22 10:56 98-rpi.conf
lrwxrwxrwx  1 root root   14 Apr 29 12:32 99-sysctl.conf -> ../sysctl.conf
-rw-r--r--  1 root root   16 Apr 22 10:56 dietpi.conf
-rw-r--r--  1 root root   70 Apr 22 10:56 dietpi-disable_ipv6.conf
-rw-r--r--  1 root root  639 Apr 22 10:56 README.sysctl

MichaIng, 2019-04-29:
@coolduke666 
Strange, did you install OpenVPN via first run installs (dietpi.txt automation settings) or after initial update has finished?

coolduke666, 2019-04-29:
hi

first I installed dietpi and after reboot I installed pihole and pivpn via
dietpi-software tool.

pihole working flawlessly. I only ran into troubles with pivpn. I needed
some time to figure out where the problem was namely the disabled
ip_forward setting.

HTH... best

MichaIng, 2019-04-29:
@coolduke666 
Ahh okay that explains it. With PiVPN we do not run our OpenVPN install code but only install the APT package and run the official PiVPN installer: https://raw.githubusercontent.com/pivpn/pivpn/master/auto_install/install.sh

Will check it by times if somehow IP redirection is missing and in case report it there or open a PR to add it.

coolduke666, 2019-04-30:
that would be great.

huntmanday, 2020-02-14:
@maartenlangeveld @MichaIng @coolduke666 

Hey guys (or gals), just popping in to say that I just ran into this same exact issue, and after a quick google search about it, I found this github issue. I'm new to github so please excuse my ignorance if commenting here like this isn't the correct way, etc.

The solution that @maartenlangeveld came up with symlinking /etc/sysctl.conf to /etc/sysctl.d/99-sysctl.conf worked like a charm, and now after rebooting my pi3, the net.ipv4.ip_forward=1 setting stays persistent.

Quick Recap of how I got here:

1. Installed latest DietPi (Current version : v6.28.0) on my RPi 3 Model B (armv7l)
2. Finished the DietPi setup and installation WITHOUT installing PiVPN during setup process

3. After everything was done, fully updated and I'd rebooted a few times, I installed PiVPN via dietpi-software

4. Configured PiVPN with Wireguard, setup my configs, realized I wasn't able to connect to other devices on my subnet 192.168.1.0/24

5. Ran pivpn -d and got the:
```
:: [ERR] IP forwarding is not enabled, attempt fix now? [Y/n] Y
net.ipv4.ip_forward = 1
Done
```

6. After that, I did not reboot since everything was now working correctly.

DAYS LATER

7. I needed to reboot my pi for an unrelated reason, figured everything was okay since pivpn was set to autostart and I was able to connect to wireguard from my clients, but I realized I couldn't connect to any other devices.... "Did net.ipv4.ip_forward setting get disabled?"

8. Sure enough, after running pivpn -d , I got the same message that it was disabled, and did I want to enable it. Hit Y, everything is good again.

9. After grepping around trying to find what file this specific parameter was being defined in, turned to google and found this.

Here's what I had to do (spacing stuff out to make it look better):

Tried to immediately make the symbolic link, but it wouldn't since `/etc/sysctl.d/99-sysctl.conf` already existed:
```
root@myrpi3host:/etc# ln -s /etc/sysctl.conf /etc/sysctl.d/99-sysctl.conf
ln: failed to create symbolic link '/etc/sysctl.d/99-sysctl.conf': File exists
```

So I checked the difference between them to see if removing it first would be dangerous:
```
root@myrpi3host:/etc# diff /etc/sysctl.conf /etc/sysctl.d/99-sysctl.conf
28c28
< net.ipv4.ip_forward=1
---
> #net.ipv4.ip_forward=1
```

Nope, safe to proceed:
```
root@myrpi3host:/etc# cd /etc/sysctl.d/
root@myrpi3host:/etc/sysctl.d# ls
98-rpi.conf  99-sysctl.conf  dietpi.conf  protect-links.conf  README.sysctl

root@myrpi3host:/etc/sysctl.d# rm -rf 99-sysctl.conf
root@myrpi3host:/etc/sysctl.d# ln -s /etc/sysctl.conf /etc/sysctl.d/99-sysctl.conf
root@myrpi3host:/etc/sysctl.d# 
```

Now after a reboot, everything sticks and I can connect to wireguard and all my devices on my LAN perfectly. Any way we can get this to be automatically baked in for new users? Thanks

MichaIng, 2020-02-14:
@huntmanday 
Many thanks for reporting. Sounds like PiVPN installer is the culprit here which removes/replaces the symlink by a file. Not sure why it comments out IPv4 forwarding, since this is of course required to grant internet access for VPN clients... Probably I find some time to check their install script and raise an issue and/or PR for fix.

huntmanday, 2020-02-14:
@MichaIng Sounds good. Glad I could help!

Joulinar, 2020-02-14:
@MichaIng  
this issue regarding `net.ipv4.ip_forward=1` I have seen on the forum board as well. What is still strange for me, that it is needed for WireGuard. On the board we found that it is needed for OpenVPN. For me personally WireGuard is working with the following setting. I'm able to connect with my mobile clients and access internet as well as intranet. But I have WireGuard installed from `dietpi-software` and not via `PiVPN`

```
root@DietPi4:/etc# cat sysctl.conf|grep ipv4.ip
#net.ipv4.ip_forward=1
root@DietPi4:/etc#
```
The board entried for your reference:

https://dietpi.com/phpbb/viewtopic.php?f=9&t=7251

https://dietpi.com/phpbb/viewtopic.php?f=11&t=7254&p=22279&hilit=wireguard#p22279

MichaIng, 2020-02-14:
@Joulinar 
Yes in case of WireGuard, applying this setting system-wide is not required, since we apply it right via the server config on service start, for the involved network interfaces:
```
PostUp = sysctl net.ipv4.conf.%i.forwarding=1 net.ipv4.conf.\$(sed -n 3p /DietPi/dietpi/.network).forwarding=1
PostUp = sysctl net.ipv6.conf.\$(sed -n 3p /DietPi/dietpi/.network).accept_ra=2
PostUp = sysctl net.ipv6.conf.%i.forwarding=1 net.ipv6.conf.\$(sed -n 3p /DietPi/dietpi/.network).forwarding=1
```

Joulinar, 2020-02-14:
@MichaIng 
Ahhhh now it's getting clear (I hope). As WireGuard is available via PiVPN as well, this system wide setting is not done. Means for both OpenVPN as well as WireGuard installations done by PiVPN, `net.ipv4.ip_forward=1` would need to be set in `/etc/sysctl.conf`. Correct?

MichaIng, 2020-02-14:
@Joulinar 
Ah, I didn't know that PiVPN supports WireGuard as well now, great. Jep for this the system-wide setting is required then.

Joulinar, 2020-02-14:
@MichaIng 
Yes discovered it last days during some OpenVPN issue investigation.

Joulinar, 2020-02-14:
@MichaIng 
just to confirm. I did a quick comparison between both version (`dietpi-software` and PiVPN) of WireGuard `wg0.conf`. And yep the PiVPN version contains basically nothing except the keys and IP's 😃