MichaIng/DietPi #2507: Issue installing LASP

XRay437, 2019-02-06:
#### Details:
- Date           | Wed  6 Feb 17:20:49 GMT 2019
- Bug report     | N/A
- DietPi version | v6.20.6 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | Unknown Device (armv7l) (index=0)  (I've a RPI 3B+, why is not recognized?)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | buster (index=5)
- Command        | G_AGI: sqlite3
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

dietpi-software, select LASP and install
#### Expected behaviour:
Install LASP
#### Actual behaviour:

`[  OK  ] DietPi-Services | unmask all: cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : apache2
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : proftpd
[  OK  ] DietPi-Services | stop : fail2ban

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT

[  OK  ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://archive.raspbian.org/raspbian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian sid InRelease
Reading package lists...

[  OK  ] DietPi-Software | G_AGUP
[ INFO ] DietPi-Software | Running apt simulation to check for errors, please wait...
[  OK  ] DietPi-Software | APT installation for: bash-doc -s, please wait...

[  OK  ] DietPi-Software | G_AGI: bash-doc -s
[  OK  ] DietPi-Software | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-Software | G_AGUG

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : apache2
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : proftpd
[  OK  ] DietPi-Services | stop : fail2ban
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : apache2
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : proftpd
[  OK  ] DietPi-Services | stop : fail2ban

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Apache2: webserver

[  OK  ] DietPi-Software | APT installation for: apache2 python-certbot-apache, please wait...

[  OK  ] DietPi-Software | G_AGI: apache2 python-certbot-apache

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing SQlite: database

[  OK  ] DietPi-Software | APT installation for: sqlite3, please wait...
E: Unable to correct problems, you have held broken packages.

[FAILED] DietPi-Software | G_AGI: sqlite3`

#### Extra details:
I've previoulsy unistalled lighttpd because it wasn't starting correctly (for my PiHole) and now I'm trying to reinstall it.

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```

--------------------------------------------------------------------
#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
root@DietPi:~# cat /DietPi/dietpi/.version
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=20
G_DIETPI_VERSION_RC=6
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
root@DietPi:~# echo $G_DISTRO_NAME` or `cat /etc/debian_version
-bash: or: comando non trovato
bustercat /etc/debian_version

- Kernel version | `uname -a`
Linux DietPi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
Unknown Device (armv7l)

- Power supply used | 5V 2.5A
- SDcard used | Samsung Class 10

#### Additional Information (if applicable)
- LASP
- Uninstalled, then tried to reinstall 
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`:

5abc24f7-3a74-4c1e-a891-0c185bc56309

I've tried to manually reinstall libsqlite3-0 without success since its seems to be the "blocking" pkg, otherwise if I try to remove it:

root@DietPi:~# apt-get install libsqlite3-0       
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze       
Lettura informazioni sullo stato... Fatto
libsqlite3-0 is already the newest version (3.26.0+fossilbc891ac6b-2).
0 aggiornati, 0 installati, 0 da rimuovere e 0 non aggiornati.
root@DietPi:~# apt-get install -y libsqlite3-0
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze       
Lettura informazioni sullo stato... Fatto
libsqlite3-0 is already the newest version (3.26.0+fossilbc891ac6b-2).
0 aggiornati, 0 installati, 0 da rimuovere e 0 non aggiornati.
root@DietPi:~# apt-get remove libsqlite3-0
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze       
Lettura informazioni sullo stato... Fatto
I seguenti pacchetti sono stati installati automaticamente e non sono più richiesti: "huge number of pkgs"
....
ATTENZIONE: i seguenti pacchetti essenziali stanno per essere rimossi.
Questo non dovrebbe essere fatto a meno che non si sappia esattamente cosa si sta facendo.
  apt raspbian-archive-keyring (a causa di apt)
0 aggiornati, 26 installati, 90 da rimuovere e 0 non aggiornati.
È necessario scaricare 2.978 kB di archivi.
Dopo quest'operazione, verranno liberati 494 MB di spazio su disco.
Si sta per compiere un'azione potenzialmente pericolosa.
Per continuare scrivere la frase "Sì, esegui come da richiesta."
 ?] 
Sorry for the italian bash language.

XRay437, 2019-02-06:
Hello, I think that my sources.list was messed up.
Following several topics, I've ended with  "apt-get install debian-keyring debian-archive-keyring && apt-get update". After few reboots, I've removed all webserver/DB software, performed a "apt-get REMOVE debian-keyring debian-archive-keyring", apt-get update and then the install completed correctly. TY

closed 2019-02-06 by XRay437

MichaIng, 2019-02-06:
@XRay437 
`debian-keyring` should be not required. It is the developer/maintainer keyring, for uploading packages to the repo, AFAIK.
`debian-archive-keyring` indeed is required to download packages from the repo, so you should not remove it. If done, reinstall, otherwise further APT steps should fail: `G_AGI debian-archive-keyring`