MichaIng/DietPi #2508: dietpi-drive_manager keeps forgettind and adding mounts

ilgrank, 2019-02-06:
Hi
After the upgrade from 6.17.20 to 6.20 (but could be unrelated, I simply noticed after the upgrade) a couple of strange things began to happen:
1) dietpi-drive_manager doesn't remove entries of mounted smb shares (or better, when I do remove them, they disappear, but after a while they do re-appear in the list of mounted shares) 
I tried removing them several times, but after a reboot or two, old entries do re-appear.

2) mounted smb shares do un-mount. Forcing a mount -a does nothing, but running dietpi-drive_manager (just opening it and quitting) does the trick.. for a while. 
(every evening after coming back from work I do find my emby server complaining it can't access the shares anymore)

Is there anything I can do to investigate this? Is there a log of drive_manager operations?
Thanks!

Fourdee, 2019-02-06:
@ilgrank 

Hi, many thanks for the report 👍 

Ok, so when unmounting SMB/NFS shares, DietPi also removes the entry from ```/etc/fstab```. 

Might be a scrape issue with ```/etc/fstab```, please can you paste results of:
```
cat /etc/fstab
```

I'll also try to replicate issue here.

Fourdee, 2019-02-06:
Issue is drive_manager not dismounting before refresh.

Fourdee, 2019-02-06:
@ilgrank 

Resolved for v6.21.

You can also apply the fix now with:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/fbdf165270f03af65b284942de0bcf3df3f61a0e/dietpi/func/dietpi-globals -O /DietPi/dietpi/func/dietpi-globals
```

Then load ```dietpi-drive_manager``` to remove the drive.

closed 2019-02-06 by Fourdee

ilgrank, 2019-02-06:
Wow! I didn't even get to arrive home and you had already a fix out!  🥇 
 Testing immediately but meanwhile.. THANKS!! 👍

ilgrank, 2019-02-10:
**UPDATE Feb 10**: after 3 days the SMB Shares remained mounted, so both issues are definitely resolved
again, thanks @Fourdee