MichaIng/DietPi #2512: Software Request: ISC+Glass DHCP Server

ethanpil, 2019-02-07:
#### Formal software information
- Software title | ISC and Glass Frontend
- Short description | DHCP Server and Web Based Frontend
- Official URL (if available) | https://github.com/Akkadius/glass-isc-dhcp
- GitHub URL (if available) | https://github.com/Akkadius/glass-isc-dhcp
- (Official) install documentation | https://github.com/Akkadius/glass-isc-dhcp#installation

#### Are there similar/alternative software titles available with DietPi-Software?
None.

#### What makes your requested software better than the above solutions, if available?
This is a great additional tool for DietPI that can work alongside PiHole... DNS and now DHCP services

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
According to the official install instructions there are a lot of manual steps to do: Setting up a Node stack and configure them accordingly. DietPi could automate those steps, which saves end users a lot of time and allows installation as well to less experienced Linux users.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. Install NodeJS
2. Install Glass (as root)
3. Open Firewall port
4. Setup default settings
5. Setup system daemon and keepalive

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
Yes, I could help out with this.

MichaIng, 2019-02-07:
@ethanpil 
Thanks for your request.

Do I get it right that Glass is moreless web UI frontend for the `isc-dhcp-server`?
Ah actually that explains it:
> Glass was created as a quick side project out of necessity to fulfill requirements not readily available through our production ISC DHCP Servers for our organization - I've decided to share it for anyone who would also find it useful

So its a fork with additional features, e.g. web UI 😉.
_________
Actually with the WiFi HotSpot we already install `isc-dhcp-server` to provide DHCP functionality. Makes sense to as well allow this for wired connected devices.
`glass-isc-dhcp` could then be an alternative, if those additional features are desired.

ethanpil, 2019-02-07:
@MichaIng thanks for the interest. I think may people, including myself, use DietPI for network services... PiHole, LAMP/LEMP, Hotspot, etc.

So I am thinking it would be great to have single instance network services... And thats how I found Glass. Thanks!

MichaIng, 2021-05-27:
Just as Pi-hole was mentioned: Since Pi-hole ships an internal DHCP server and has a web UI, there is no point to use Glass with it. But when no Pi-hole is used, it is nice of course.

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not addressed soon. The FeatHub request stays however open to allow votes and if anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng