MichaIng/DietPi #2515: USB 3 on DietPi?

kuerious, 2019-02-08:
I have almost completely converted new server and fan builds sensimilla Debian to diet pi. I just now ran into my first major issue, I am unable to have a usb3 external drive be seen by the operating system. Is there something special I'm the same?

I will take screenshots, provide system specs, and hand over more info in the morning. All that I remember about today in my hours of trying were seeing something about xhci = m

MichaIng, 2019-02-08:
@kuerious 
Thanks for your report.

Jep, please provide some more specs about the system (SBC), drive and power supply, to allow us investigate.

Just some general thoughts:
- If it's a 2,5" or 3,5" drive, attached to an SBC, not that you need an external power supply for the drives. The SBC USB slots usually do not provide enough power for more than a USB stick.
- Check `fdisk -l` if the drive is recognized at all by the system.
- Check `dmesg` for any kernel/device related errors, e.g. power/voltage, I/O errors, driver errors and such.

kuerious, 2019-02-08:
Okay, time for details.
- PC is HP EliteDesk 800 G1 USDT; one built-in SATA/power, a 2nd SATA port but 4-pin power
- This is a 2.5" SSD, very low power needs
- I was able to "see" the disk, delete data, and show the correct size under Windows
- I have a USB-to-SATA_power adapter; I tried w/ a SATA connector, still no joy

I'll boot it on up and do an output of the above momentarily.

kuerious, 2019-02-08:
Results of `fdisk -l` are as follows:

```
root@docker1:~# fdisk -l
Disk /dev/sdb: 59.6 GiB, 64023257088 bytes, 125045424 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt

Disk identifier: 70902C9A-A893-4CF2-816E-0DE039FB2288
Device       Start      End  Sectors  Size Type
/dev/sdb1     2048  1050623  1048576  512M EFI System
/dev/sdb2  1050624 29587455 28536832 13.6G Linux filesystem

Disk /dev/sda: 111.8 GiB, 120040980480 bytes, 234455040 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 6EDB5609-6D5F-4403-B841-B429FE75390D

Device       Start      End  Sectors  Size Type
/dev/sda1     2048  1050623  1048576  512M EFI System
/dev/sda2  1050624 60620766 59570143 28.4G Linux filesystem
```

root@docker1:~#

kuerious, 2019-02-08:
What follows is the output of dmesg of any- and everything "USB"-related:

```
root@docker1:~# dmesg

[    1.565327] ACPI: bus type USB registered
[    1.565341] usbcore: registered new interface driver usbfs
[    1.565346] usbcore: registered new interface driver hub
[    1.565361] usbcore: registered new device driver usb
[    1.565728] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.566185] ehci-pci: EHCI PCI platform driver
[    1.566254] libata version 3.00 loaded.
[    1.571692] AVX2 version of gcm_enc/dec engaged.
[    1.571692] AES CTR mode by8 optimization enabled

[    1.778348] ehci-pci 0000:00:1a.0: EHCI Host Controller
[    1.778352] ehci-pci 0000:00:1a.0: new USB bus registered, assigned bus number 1
[    1.778369] ehci-pci 0000:00:1a.0: debug port 2
[    1.782291] ehci-pci 0000:00:1a.0: cache line size of 64 is not supported
[    1.782305] ehci-pci 0000:00:1a.0: irq 16, io mem 0xf7c2c000
[    1.797562] ehci-pci 0000:00:1a.0: USB 2.0 started, EHCI 1.00
[    1.797603] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    1.797604] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.797605] usb usb1: Product: EHCI Host Controller
[    1.797606] usb usb1: Manufacturer: Linux 4.9.0-8-amd64 ehci_hcd
[    1.797607] usb usb1: SerialNumber: 0000:00:1a.0
[    1.797767] hub 1-0:1.0: USB hub found
[    1.797771] hub 1-0:1.0: 3 ports detected
[    1.797970] ehci-pci 0000:00:1d.0: EHCI Host Controller
[    1.797973] ehci-pci 0000:00:1d.0: new USB bus registered, assigned bus number 2
[    1.797982] ehci-pci 0000:00:1d.0: debug port 2

[    1.801881] ehci-pci 0000:00:1d.0: cache line size of 64 is not supported
[    1.801899] ehci-pci 0000:00:1d.0: irq 23, io mem 0xf7c2b000
[    1.817557] ehci-pci 0000:00:1d.0: USB 2.0 started, EHCI 1.00
[    1.817590] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002
[    1.817592] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.817593] usb usb2: Product: EHCI Host Controller
[    1.817594] usb usb2: Manufacturer: Linux 4.9.0-8-amd64 ehci_hcd
[    1.817594] usb usb2: SerialNumber: 0000:00:1d.0
[    1.817750] hub 2-0:1.0: USB hub found
[    1.817753] hub 2-0:1.0: 3 ports detected
[    1.817891] ahci 0000:00:1f.2: version 3.0
[    1.818042] ahci 0000:00:1f.2: AHCI 0001.0300 32 slots 5 ports 6 Gbps 0x29 impl SATA mode
[    1.818045] ahci 0000:00:1f.2: flags: 64bit ncq pm led clo pio slum part ems apst
[    1.842177] scsi host0: ahci
[    1.842384] scsi host1: ahci
[    1.842566] scsi host2: ahci
[    1.842675] scsi host3: ahci
[    1.842749] scsi host4: ahci
[    1.842824] scsi host5: ahci
[    1.842862] ata1: SATA max UDMA/133 abar m2048@0xf7c2a000 port 0xf7c2a100 irq 27
[    1.842863] ata2: DUMMY
[    1.842864] ata3: DUMMY
[    1.842866] ata4: SATA max UDMA/133 abar m2048@0xf7c2a000 port 0xf7c2a280 irq 27
[    1.842866] ata5: DUMMY
[    1.842868] ata6: SATA max UDMA/133 abar m2048@0xf7c2a000 port 0xf7c2a380 irq 27
[    1.842973] i801_smbus 0000:00:1f.3: SPD Write Disable is set
[    1.842997] i801_smbus 0000:00:1f.3: SMBus using PCI interrupt
[    2.125571] usb 1-1: new high-speed USB device number 2 using ehci-pci
[    2.145557] usb 2-1: new high-speed USB device number 2 using ehci-pci
[    2.157206] ata4: SATA link up 1.5 Gbps (SStatus 113 SControl 300)
[    2.157233] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    2.157256] ata6: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    2.158278] ata1.00: ATA-9: SanDisk SSD PLUS 120GB, UE4500RL, max UDMA/133
[    2.158282] ata1.00: 234455040 sectors, multi 1: LBA48 NCQ (depth 31/32)
[    2.158429] ata6.00: ACPI cmd ef/10:06:00:00:00:00 (SET FEATURES) succeeded
[    2.158434] ata6.00: ACPI cmd f5/00:00:00:00:00:00 (SECURITY FREEZE LOCK) filtered out
[    2.158438] ata6.00: ACPI cmd b1/c1:00:00:00:00:00 (DEVICE CONFIGURATION OVERLAY) filtered out
[    2.158927] ata6.00: ATA-9: SAMSUNG MZMPC064HBDR-00000, CXM1201Q, max UDMA/133
[... 41 lines cut ...]
[    2.683698] usb 1-1.6: New USB device found, idVendor=03f0, idProduct=344a
[    2.683700] usb 1-1.6: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    2.683701] usb 1-1.6: Product: HP USB Slim Keyboard

[    3.192017] usbcore: registered new interface driver usbhid
[    3.192018] usbhid: USB HID core driver
[    3.205974] fbcon: inteldrmfb (fb0) is primary device
[    3.217626] iTCO_vendor_support: vendor-support=0
[    3.219474] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[    3.219493] iTCO_wdt: Found a Lynx Point TCO device (Version=2, TCOBASE=0x0460)
[    3.219635] iTCO_wdt: initialized. heartbeat=30 sec (nowayout=0)
[    3.222364] input: HP WMI hotkeys as /devices/virtual/input/input10
[    3.225265] intel_rapl: Found RAPL domain package
[    3.225266] intel_rapl: Found RAPL domain core
[    3.225267] intel_rapl: Found RAPL domain uncore
[    3.225268] intel_rapl: Found RAPL domain dram
[    3.225270] intel_rapl: RAPL package 0 domain package locked by BIOS
[    3.225277] intel_rapl: RAPL package 0 domain dram locked by BIOS
[    3.231931] input: HP USB Slim Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:03F0:344A.0001/input/input11
[    3.265869] Console: switching to colour frame buffer device 160x64
[    3.281099] i915 0000:00:02.0: fb0: inteldrmfb frame buffer device
[    3.289748] hid-generic 0003:03F0:344A.0001: input,hidraw0: USB HID v1.10 Keyboard [HP USB Slim Keyboard] on usb-0000:00:1a.0-1.6/input0
[    3.291163] input: HP USB Slim Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.1/0003:03F0:344A.0002/input/input12
[    3.313178] input: HDA Intel HDMI HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:03.0/sound/card0/input13
```

kuerious, 2019-02-13:
Is this a low-pri, or did I neglect to include/exclude some information, or ...

MichaIng, 2019-02-13:
@kuerious 
Sorry for delay, was distracted here and there.
```
[    2.157206] ata4: SATA link up 1.5 Gbps (SStatus 113 SControl 300)
[    2.157233] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    2.157256] ata6: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
```
- 128M, 64M and ROM

Hmm yeah seems just the USB keyboard is detected.
```
[    1.797771] hub 1-0:1.0: 3 ports detected
[    1.817753] hub 2-0:1.0: 3 ports detected
[    2.278506] hub 1-1:1.0: 6 ports detected
[    2.294219] hub 2-1:1.0: 8 ports detected
[    2.683701] usb 1-1.6: Product: HP USB Slim Keyboard
```

Does the same external drive work on  different USB port, e.g. non-USB3.0?

What does `lsusb` show?

MichaIng, 2019-02-28:
@kuerious 
Due to some other reason I stumbled over UAS (USB-attached-SATA) support on RPi, which requires two kernel modules: `sg` and `uas`
Could you verify these are active on your system:
lsmod | grep -E '^(sg|uas)'

In case try to enable them (`uas`  depends on `sg`, AFAIK):
```
modprobe sg
modprove uas
```

MichaIng, 2019-05-06:
I mark this a closed. Feel free to reopen if required.

closed 2019-05-06 by MichaIng