MichaIng/DietPi #2516: Dietpie | Software: Pihole upgrade problem

jonare77, 2019-02-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
`G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee`
- Distro version | `9.7`
- Kernel version | `Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux`
- SBC device | `Virtual Machine (x86_64)`

#### Additional Information (if applicable)
- Software title | Pihole
- Was the software title installed freshly or updated/migrated: Freshly
- Can this issue be replicated on a fresh installation of DietPi: Not sure
- `9b1ac4a2-8c3b-4d4c-a6b4-fa83e8a54b2b` ID

#### Expected behaviour

- Running "pihole -up" should upgrade Pihole.

#### Actual behaviour

- Pihole is throwing: ": Could not update local repository. Contact support."

MichaIng, 2019-02-08:
@jonare77 
Thanks for your report.

I will run local tests here.

MichaIng, 2019-02-08:
Okay I found the reason: https://github.com/pi-hole/pi-hole/issues/2592#issuecomment-461795981

Do the following to update:
```
cd /var/www/html/admin
git reset --hard HEAD
pihole -up
```

@Fourdee 
We must not change any webroot file permissions via: `/DietPi/dietpi/func/dietpi-set_software setpermissions`
I already thought that before, as well for `/mnt/dietpi_userdata` content. Many software titles installed via APT, installers or Git client, apply certain specific permissions to each file. We break security ideas and as above internal updaters/installers, when overriding those permissions with every `dietpi-software` run.
As long as we use `mv`, `cp -a` or `rsync -a` to handle files, permissions are preserved.
Instead I think we should only apply permissions to software titles, downloaded as non-permissions capable archive (e.g. `tar` includes permissions, `zip` not) and of course change group to `dietpi` where required `chown :dietpi` during the install itself, not afterwards anymore. Since we do not allow to move `dietpi_userdata` to non-permissions capable file systems, there should be no reason for this afterwards.

Fourdee, 2019-02-08:
@MichaIng 

> /DietPi/dietpi/func/dietpi-set_software setpermissions

The whole purpose of the above is a "cover all" method designed for simple patching and consistency. 
Ensuring that if users change permissions accidentally (or on purpose), the above will reset everything we install back to a working condition.

I agree we could look at moving over to ```.tar``` for some software titles, then no change for permissions required.

> We break security ideas

I somewhat disagree. ```/var/www``` as ```www-data``` is standard use.  In terms of ```dietpi_userdata```, this is our system.

Although, no harm in us verifying all our permissions set, match stock install APT etc where required.

MichaIng, 2019-02-08:
@Fourdee 
> Ensuring that if users change permissions accidentally (or on purpose), the above will reset everything we install back to a working condition.

Generally this makes sense, but indeed there are several software titles requiring or expecting quite specific permissions.

> I agree we could look at moving over to `.tar` for some software titles, then no change for permissions required.

I already did that in some cases, also because they are better packed (smaller download) and some internal checksum test is possible (?).
Generally if the software is installed by downloading an archive, most likely no specific permissions are required (besides run oder and execution, based on our systemd unit). If (only) tar is available, at least specific permissions are most likely included already.

But only in case of Git clones and APT installs, specific permissions might be really required. Git actively checks file permissions before allowing checkout and such without complains. This is not a Pi-hole issue, but Git internal and new, AFAIK. At least we never faced that before.

> I somewhat disagree. `/var/www` as `www-data` is standard use.

Again, generally I agree. However there are as well other software titles that explicitly expect certain scripts and config files not being R/W accessible for `www-data`. E.g. ownCloud/Nextcloud/Pydio solve this via specific (but quite complicated) webserver configurations, and Pi-hole obviously solves this by having those files owned by `root`. So it's not about what is the usual or consistent way to achieve desired security, but about how different software titles in fact implement this 🤔.

In case of `dietpi_userdata` it is as well only relevant for the cases where we drop the whole install dir of a software title into `dietpi_userdata`. E.g. having all files R/W/X accessible (at least for dietpi user) might be easy/simple but not the safest solution.

> Although, no harm in us verifying all our permissions set, match stock install APT etc where required.

The problem with trying to match everything exactly is that permissions by times are veeery individual for each file. Impossible or very high effort to add all that single files/dirs and modes to the script and an unacceptable high maintenance effort to keep those up-to-date.
__________
Perhaps we can do like that:
- Keep `setpermissions` command available.
- Copy all software related permissions setting to their install/config code.
- Remove setpermissions from `dietpi-set_userdata` and `dietpi-software` default run.
- If users run into issues, we can offer the command. If no related issues reported for a while, we can remove `setpermissions` completely.

MichaIng, 2019-02-20:
PR up for fix: https://github.com/Fourdee/DietPi/pull/2570
- I now just added `git reset --hard HEAD` to the permissions script, to have a solution aside from the discussion if we should use it or not.

closed 2019-03-23 by MichaIng