MichaIng/DietPi #2517: usbip no longer working

TBail, 2019-02-08:
### Creating a bug report/issue
After updateing a raspberrypi Version 1 vpm Dietpi 6.19.7 to latest 6.21.1 my usbip setup is not longer workung. Conneting to this raspberry from another machine results in the following error message

`usbip: error: could not connect to SomeIP:3240: System error`

On the the DietPi System i see no errors as far as i am able to see them

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version
stretch

- Kernel version
Linux btscanner02 4.14.79+ #1159 Sun Nov 4 17:28:08 GMT 2018 armv6l GNU/Linux

- SBC device
RPi B (armv6l)

- Power supply used
Some kind of 2A Samsung USB Powersupply

- SDcard used
SanDisk Externe 16GB

#### Additional Information (if applicable)
- Software title
usbip
- Was the software title installed freshly or updated/migrated?
Updated
- Can this issue be replicated on a fresh installation of DietPi?
Yes, tried that with y fresh 6.20.6
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`
7c4f85b2-b2b1-4dcd-b881-168a07585dbb

#### Steps to reproduce
Install ist on an older Distro and the upgrade to latest.

#### Expected behaviour
Usb Device should be forwarded to anothe system

#### Actual behaviour
Cliet Software on other system annot connect

#### Extra details

Fourdee, 2019-02-08:
@TBail 

Hi,

Not had any experience with ```usbip```, however:
On the RPi, is there a service running for ```usbip```?

If not, try re-running the daemon:
```
usbipd -D
```

TBail, 2019-02-08:
@Fourdee 
Indeed the usbipd was not running, but should be loaded in rc.local. This is working with version 6.19.7 .
Used the following code
`usbipd -D`
in rc.local. Changed it to
`/usr/sbin/usbipd -D`
but while rebooting it did not start. So i put some echo statements into the rc.local to create a log. And as far as i understand it the rc.local is never executed.

Fourdee, 2019-02-08:
@TBail 

Strange, please check the ```rc.local``` service:
```
systemctl status rc.local -l
```

Fourdee, 2019-02-08:
Simple systemD service I created, this replaces ```rc.local``` and offers improved logging with:
```
systemctl status usbip -l
```

### Copy + paste all
```
cat << _EOF_ > /etc/systemd/system/usbip.service
[Unit]
Description=usbip
After=network.target network-online.target

[Service]
Type=simple
ExecStart=$(which usbipd)

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
systemctl enable usbip.service
systemctl start usbip
```

MichaIng, 2019-02-08:
Found some more user reports about disabled `rc.local`. It is disabled by default on fresh Debian, but I also found user reports where it was working before an update. Perhaps a recent `systemd` update actively disabled it or changed something. `systemctl enable rc-local` however still works.

But yeah the feature is deprecated and own systemd unit preferred.
Or use DietPi-PostBoot implementation: `/var/lib/dietpi/postboot/my_script`

TBail, 2019-02-11:
because of the deprication of rc.Local i moved over to a own systemd service. This works perfect.

Thnaks for your help.

Thomas

closed 2019-02-11 by TBail