MichaIng/DietPi #2518: Fresh install of piVPN, unable to reach network/internet

XRay437, 2019-02-08:
#### Required Information
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee
- stretch
- Linux DietPi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
-RPi3B+

#### Additional Information (if applicable)
- Software title | piVPN
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes
-  4dbf4340-cc8a-4b55-b691-377e8007f2bb

#### Steps to reproduce
1. Fresh install of DietPi, piHole
2. Configure piHole
3. Install piVPN
4. Follow the piVPN wizard

#### Expected behaviour
Full VPN tunnel using piVPN.
Currently using an iOS client (on a previous dietpi/pivpn install everything was working), in order to reach both LAN and internet (using piHole DNS).

#### Actual behaviour
Using the .ovpn file I'm able to connect to the VPN on my iPhone, but I'm only able to ping the VPN end-point (10.8.0.1). 
No ping towards gateway, pi's local IP, no DNS resolution with piHole, no DNS resolution (using 1.1.1.1), no internet ping.
Using the stock config (as reported on the dietpi-software page).

Any advice?
Thank you!

MichaIng, 2019-02-08:
@XRay437 
Thanks for your report.

We will investigate.

XRay437, 2019-02-08:
It's very strange, after the installation the ip_forwarding seems to be still 0.
As reported here: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5509, after enabling it, everything seems to works!
During the piVpn/openvpn setup the ip_forwarding should be automatically set?
Otherwise, looking on the dietpi-software post on the forum, it would be appropriate to write a configuration reminder!

MichaIng, 2019-02-08:
@XRay437 
> During the piVpn/openvpn setup the ip_forwarding should be automatically set?

During PiVPN/OpenVPN setup it is not by default, AFAIK, since as well by default only requests to the VPN server are tunnelled, nothing else.
I check our docs and in case add the hint that this needs to be enabled (and how to do best).

During WireGuard install on the other hand it is enabled by default, as well the default client configs tunnel all traffic. Hmm we should somehow align both VPN solutions to have the same default behaviour 🤔.

XRay437, 2019-02-08:
> @XRay437
>
>
> During PiVPN/OpenVPN setup it is not by default, AFAIK, since as well by default only requests to the VPN server are tunnelled, nothing else.
> I check our docs and in case add the hint [...]

Yes, I think it's should be appropriate to align both VPNs!
Anyway, I've just reinstalled DietPi, so before trying Wireguard again I'll perform a complete backup!

XRay437, 2019-02-08:
@MichaIng something is still broken, after a reboot, the situation reverted to the OP, no network, no browsing

MichaIng, 2019-02-08:
@XRay437 
Note that sysctl.conf is actually an outdated file. For compatibility reasons is it symlinked usually from within /etc/sysctl.d/99-sysctl.conf. Not sure why, but in some cases this symlink does not exist, so sysctl.conf is not parsed at all on boot. At best use a dedicated file, like we do on WireGuard install, e.g. `echo 'net.ipv4.ip_forward = 1' > /etc/sysctl.d/ipv4_forward.conf`

MichaIng, 2019-02-10:
I will mark this as closed. Feel free to use or reopen, if required.

OpenVPN will now add the ip_forwarding the above way automatically: https://github.com/Fourdee/DietPi/pull/2526/files#diff-d92a6ee04e02fd2a2dc23d5bec3e6a98R10194

On PiVPN we use their install script, which uses still `sysctl.conf`: https://github.com/pivpn/pivpn/blob/master/auto_install/install.sh#L909
- I will see if I can open a PR there to switch to sysctl.d and until then we could check for and in case recreate the required symlink `/etc/sysctl.d/99-sysctl.conf` -> `/etc/sysctl.conf`

closed 2019-02-10 by MichaIng