MichaIng/DietPi #2519: DietPi-Drive_Manager | Allow NFS (and Samba?) clients to mount sub directories

MichaIng, 2019-02-08:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=12&t=5488

Should be easy to add. Of course the user should assure that this directory is available on the server. Perhaps doing a test mount, and in case re-prompt for host/path entry.

echable9999, 2019-05-20:
I agree with this, essential feature. 

Relatedly, I'm getting confusing inconsistent results with NFS mapping - on two separate, identical Pies, with fresh DietPi installed with nothing else, one will mount NFS from my NAS, the other not. Don't know how to troubleshoot this. 

And the separate NFS and SAMBA client softwares that can be installed - are they automatically installed when you try to mount a drive ? Or do they offer additional functionality, if so, what ? Thank you for your help.

MichaIng, 2019-05-20:
@echable9999 
Actually to mount network drives, indeed there are two install ways currently, once via `dietpi-software` (but this just installs the software, does not configure the mount) and once via `dietpi-drive_manager`, where the packages are installed automatically (outside of DietPi-Software) and the mount configured accordingly.

I agree this is a bid confusing and perhaps we should simply remove the client software from DietPi-Software list. Otherwise, possibly users want to have the client software but manually add the mounts in a different way than drive manager does 🤔. Not sure currently what is best here.
______________
Okay but about the NFS mounts, do you face any error messages within drive manager or when attempting to mount the share manually?

echable9999, 2019-05-20:
I really appreciate an important function such as mounting shares having
it's own separate interface.

I agree, you should remove the client software from the DietPi software
list, and have the Drive Manager as it's own section in config and/or
launcher.

I was expecting this to be a nightmare in Linux, with all the pitfalls for
a novice of case-sensitivity, "devices that are files" and whatever .... :)
But the way it's handled in DietPi is excellent.

I think what you mention, there being two ways to go about this, is somehow
how I must have gotten these errors. I did a fresh reinstall anyway, and it
worked without any problems this time.

Another problem I keep coming across is: packages that have their own
webinterface and port, I can install and use and find at their port (after
port forwarding in router). Packages that are supposed to just have some
letters (e.g. <Pi.IP.adress>/OMPD) never worked so far.

Regards
Eric

MichaIng, 2019-05-20:
@echable9999 
> Packages that are supposed to just have some
letters (e.g. <Pi.IP.adress>/OMPD) never worked so far.

These use a shared web server, if you didn't choose a different one, then it's Lighttpd.

To debug, could you paste the following:
```
systemctl status lighttpd
systemctl status php7.3-fpm
cat /var/log/lighttpd/error.log
cat /var/log/php7.3-fpm.log
```

Ah at best open a new issue for that 😉.

StephanStS, 2021-03-20:
Additionally, the display of the mounted directories in dietpi-drive_manager only show one single mount. If there are several mounts done and if you unmount ("Remove") nfs mounts the next mounted share will appear.

[image]

It also would be a good idea to show mountable shares (via `showmount -e <server_name>`) to be able to select the share to mount (reduces error possibilities).

MichaIng, 2021-03-23:
Good find. Yes basically it expects a single export without subdirectory from the server. I mean it does not even allow to add a path (unless it's an NFSv3 server where it's required), so it's not possible to add multiple exports from the same server.

Great idea with `showmount -e`:
- So basically we let the user select the export from the output.
- Then we allow to add additional path elements as subdirectory of the actual export, if this is possible? AFAIK I misunderstood the initial request in the forum, as it was not to mount subdirectories of an export but simply to add the export path itself so that the NFS mountpoint does not contain the whole path, like `/mnt/nfs_client/mnt/dietpi_userdata`, with the actual files being all in the last directory. So NFSv4 allows to mount a servers root and creates the path to the export inside of the mount point automatically, which is smart, but the result is not very handy.

StephanStS, 2021-03-23:
I played a bit with dietpi-drive_manager to investigate the behaviour.

First, I changed in Mount_NFS():  
- orig: `G_WHIP_INPUTBOX 'Please enter the NFS servers IP address (eg: 192.168.0.2).' || return`
- new: `G_WHIP_INPUTBOX 'Please enter the NFS servers hostname or IP address (eg: 192.168.0.2).' || return`
  to clearify that also a hostname is allowed

Then I added
- `local nfs_fp_server_sharedir='/'`
- and below the input of the NFS servers hostname or IP:

  ```
  G_WHIP_DEFAULT_ITEM=$nfs_fp_server_sharedir
  G_WHIP_INPUTBOX 'Please enter the NFS share path (e.g. /home)' || return
  nfs_fp_server_sharedir=$G_WHIP_RETURNED_VALUE

- and also further below: `if mount -vt nfs -o port=2049 "$nfs_server_ip":"$nfs_fp_server_sharedir" "$nfs_fp_mount_target" &>> $fp_tmp; then`

This gives a basic option to enter the share directory and mount a remote dir.  
If you then check with a wrong remote dir, further error mechanism take control which I did not investigate further on. This will be the harder work to get it robust.

This would be a basic idea, instead of using a browsing option via `showmount -e`.
```

MichaIng, 2021-03-23:
> If you then check with a wrong remote dir, further error mechanism take control which I did not investigate further on. This will be the harder work to get it robust.

Not that hard. An error loop could bring you back to the  IP/hostname input (which includes repeating the path, since each could be the reason for the connection failure). But `showmount -e` IMO is better anyway:
- Put IP/hostname input into a loop which contains the `showmount -e` selection menu.
- So when the `showmount -e` command itself fails or does not show any result, the IP/hostname can be corrected directly.

> And - what happened (magic): Adding the lines above showed two mounted NFS mounts in the dietpi-drive_manager main menu.

Yes that is expected. The menu shows each mount source only once and only the path element allows to differentiate between multiple exports from a single NFS server.

Joulinar, 2021-05-08:
AHH that was the one I was searching for. Just found the other one 🙂

MichaIng, 2022-03-17:
PR up: #5363

MichaIng, 2022-03-21:
Added for DietPi v8.3.

closed 2022-03-26 by MichaIng