MichaIng/DietPi #2523: "local: can only be used in a function" after update from v.6.20.6 to v.6.21.1

BerndKohl, 2019-02-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.20.6
- Distro version | stretch 9.6
- Kernel version | Linux 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | 5V 2A
- SDcard used | SanDisk ultra 16 GB

#### Additional Information (if applicable)
- Software title | dietpi-update
- Was the software title installed freshly or updated/migrated? updated
- Can this issue be replicated on a fresh installation of DietPi? don't know
- `dietpi-bugreport` ID | f21e0678-caf7-4b93-9475-08e8a85f9234

#### Steps to reproduce

1. run dietpi-update on version 6.20.6 and watch last script output before reboot (see below)
#### Extra details

- ...

# dietpi-update 
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.20.6
[ INFO ] DietPi-Update | Latest version  : v6.21.1
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=55679 MB | required=100 MB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : netdata
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying critical pre-patches

[  OK  ] Successfully applied critical pre-patches

[  OK  ] DietPi-Update | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:3 https://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:4 https://archive.raspberrypi.org/debian stretch/main armhf Packages [213 kB]
Fetched 11.9 MB in 15s (772 kB/s)
Reading package lists...

[  OK  ] DietPi-Update | G_AGUP
[  OK  ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  curl libcurl3 libcurl3-gnutls
3 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 741 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf curl armhf 7.52.1-5+deb9u9 [220 kB]
Get:2 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libcurl3 armhf 7.52.1-5+deb9u9 [262 kB]
Get:3 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libcurl3-gnutls armhf 7.52.1-5+deb9u9 [260 kB]
Fetched 741 kB in 0s (1,130 kB/s)
(Reading database ... 46888 files and directories currently installed.)
Preparing to unpack .../curl_7.52.1-5+deb9u9_armhf.deb ...
Unpacking curl (7.52.1-5+deb9u9) over (7.52.1-5+deb9u8) ...
Preparing to unpack .../libcurl3_7.52.1-5+deb9u9_armhf.deb ...
Unpacking libcurl3:armhf (7.52.1-5+deb9u9) over (7.52.1-5+deb9u8) ...
Preparing to unpack .../libcurl3-gnutls_7.52.1-5+deb9u9_armhf.deb ...
Unpacking libcurl3-gnutls:armhf (7.52.1-5+deb9u9) over (7.52.1-5+deb9u8) ...
Setting up libcurl3:armhf (7.52.1-5+deb9u9) ...
Setting up libcurl3-gnutls:armhf (7.52.1-5+deb9u9) ...
Processing triggers for libc-bin (2.24-11+deb9u3) ...
Setting up curl (7.52.1-5+deb9u9) ...

[  OK  ] DietPi-Update | G_AGUG
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   119    0   119    0     0    249      0 --:--:-- --:--:-- --:--:--   250
100 2494k  100 2494k    0     0  1143k      0  0:00:02  0:00:02 --:--:-- 2265k
[  OK  ] DietPi-Update | Unpack update archive
[  OK  ] DietPi-Update | Copy DietPi core files to RAMdisk
[  OK  ] DietPi-Update | Copy DietPi rootfs files in place
[  OK  ] DietPi-Update | Set execute permissions for DietPi scripts
[  OK  ] DietPi-Set_software | Connection test: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi.txt
[  OK  ] DietPi-Set_software | Patching dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Running incremental patching

[ INFO ] DietPi-Update | Current version : v6.20.6
[ INFO ] DietPi-Update | Latest version  : v6.21.1
[ INFO ] DietPi-Update | Patching 6.20.6 to 6.21

[  OK  ] DietPi-Patchfile | DietPi has applied a new G_DIETPI_INSTALL_STAGE system to the device.
[  OK  ] DietPi-Patchfile | Re-running DietPi-Update, to apply this new system...
[ SUB2 ] DietPi-Update > Checking for DietPi updates
[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.20.6
[ INFO ] DietPi-Update | Latest version  : v6.21.1
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=55670 MB | required=100 MB
[  OK  ] DietPi-Update | Update is being applied, please wait...
[ SUB3 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : netdata
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying critical pre-patches

[  OK  ] Successfully applied critical pre-patches

[  OK  ] DietPi-Update | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...

[  OK  ] DietPi-Update | G_AGUP
[  OK  ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-Update | G_AGUG
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   119    0   119    0     0    312      0 --:--:-- --:--:-- --:--:--   312
100 2494k  100 2494k    0     0  1236k      0  0:00:02  0:00:02 --:--:-- 3297k
[  OK  ] DietPi-Update | Unpack update archive
[  OK  ] DietPi-Update | Copy DietPi core files to RAMdisk
[  OK  ] DietPi-Update | Copy DietPi rootfs files in place
[  OK  ] DietPi-Update | Set execute permissions for DietPi scripts
[  OK  ] DietPi-Set_software | Connection test: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi.txt
[  OK  ] DietPi-Set_software | Patching dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed
[ SUB2 ] DietPi-Update > Running incremental patching
[ INFO ] DietPi-Update | Current version : v6.20.6
[ INFO ] DietPi-Update | Latest version  : v6.21.1
[ INFO ] DietPi-Update | Patching 6.20.6 to 6.21
[  OK  ] DietPi-Update | Patching to v6.21.1 completed

[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com
[ SUB2 ] DietPi-Update > Completed
[ INFO ] DietPi-Update | Current version : v6.21.1
[ INFO ] DietPi-Update | Latest version  : v6.21.1
/DietPi/dietpi/dietpi-update: line 528: local: can only be used in a function
[  OK  ] DietPi-Update | systemctl restart dietpi-ramdisk

BerndKohl, 2019-02-09:
Same for upgrade from v.6.19.7 to v.6.21.1

MichaIng, 2019-02-09:
@BerndKohl 
Thanks for your report and good spot!

Jep wrong coding there. However luckily visual only, since the result due to the syntax error is the same as it should be, which is why this passed testing as well.

Fixed with: https://github.com/Fourdee/DietPi/commit/96b5b6f8167dcd10d306eedbce8bf0a83ece7e68

Background:
- It is to control the reboot after the update.
- Declaring the variable as `local` fails, since this part of code is indeed not inside a function: https://github.com/Fourdee/DietPi/commit/96b5b6f8167dcd10d306eedbce8bf0a83ece7e68#diff-c613a85da508fb885b67c34f9661e243R528
- When update is on first run update or user chooses to reboot when asked, the variable will be set to value `1` and by this declared the correct way as well: https://github.com/Fourdee/DietPi/commit/96b5b6f8167dcd10d306eedbce8bf0a83ece7e68#diff-c613a85da508fb885b67c34f9661e243R536
- Afterwards based on the value, reboot is done or not. If one did not choose to reboot, then a second error should show up about an expected arithmetic value, since we check a non-existing variable.
- However the reboot is skipped correctly, since value `0` and error both result in the same "false": https://github.com/Fourdee/DietPi/commit/96b5b6f8167dcd10d306eedbce8bf0a83ece7e68#diff-c613a85da508fb885b67c34f9661e243R553
________
I mark this as closed.

closed 2019-02-09 by MichaIng

BerndKohl, 2019-02-09:
Thank you so much for a) the quick response and b) the very good explanation of what went wrong.
As always a pleasure.