MichaIng/DietPi #2524: Error during JAVA Installation

WTFMaster, 2019-02-09:
DietPi-Software: G_AGI: openjdk-8-jre-headless openjdk-8-jdk-headless
                                    │  - Exit code: 100
                                    │  - DietPi version: v6.21.1 (Fourdee/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4
                                    │  - Image creator: DietPi Core Team
                                    │  - Pre-image: Raspbian Lite
                                    │
                                    │ Log file contents:
                                    │ Running hooks in /etc/ca-certificates/update.d...^M
                                    │ done.^M
                                    │ Setting up libx11-data (2:1.6.4-3+deb9u1) ...^M
                                    │ Setting up libxau6:armhf (1:1.0.8-1) ...^M
                                    │ Setting up libavahi-common-data:armhf (0.6.32-2) ...^M
                                    │ Setting up fontconfig-config (2.11.0-6.7) ...^M
                                    │ Setting up libnss3:armhf (2:3.26.2-1.1+deb9u1) ...^M
                                    │ Setting up libavahi-common3:armhf (0.6.32-2) ...^M
[image]

MichaIng, 2019-02-09:
@WTFMaster 
Thanks for your report.

I will try to replicate. Meanwhile could you force the error again and scroll down to the last error log line please:
`G_AGI openjdk-8-jre-headless openjdk-8-jdk-headless`
Use the Page up/down keys do navigate through the log. Somehow the scroll bar is not shown here 🤔.

WTFMaster, 2019-02-10:
`apt-get install default-jre-headless`

`
Setting up ca-certificates-java (20170531+nmu1) ...
Error: missing `server' JVM at `/usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/server/libjvm.so'.
Please install or use the JRE or JDK that contains these missing components.
dpkg: error processing package ca-certificates-java (--configure):
 subprocess installed post-installation script returned error exit status 4
dpkg: dependency problems prevent configuration of openjdk-8-jre-headless:armhf:
 openjdk-8-jre-headless:armhf depends on ca-certificates-java; however:
  Package ca-certificates-java is not configured yet.

dpkg: error processing package openjdk-8-jre-headless:armhf (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of default-jre-headless:
 default-jre-headless depends on openjdk-8-jre-headless; however:
  Package openjdk-8-jre-headless:armhf is not configured yet.

dpkg: error processing package default-jre-headless (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.24-11+deb9u3) ...
Processing triggers for systemd (232-25+deb9u8) ...
Processing triggers for ca-certificates (20161130+nmu1+deb9u1) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...

Error: missing `server' JVM at `/usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/server/libjvm.so'.
Please install or use the JRE or JDK that contains these missing components.
E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.
done.
Errors were encountered while processing:
 ca-certificates-java
 openjdk-8-jre-headless:armhf
 default-jre-headless
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

MichaIng, 2019-02-10:
@WTFMaster 
First of all, use `apt-get install openjdk-8-jre-headless`, just to assure the desired Java version is installed. But okay on Stretch currently the result is the same.

Hmm, I remember this `ca-certificates-java` occurring already in the past...

I think it is somehow due to a dependency and install order loop.
- ca-certificates-java > openjdk-8-jre-headless: https://packages.debian.org/de/stretch/ca-certificates-java
- and openjdk-8-jre-headless > ca-certificates-java: https://packages.debian.org/de/stretch/openjdk-8-jre-headless

AFAIK the code in the past included both, try:
`apt-get -y install openjdk-8-jre-headless openjdk-8-jdk-headless ca-certificates-java`

WTFMaster, 2019-02-10:
`
Reading package lists... Done
Building dependency tree
Reading state information... Done
ca-certificates-java is already the newest version (20170531+nmu1).
ca-certificates-java set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:~#
`

MichaIng, 2019-02-10:
@WTFMaster 
Hmm installed already but not configured, argh.

So first try: `apt-get -f install`
Somewhere when installing openjdk the error message about non-configured package should appear. The following should fail:
```
dpkg --configure ca-certificates-java # Since it depends on JRE
dpkg --configure openjdk-8-jre-headless # Since it depends on ca-certificates-java..
```
If this is the case, damn loop. Try to restart from scratch:
```
apt purge ca-certificates-java openjdk-8-jre-headless openjdk-8-jdk-headless default-jre-headless
dpkg -P ca-certificates-java openjdk-8-jre-headless openjdk-8-jdk-headless default-jre-headless
apt install ca-certificates-java openjdk-8-jre-headless openjdk-8-jdk-headless
```

WTFMaster, 2019-02-10:
`
root@DietPi:~# apt-get -f install
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`
`
root@DietPi:~# dpkg --configure ca-certificates-java
dpkg: error processing package ca-certificates-java (--configure):
 package ca-certificates-java is already installed and configured
Errors were encountered while processing:
 ca-certificates-java
`
`
root@DietPi:~# dpkg --configure openjdk-8-jre-headless
dpkg: error processing package openjdk-8-jre-headless:armhf (--configure):
 package openjdk-8-jre-headless:armhf is already installed and configured
Errors were encountered while processing:
 openjdk-8-jre-headless:armhf
`

MichaIng, 2019-02-10:
@WTFMaster 
Ah so finally every is installed and configured? Okay then the `dietpi-software` install should work now.

Still we have to assure the APT install is done without running into this. But we can do this with some tests our side.

MichaIng, 2019-02-12:
Damn me, I removed the workaround for an indeed known issue: https://github.com/Fourdee/DietPi/issues/1340#issuecomment-389902031
https://github.com/Fourdee/DietPi/commit/6490e6dc5d945043a1d41dabc62546135ef1c612

Was not carefully enough reading the link and test passed on VM, but it is an ARM specific issue, now verified to still exist. Will re-add the workaround from before.

WilburWalsh, 2019-02-20:
Tested on my RPi 1B (armv6l) and it installed without problems.
This error is still printed during installation:

Processing triggers for ca-certificates (20161130+nmu1+deb9u1) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
Error: missing `server' JVM at `/usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/server/libjvm.so'.
Please install or use the JRE or JDK that contains these missing components.
E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.

But the installation process finshes fine, java works, and if I call `update-ca-certificates` manually, the missing JVM error does not occur.

MichaIng, 2019-02-20:
@WilburWalsh 
Thanks for testing. Jep the error is expected. We simply run a second APT install step to have it resolved then. Nasty thing that actually should be resolved Raspbian repo side, but at least the workaround works.

closed 2019-03-23 by MichaIng

Gill-Bates, 2019-03-25:
Solved. Awesome!