MichaIng/DietPi #2529: SSH Login Dietpi | Banner is displayed four times

jonare77, 2019-02-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6, G_DIETPI_VERSION_SUB=21, G_DIETPI_VERSION_RC=1, G_GITBRANCH=master, G_GITOWNER=Fourdee`
- Distro version | Distro version | `9.7`
- Kernel version | `uname -a`
- SBC device | Kernel version | `Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux`
SBC device | `Virtual Machine (x86_64)`

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated: Freshly
- Can this issue be replicated on a fresh installation of DietPi?: Not sure
- `dietpi-bugreport` ID | `9b1ac4a2-8c3b-4d4c-a6b4-fa83e8a54b2b`

#### Steps to reproduce
1. Login with SSH (user: dietpi)

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- Tried SecureCRT 8.5.2 and Putty 7.0

Fourdee, 2019-02-10:
@jonare77 

Many thanks for the report 👍 

Not something I've seen in local testing, however, i'll check your bug report and see what the cause is.

Fourdee, 2019-02-10:
@jonare77 

Found the cause, 4 entries for bashrc.d in ```/etc/bash.bashrc```:
```
for i in /etc/bashrc\.d/*.sh; do [ -r "$i" ] && . $i; done
for i in /etc/bashrc\.d/*.sh; do [ -r "$i" ] && . $i; done
for i in /etc/bashrc\.d/*.sh; do [ -r "$i" ] && . $i; done
for i in /etc/bashrc\.d/*.sh; do [ -r "$i" ] && . $i; done
```

## Solution/fix for multiple entries:
```
sed -i '\#for i in /etc/bashrc#d' /etc/bash.bashrc
G_CONFIG_INJECT '.*/etc/bashrc\.d/.*' 'for i in /etc/bashrc.d/*.sh; do [ -r "$i" ] && . $i; done' /etc/bash.bashrc
```

Fourdee, 2019-02-10:
Although unable to replicate in local testing, will patch clean + re-apply v6.22, no harm.

Fourdee, 2019-02-10:
Completed.

closed 2019-02-10 by Fourdee