MichaIng/DietPi #2530: General | Cant use any navigation commands like "dir"

WapitiAcht2777, 2019-02-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee
`
- Distro version | `stretch`
- Kernel version | `Linux DietPi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux`
- SBC device | `RPi 3 Model B (armv7l)`
- Power supply used | (5V 1A)
- SDcard used | (SanDisk 32GB)

#### Additional Information (if applicable)
- Issue can be replicated on a fresh installation of DietPi
- `46bf27a0-3a2b-46fc-97ad-2d79fcba7d7a` BUGREPORT ID

#### Steps to reproduce

1. Install fresh img of DietPi
2. Let it fully install
2. Type `dir` or `ls`

#### Expected behaviour

- Showing directory

#### Actual behaviour

- Shows `server_version-6`

#### Extra details

- Redownloaded img and checked SHA-256 and SHA1
[image]

MichaIng, 2019-02-10:
@WapitiAcht2777 
Might the reason be simply that the file `server_version-6` is the only non-hidden in /root directory?
Type `ls -Al` to show file details and hidden files in a list.

Not sure why the file is there. It is not required, only downloaded once on upgrade but removed directly. Only required on GitHub server to check for updates.
`rm /root/server_version-6` to get rid of it. I will check the PREP script, if this is accidentally downloaded and forgotten there.

WapitiAcht2777, 2019-02-10:
whoops i forgot about that sorry

closed 2019-02-10 by WapitiAcht2777