MichaIng/DietPi #2531: WiFi HotSpot not working on RPi B (armv6l)  -

marcobrianza, 2019-02-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.6

- Kernel version | `uname -a`
Linux DietPi 4.14.79+ #1159 Sun Nov 4 17:28:08 GMT 2018 armv6l GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi B (armv6l)

- Power supply used | (EG: 5V 1A RAVpower)
original Raspberry power supply

- SDcard used | (EG: SanDisk ultra)
Sandisk Extreme 16GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
WiFi Hotspot

- Was the software title installed freshly or updated/migrated?
freshly installed

- Can this issue be replicated on a fresh installation of DietPi?
yes.

- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

hostapd service does not start

running : hostapd -d /etc/hostapd/hostapd.conf
I get:
Illegal instruction

#### Expected behaviour
the service should run

#### Actual behaviour

- ...

#### Extra details
it look like the hostapd binary is for the armv7

MichaIng, 2019-02-11:
@marcobrianza 
Thanks for your report.

Indeed we replace the `hostapd` binary from Raspbian APT repo by an own compiled one that obviously has/had better WiFi device support. But we use ARMv6 binaries in case: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L5006-L5027
Just checked, ours is a newer version than provided by Debian/Raspbian Stretch (v2,5 vs v2.4): https://packages.debian.org/stretch/hostapd
However perhaps those need to be redone/rechecked on RPi 1. Also v2.7 is available now.

@Fourdee 
Do you have a chance to replicate on RPi 1 (B)?

dashingdon, 2019-03-22:
I have exact same issue. if need be , I can open another issue but here are my details 

Fresh dietpi install
`
#### Details:
- Date           | Thu 21 Mar 19:55:46 PDT 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi B (armv6l) (index=1)
- Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019
- Distro         | stretch (index=4)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config
#### Additional logs:
```
Log file contents:
Job for hostapd.service failed because a fatal signal was delivered to the control process.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

--------------------------------------------------------------------

[FAILED] DietPi-Config | Unable to continue, DietPi-Config will now terminate.

root@DietPi:/etc/hostapd# systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/hostapd.service.d
   Active: activating (auto-restart) (Result: signal) since Thu 2019-03-21 19:56:00 PDT; 5s ago
  Process: 2672 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=kill
ed, signal=ILL)
      CPU: 21ms

Mar 21 19:56:00 DietPi systemd[1]: hostapd.service: Control process exited, code=killed status=4
Mar 21 19:56:00 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Mar 21 19:56:00 DietPi systemd[1]: hostapd.service: Unit entered failed state.
Mar 21 19:56:00 DietPi systemd[1]: hostapd.service: Failed with result 'signal'.

`

MichaIng, 2019-03-22:
@dashingdon 
Did you try to use the binary from the Debian repo?
`apt install --reinstall hostapd`

dashingdon, 2019-03-22:
Yes. I did. No success.

MichaIng, 2019-03-22:
@dashingdon 
Okay, we need to to some research then. Either some kernel/system setting is missing or we simply need to update the hostapd binary to support a wider range of WiFi devices. Not sure currently if sort of AP drivers are included or such.

dashingdon, 2019-03-22:
Thanks @MichaIng Seems like something to do with Realtek RTL8188CUS driver. I found few articles where people have provided recompiled drivers. I tried but still not luck. 

Just for your reference on what all I tried

[jenssegers Version](https://docs.google.com/document/d/16hJS7fnPyKUwbvR33_nMku_b6XcYaBJXVpeCIMoJzcQ/pub)

[daveconroy version](https://daveconroy.com/how-to/turn-your-raspberry-pi-into-a-wifi-hotspot-with-edimax-nano-usb-ew-7811un-rtl8188cus-chipset/)

[PlantDaddy version](https://github.com/PlantDaddy/hostapd-rtl8188) - I have no idea how to use this

This has become a royal pain and I thought it would be a walk in the park :smile:  

Great job on dietpi otherwise. simply loving it. :sparkle:

MichaIng, 2019-03-22:
@dashingdon 
> PlantDaddy version - I have no idea how to use this

There are pre-compiled armhf binaries that you can simply place into your PATH:
```
wget https://github.com/PlantDaddy/hostapd-rtl8188/raw/master/hostapd/hostapd -O $(which hostapd)
wget https://github.com/PlantDaddy/hostapd-rtl8188/raw/master/hostapd/hostapd_cli -O $(which hostapd_cli)
```

dashingdon, 2019-03-22:
No Luck. Same result as [My original post](https://github.com/MichaIng/DietPi/issues/2531#issuecomment-475478826)

Happy to help test if needed.

whatsupskip, 2019-04-06:
I have hit the same problem with RPi Zero W using the built in WiFi adapter.

MichaIng, 2019-04-06:
@whatsupskip 
Does the original Debian package work in your case?
`apt install --reinstall hostapd`

whatsupskip, 2019-04-07:
@MichaIng  Yes, apt install --reinstall hostapd fixed it. Thank you!

I had previously tried uninstalling it and reinstalling it, but I think that changed some of the settings that shouldn't have been changed.

MichaIng, 2019-04-07:
@whatsupskip 
Thanks for testing and reporting back. DietPi-Software installs currently on RPi indeed an older version of `hostapd` which seems to not (yet) support the WiFi card.

Background:
- Debian Stretch shipped and still ships a quite outdated version of hostapd: https://packages.debian.org/stretch/hostapd
- On Debian Wheezy and Jessie this is was more drastic of course. For this reason Fourdee compiled own binaries on the that time current v2.5.
- Debian Stretch at least ships v2.4 already
- But Raspbian Stretch has meanwhile updated hostapd to v2.6: http://archive.raspberrypi.org/debian/dists/stretch/main/binary-armhf/Packages

So at least on Raspberry Pi's we need to revert to APT repo package.

MichaIng, 2019-04-07:
@dashingdon 
Btw. we also ship a fixed version for RTL8188C* chips that do not work with native hostapd: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L5010-L5015
And ours is with v2.5 newer than the ones you linked.

https://github.com/PlantDaddy/hostapd-rtl8188 => v2.4
https://docs.google.com/document/d/16hJS7fnPyKUwbvR33_nMku_b6XcYaBJXVpeCIMoJzcQ/pub => v2.0
https://daveconroy.com/how-to/turn-your-raspberry-pi-into-a-wifi-hotspot-with-edimax-nano-usb-ew-7811un-rtl8188cus-chipset/ => Download link not functional but since from 2013 it should be the oldest version.

Can you please run and paste output of: `hostapd -d /etc/hostapd/hostapd.conf`

dashingdon, 2019-04-07:
Output : 

root@DietPi:~# hostapd -d /etc/hostapd/hostapd.conf
Illegal instruction

Error Log : 

`
FAILED] DietPi-Config | systemctl start hostapd

--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Sat  6 Apr 20:51:36 PDT 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi B (armv6l) (index=1)
- Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019
- Distro         | stretch (index=4)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config

#### Additional logs:
```
Log file contents:
Job for hostapd.service failed because a fatal signal was delivered to the control process.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

--------------------------------------------------------------------

[FAILED] DietPi-Config | Unable to continue, DietPi-Config will now terminate.

`

Output of systemctl status hostapd.service

`
root@DietPi:~# systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: signal) since Sat 2019-04-06 20:55:50 PDT; 633ms ago
  Process: 2608 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=kill
ed, signal=ILL)

Apr 06 20:55:50 DietPi systemd[1]: hostapd.service: Control process exited, code=killed status=4
Apr 06 20:55:50 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP A
uthenticator.
Apr 06 20:55:50 DietPi systemd[1]: hostapd.service: Unit entered failed state.
Apr 06 20:55:50 DietPi systemd[1]: hostapd.service: Failed with result 'signal'.

`
Output of journalctl -xe

`
-- Logs begin at Thu 2016-11-03 10:16:44 PDT, end at Sat 2019-04-06 20:58:48 PDT. --
Apr 06 20:53:28 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
-- Subject: Unit hostapd.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit hostapd.service has finished shutting down.
Apr 06 20:53:28 DietPi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
-- Subject: Unit hostapd.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit hostapd.service has begun starting up.
Apr 06 20:53:28 DietPi systemd[1]: hostapd.service: Control process exited, code=killed status=4
Apr 06 20:53:28 DietPi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP A
uthenticator.
-- Subject: Unit hostapd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit hostapd.service has failed.
-- 
-- The result is failed.
Apr 06 20:53:28 DietPi systemd[1]: hostapd.service: Unit entered failed state.
Apr 06 20:53:28 DietPi systemd[1]: hostapd.service: Failed with result 'signal'.
Apr 06 20:53:30 DietPi systemd[1]: hostapd.service: Service hold-off time over, scheduling restart.
Apr 06 20:53:30 DietPi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.

`

dashingdon, 2019-04-07:
hostapd.conf 

> interface=wlan0
driver=rtl871xdrv
ssid=wifi
hw_mode=g
channel=12
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=YourPassPhrase
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

MichaIng, 2019-04-07:
@dashingdon 
Hmm, illegal instructions as well. Actually this error typically shows up when using a binary for wrong architecture. I just rechecked the binary set we ship and the armv6 binaries are larger then the armv7 are, while it's typically (but not always) the other way round. Since both yours devices are armv6 and those binaries would work on armv7 as well, and just to sort out that we mixed them up, could you please try with the following:
[hostapd.zip](https://github.com/MichaIng/DietPi/files/3051589/hostapd.zip)
```
wget https://github.com/MichaIng/DietPi/files/3051589/hostapd.zip
unzip hostapd.zip
rm hostapd.zip
cp -a $(command -v hostapd) hostapd_bak
mv hostapd $(command -v hostapd)
chmod +x $(command -v hostapd)
hostapd -d /etc/hostapd/hostapd.conf
```

dashingdon, 2019-04-09:
Create a temporary folder 

Illegal instruction

> root@DietPi:~/dietpi_hostapd# rm hostapd.zip
root@DietPi:~/dietpi_hostapd# cp -a $(command -v hostapd) hostapd_bak
root@DietPi:~/dietpi_hostapd# mv hostapd $(command -v hostapd)
root@DietPi:~/dietpi_hostapd# chmod +x $(command -v hostapd)
root@DietPi:~/dietpi_hostapd# hostapd -d /etc/hostapd/hostapd.conf
Illegal instruction
root@DietPi:~/dietpi_hostapd#

MichaIng, 2019-04-09:
@dashingdon 
Okay so now if this is ARMv7 the illegal instruction error is expected.
[hostapd.zip](https://github.com/MichaIng/DietPi/files/3061336/hostapd.zip)

So last try now just in case somehow during dietpi-software install the wrong binary was picked:
- This is now the ARMv6 rtl8188c binary.
```
wget https://github.com/MichaIng/DietPi/files/3061336/hostapd.zip
unzip hostapd.zip
rm hostapd.zip
cp -a $(command -v hostapd) hostapd_bak
mv hostapd $(command -v hostapd)
chmod +x $(command -v hostapd)
hostapd -d /etc/hostapd/hostapd.conf
```

Aside from that the only other thing we can try is to rebuild the patched hostapd binary with v2.6 according to: https://github.com/pritambaral/hostapd-rtl871xdrv#installation

dashingdon, 2019-04-11:
unfortunately no go
```
Illegal instruction
```
I am going to try and rebuild as per the instructions. Fingers crossed.

dashingdon, 2019-04-11:
Built hostapd from source. - http://w1.fi/hostapd/

Instructions for anyone who is interested in building hostapd on dietpi fresh install. Make sure to be at root prompt 

 Patch instructions from [here](https://github.com/pritambaral/hostapd-rtl871xdrv#installation)

```
sudo apt-get install build-essential pkg-config libnl-3-200 libnl-utils libnl-3-dev libssl-dev libnl-genl-3-dev git
export PKG_CONFIG_PATH=/usr/include/libnl3

// Download driver
git clone https://github.com/pritambaral/hostapd-rtl871xdrv.git

wget http://w1.fi/releases/hostapd-2.7.tar.gz
tar -xf hostapd-2.7.tar.gz
cd /hostapd-2.7/hostapd
cp defconfig .config

// Apply patch
patch -Np1 -i /root/hostapd-rtl871xdrv/rtlxdrv.patch
echo CONFIG_DRIVER_RTW=y >> .config

// Alternatively open .config in nano and make sure that section looks like below
// # Driver interface for Host AP driver
// #CONFIG_DRIVER_HOSTAP=y
// CONFIG_DRIVER_RTW=y

make
make install 
```

if make errors out with 
```
../src/utils/wpabuf.o: file not recognized: File truncated. 
```
Navigate to  /hostapd-2.7./src/utils/ Delete 0 byte file and rerun make

dashingdon, 2019-04-11:
Output of 
```
```

```
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
drv->ifindex=3
l2_sock_recv==l2_sock_xmit=0x0x1c608a0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Completing interface initialization
Mode: IEEE 802.11g  Channel: 3  Frequency: 2422 MHz
DFS 0 channels required radar detection
RATE[0] rate=10 flags=0x1
RATE[1] rate=20 flags=0x1
RATE[2] rate=55 flags=0x1
RATE[3] rate=110 flags=0x1
RATE[4] rate=60 flags=0x0
RATE[5] rate=90 flags=0x0
RATE[6] rate=120 flags=0x0
RATE[7] rate=180 flags=0x0
RATE[8] rate=240 flags=0x0
RATE[9] rate=360 flags=0x0
RATE[10] rate=480 flags=0x0
RATE[11] rate=540 flags=0x0
hostapd_setup_bss(hapd=0x1c60f10 (wlan0), first=1)
wlan0: Flushing old station entries
wlan0: Could not connect to kernel driver
wlan0: Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=0)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=1)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=2)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=3)
Using interface wlan0 with hwaddr 80:1f:02:b5:ea:27 and ssid "DietPi-HotSpot"
Deriving WPA PSK based on passphrase
SSID - hexdump_ascii(len=14):
     44 69 65 74 50 69 2d 48 6f 74 53 70 6f 74         DietPi-HotSpot  
PSK (ASCII passphrase) - hexdump_ascii(len=13): [REMOVED]
PSK (from passphrase) - hexdump(len=32): [REMOVED]
random: Got 20/20 bytes from /dev/random
GMK - hexdump(len=32): [REMOVED]
Key Counter - hexdump(len=32): [REMOVED]
WPA: Delay group state machine start until Beacon frames have been configured
rtl871x_set_beacon_ops
rtl871x_set_hidden_ssid_ops
Failed to set beacon parameters
wlan0: Flushing old station entries
wlan0: Could not connect to kernel driver
wlan0: Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=3
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=0)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=1)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=2)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=3)
hostapd_free_hapd_data(wlan0)
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
hostapd_interface_deinit_free(0x1c602e0)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x1c602e0)
wlan0: interface state DISABLED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: Flushing old station entries
wlan0: Could not connect to kernel driver
wlan0: Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=3
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=0)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=1)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=2)
rtl871x_set_key_ops
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=3)
wlan0: AP-DISABLED 
hostapd_cleanup(hapd=0x1c60f10 (wlan0))
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=0x95be0 drv_priv=0x1c60810 -> hapd_deinit
hostapd_interface_free(0x1c602e0)
hostapd_interface_free: free hapd 0x1c60f10
hostapd_cleanup_iface(0x1c602e0)
hostapd_cleanup_iface_partial(0x1c602e0)
hostapd_cleanup_iface: free iface=0x1c602e0
ELOOP: remaining socket: sock=6 eloop_data=0x1c60948 user_data=(nil) handler=0x6964c
```

MichaIng, 2019-04-17:
@marcobrianza 
Note that as mentioned the patch only supports hostapd v2.6 currently: http://w1.fi/releases/hostapd-2.7.tar.gz

dashingdon, 2019-04-20:
Thanks @MichaIng for the help. Appreciate you following up.

I rebuilt for version 2.6. Error message is persistent. 
```
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
```
Hostapd version Info
```
root@DietPi:~# hostapd -v
hostapd v2.6 for Realtek rtl871xdrv
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2016, Jouni Malinen <j@w1.fi> and contributors
```

Not sure what else to do ...

MichaIng, 2019-04-22:
> Unpacking libssl1.0.0:armhf (1.0.1t-1+deb8u7) over (1.0.1t-1+deb8u7) ...

Okay was already installed in your case.

Hmm I am also out of ideas for now. We tried so many binaries now and ruled out the possible libssl issue and `rtl871xdrv` driver. A pain with this rtl8188c chips that official AP drivers for those had never been really finished 😞.

So I guess you have to test another WiFi dongle, at best one that is supported by the default hostapd.
___________
Little seeking for the error message: https://groups.google.com/forum/#!topic/linux-sunxi/Jrd_JedPlDs
- Possibly related to kernel <> driver combination?
Also from the patch repo itself: https://github.com/pritambaral/hostapd-rtl871xdrv/issues/1
- Very old, no helpful answer.
- You could try to open an issue on the repo with the exact WiFi dongle and steps you did to build the binary (source, patch commands etc).

dashingdon, 2019-04-22:
Thanks @MichaIng . 

I read somewhere that overclocking solved the issue of hostapd not starting. So I overclocked the pi and to **my surprise**, patched hostapd 2.6 version service started. Still no AP. Now that atleast the service is running , any other thoughts would be much appreciated. 

Just for the record , I built the new kernel as per [link](https://www.raspberrypi.org/documentation/linux/kernel/building.md) Took more than 30hrs and unfortunately didn't help

```
root@DietPi:~# systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-04-21 17:35:05 PDT; 5min ago
  Process: 614 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, stat
us=0/SUCCESS)
 Main PID: 626 (hostapd)
   CGroup: /system.slice/hostapd.service
           └─626 /usr/sbin/hostapd -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf

Apr 21 17:35:04 DietPi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator..
.
Apr 21 17:35:04 DietPi hostapd[614]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Apr 21 17:35:04 DietPi hostapd[614]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Apr 21 17:35:05 DietPi hostapd[614]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Apr 21 17:35:05 DietPi hostapd[614]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Apr 21 17:35:05 DietPi hostapd[614]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Apr 21 17:35:05 DietPi hostapd[614]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Apr 21 17:35:05 DietPi systemd[1]: hostapd.service: PID file /run/hostapd.pid not readable (yet?) after start: N
o such file or directory
Apr 21 17:35:05 DietPi systemd[1]: Started Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.

```
Service isc-dhcp-server
```
root@DietPi:~# systemctl status isc-dhcp-server
● isc-dhcp-server.service - LSB: DHCP server
   Loaded: loaded (/etc/init.d/isc-dhcp-server; generated; vendor preset: enabled)
   Active: active (running) since Sun 2019-04-21 17:35:07 PDT; 33min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 608 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/isc-dhcp-server.service
           └─645 /usr/sbin/dhcpd -4 -q -cf /etc/dhcp/dhcpd.conf wlan0

Apr 21 17:35:04 DietPi systemd[1]: Starting LSB: DHCP server...
Apr 21 17:35:05 DietPi isc-dhcp-server[608]: DHCPv4 interfaces are no longer set by the INTERFACES variable in
Apr 21 17:35:05 DietPi isc-dhcp-server[608]: /etc/default/isc-dhcp-server.  Please use INTERFACESv4 instead.
Apr 21 17:35:05 DietPi isc-dhcp-server[608]: Migrating automatically for now, but this will go away in the futur
e.
Apr 21 17:35:05 DietPi isc-dhcp-server[608]: Launching IPv4 server only.
Apr 21 17:35:05 DietPi dhcpd[634]: Wrote 0 leases to leases file.
Apr 21 17:35:05 DietPi dhcpd[645]: Server starting service.
Apr 21 17:35:07 DietPi isc-dhcp-server[608]: Starting ISC DHCPv4 server: dhcpd.
Apr 21 17:35:07 DietPi systemd[1]: Started LSB: DHCP server.
```
Output of hostapd /etc/hostapd/hostapd.conf
```
root@DietPi:~# hostapd  /etc/hostapd/hostapd.conf
Configuration file: /etc/hostapd/hostapd.conf
drv->ifindex=3
l2_sock_recv==l2_sock_xmit=0x0x1a57cd8
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
wlan0: Could not connect to kernel driver
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Using interface wlan0 with hwaddr 80:1f:02:b5:ea:27 and ssid "wifi"
rtl871x_set_beacon_ops
rtl871x_set_hidden_ssid_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set beacon parameters
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
wlan0: Could not connect to kernel driver
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=3
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
wlan0: Could not connect to kernel driver
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=3
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
ELOOP: remaining socket: sock=6 eloop_data=0x1a57d80 user_data=(nil) handler=0x52748
```

dashingdon, 2019-04-22:
I am going to try [this](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=154413) tomorrow. I will update this thread with results.

MichaIng, 2019-04-22:
@dashingdon 
Very interesting. According to this at least for this `RTL8188CUS` chip the default hostapd package with default `driver=nl80211` works? Strange that this is not reported anywhere else...

Yeah please try this out. Also try to override with default hostapd package then: `apt install --reinstall hostapd` and `driver=nl80211` (the first you tried already, but not sure if the driver setting was correct that time?)

Confusing as this was reported in Juli 2016 and last work on the patch is done October 2016.

dashingdon, 2019-04-25:
@MichaIng Just a quick update. I am continuing to try different things. Here is what I am able to confirm till now. 
- [This](https://github.com/MichaIng/DietPi/issues/2531#issuecomment-485337904) did not work on dietpi.
- Fresh Raspbian + [This](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=154413) + [Bridge Mode](https://daveconroy.com/how-to/turn-your-raspberry-pi-into-a-wifi-hotspot-with-edimax-nano-usb-ew-7811un-rtl8188cus-chipset/) excluding the hostapd version provided in the article : **AP Works** 
- Fresh Raspbian +  [This](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=154413) + [Non Bridge Mode](https://daveconroy.com/how-to/using-your-raspberry-pi-as-a-wireless-router-and-web-server/) **DID NOT WORK**  _I only used relevant configuration parts of the article_

Please note that Hostapd service always starts in **raspbian**. dhcpcd5 / isc-dhcp-server is the one which  errors out. While hostapd service always fails in dietpi but isc-dhcp-server service runs fine. **I used debian version of hostapd.with driver=nl80211**

I am still trying to figure **why** part of the above behavior. I will re-try on the fresh dietpi install in next couple of days.

MichaIng, 2019-04-27:
@dashingdon 
Thanks for investigating.

So with "This" you mean running the Raspbian APT repo hostapd binary (`apt install --reinstall hostapd`) with `driver=nl80211` works well on Raspbian but not on DietPi?
This is actually very unlikely since DietPi on RPi is Raspbian, so if the binary works on the one system it must run on the other as well.
> I used debian version of hostapd.with driver=nl80211

You mean the "Raspbian" version, right?

So I guess you tried with different configurations (e.g. `hostapd.conf`) on both systems and one works with the binary while one doesn't. This is important since I want to use the Raspbian APT hostapd which obviously works, regardless of RTL8188C* chip. But it is important then to know what we need to change about our configuration in case.

To make tests as clean as possible and get the hostapd binary running first:
`hostapd -d /etc/hostapd/hostapd.conf`
- In case compare `hostapd.conf` about differences that might have an effect and `/etc/network/interfaces` about the `wlan0` interface that hostapd relies on.

When finally the hostapd binary works, then the `hostapd.service` most likely does as well. Next is then to have a look at the `isc-dhcp-server`, if it fails. If it fails on Raspbian, then there must be something wrong about the configuration. Could be compared to how we do on DietPi, as well in comparison to `/etc/network/interfaces`. `dhcpcd5` is not what we want or use, as it requires different `/etc/network/interfaces` and conflicts with the `isc-dhcp-client` we use, which is better integrated into `ifupdown` per-interface start/stop hooks. Also `dhcpcd5` and `isc-dhcp-client` break each other, AFAIK.

The services above should run independently of any redirection/iptables/bridging. So finally WiFi client connections are the last step. Bridging should be not required, but the iptables accept/masquerade rules and sysctl forwarding rules. Note that the guide you followed for Raspbian does not contain IPv6 forwarding nor ip6tables rules. Depending on the DNS nameserver this could break connections when the client tries to access via IPv6.

MichaIng, 2019-04-30:
@Fourdee 
Do you have a RTL8188 dongle around to test with newest `hostapd` APT packages, especially on RPi (Raspbian) where they ship a newer version, possibly with custom fixes.

Possibly related: https://github.com/MichaIng/DietPi/issues/1955

Since there are already older reports that on Raspbian it works with APT repo binary, I am thinking to skip our custom binaries on RPi: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=152603#p1010357

bowiz2, 2020-03-22:
@MichaIng 

> So with "This" you mean running the Raspbian APT repo hostapd binary (apt install --reinstall hostapd) with driver=nl80211 works well on Raspbian but not on DietPi?

On the most recent version of dietpi (flashed today), and with no other software. Raspberrpi Pi 1B with a Edimax EW-7811Un dongle, which [should be ](https://deviwiki.com/wiki/Edimax_EW-7811Un) using RTL8188CUS.

At first, got the same "Invalid Instruction" and "Could not find driver" errors. Did those two steps, and hostapd runs without issue.
Still can't vouch for long term stability, but I'll check back in a few days and let you know if anything goes wrong.

If there's anything you want me to check/verify, just let me know!

MichaIng, 2020-03-22:
I exclude RPi now for our self-compiled RTL binaries: https://github.com/MichaIng/DietPi/commit/c63fa4f48c5cf4a8963444f35ae0d36da44e7955
Enough reports that the RPi repo drivers work with RTL8188C* adapters already.

sphh, 2020-06-28:
I had the same or similar problem on my RPi1 during a fresh install on a RPi 1 with an Edimax EW-7811UN USB dongle: I enabled the WiFi Hotspot, everything was downloaded, installed and configured automatically. But the `hostapd` service failed to start. Luckily I was just watching that DietPi added the line `driver=rtl871xdrv` to the `/etc/hostapd/hostapd.conf` file. After commenting out this line (because I knew from Raspbian, that this does not work), and everything works.

This might or might not be related to the above problem ...

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`:
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`:
`buster` or `10.4`
- Kernel version | `uname -a`:
`Linux DietPi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3): `1`

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi?: YES

MichaIng, 2020-06-28:
Interesting actually, since AFAIK on Raspbian that entry should not be added anymore 🤔.

... checked, ah nope actually we do not install our special Realtek binaries on RPi anymore but still apply the setting.

@sphh 
Could you please run the following and paste the output?
```
lsusb
G_CONFIG_INJECT 'driver=' 'driver=rtl871xdrv' /etc/hostapd/hostapd.conf
hostapd
```
To revert to functional config:
```
G_CONFIG_INJECT 'driver=' 'driver=nl80211' /etc/hostapd/hostapd.conf
```

sphh, 2020-06-28:
`lsusb`:
```
Bus 001 Device 006: ID 4852:0006  
Bus 001 Device 005: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 008: ID 12d1:14db Huawei Technologies Co., Ltd. E353/E3131
Bus 001 Device 004: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`G_CONFIG_INJECT 'driver=' 'driver=rtl871xdrv' /etc/hostapd/hostapd.conf`:
```
[  OK  ] G_CONFIG_INJECT | Comment in /etc/hostapd/hostapd.conf converted to setting: driver=rtl871xdrv
```

`hostapd /etc/hostapd/hostapd.conf`:
```
Configuration file: /etc/hostapd/hostapd.conf
Line 2: invalid/unknown driver 'rtl871xdrv'
1 errors found in configuration file '/etc/hostapd/hostapd.conf'
Failed to set up interface with /etc/hostapd/hostapd.conf
Failed to initialize interface
```

MichaIng, 2020-06-28:
Okay great, many thanks:
> Line 2: invalid/unknown driver 'rtl871xdrv'

That clarifies that the RPi binary did not add the Realtek driver but either patched the nl80211 driver to support those RTL boards or something like that. I'll prevent the driver switch on RPi, lets hope this does not cause opposite reports as if deeper differentiation would be required 😅.

MichaIng, 2020-08-27:
I mark this as solved. Compared to the Debian hostapd binary, the one from the RPi foundation indeed seems to support (most of) those, especially since RPi Zero W has an RTL8188 WiFi chip as well.

closed 2020-08-27 by MichaIng

Joulinar, 2020-11-27:
@MichaIng 
can you have a look to this forum post. It's fitting to this issue https://dietpi.com/phpbb/viewtopic.php?f=11&t=8318

MichaIng, 2020-11-28:
Indeed, the driver is applied exactly the opposite way as intended. Should be fixed now (see forum thread).