MichaIng/DietPi #2532: NanoPi Neo 2 i2c OLED Display

Velociraptor85, 2019-02-10:
### Creating a feature request
I would like to have the ability to use a NanoHat OLED Display for my NanoPi Neo2. Also I think that this board is used alot so maybe other people would probably also benefit ^^

#### Is your feature request related to a problem? Please describe:

- Well no and yes =) If I go to Display Options->LCD Panel addon option in the DietPi-Config  it sais This option is not available for NanoPi NEO2

#### Describe the solution you'd like:

- That the previous mentioned Option enables the OLED Hat.

#### Describe alternatives you've considered:

- Could also be as default in the Setup or as an Software Package in the DietPi-Software. 

#### Additional context

Here some links to libraries and documentation
- http://wiki.friendlyarm.com/wiki/index.php/NanoHat_OLED
- https://github.com/friendlyarm/NanoHatOLED
- https://github.com/squidpickles/nanohat-oled
- https://github.com/creekpld/nanohat-oled
- [alternative HAT](https://www.aliexpress.com/item/NanoPi-NEO-NEO2-all-metal-aluminum-shell-with-NanoHat-OLED-display-heat-sink/32953738364.html?spm=a2g0s.12269583.0.0.71e15e02hBGLJM) based on the same OLED driver (SSD1306)

MichaIng, 2019-02-11:
@Velociraptor85 
Thanks for your request.

I add this as software request since the install script/pre-req match many of what we install via `dietpi-software` as well, too much to do this inside the `dietpi-config` script.
We can add it to `dietpi-config`, which then redirects you to dietpi-software or even triggers an automated install via: `dietpi-software install <ID>`

The I²C module/overlay is enabled by default on NanoPi?

Velociraptor85, 2019-02-12:
I will check on the weekend.
What should I do to check if the I²C module/overlay is enabled?

Also Thanks for thinking of implementing it ^^

MichaIng, 2019-02-12:
@Velociraptor85 
> What should I do to check if the I²C module/overlay is enabled?

`lsmod` should contain something related.

Velociraptor85, 2019-02-16:
@MichaIng so `lsmod` shows following:

> root@DietPi:~# lsmod
Module                  Size  Used by
ir_lirc_codec          16384  0
lirc_dev               16384  1 ir_lirc_codec
snd_soc_simple_card    16384  0
snd_soc_simple_card_utils    16384  1 snd_soc_simple_card
sunxi_cir              16384  0
rc_core                32768  4 ir_lirc_codec,lirc_dev,sunxi_cir
rtc_ds1307             24576  0
g_mass_storage         16384  0
usb_f_mass_storage     36864  1 g_mass_storage
libcomposite           45056  2 g_mass_storage,usb_f_mass_storage

meaning I cant see anything related to I²C

MichaIng, 2019-02-16:
Okay, so if not it is hardcoded into the kernel, it seems we also need to enable the overlay/module first.

Ah the `install.sh` contains this. It pulls the `BakeBit` installer which contains I²C setup: https://github.com/friendlyarm/BakeBit/blob/master/Script/install.sh

Hmm using https://github.com/friendlyarm/NanoHatOLED/blob/master/install.sh on the one hand is easy and makes sense. But it is a bid ugly using `rc.local` 🤔 and this even a way that might not work due to it's `exit 0`. So we need an own systemd unit or `postboot.d` entry for this.
Also the automated reboot in BakeBit's install.sh is something we need to work around.

Usage libraries:
- https://github.com/creekpld/nanohat-oled
  - What is the Swift package manager? 🤔
- https://github.com/squidpickles/nanohat-oled
  - Sorry I am a bid dump about this, but is there some install required or does one simply use the provided library and font files as is?

Velociraptor85, 2019-02-17:
Hi,
so regarding the libraries, these where just some references which I found in the interwebs by you common search engine.

Regarding the install.sh I could test that from NanoHatOLED.
As far as I understand it:
* it compiles the the daemon and main into `NanoHatOLED` executable 
* creates this `/usr/local/bin/oled-start` file which is started than via the mentioned `rc.local` 
* and for the driver support it starts the `BakeBit\install.sh`

For me this would mean that the actual i²C support is the BakeBit part.

Additionally squidpickles uses a device path `/dev/i2c-0` which is present on my current DietPi setup.
Not sure though if it would work. Seems like he uses RUST which might need to be installed for this to work.

`dmesg | grep i2c` showed 
> [    1.619783] i2c /dev entries driver

bdleung, 2019-05-19:
The v2 script posted by Phillski at https://dietpi.com/phpbb/viewtopic.php?t=3077 is also useful to prevent OLED burn in (which unfortunately happened to mine... :( )

Velociraptor85, 2019-05-27:
@bdleung thanks, the instructions worked like a charm

MichaIng, 2019-05-27:
@bdleung 
Many thanks for linking the forum topic, which matches/verifies the instructions above. Anti screen burn is brilliant, totally forgot about this.

bdleung, 2019-05-27:
@Velociraptor85 @MichaIng Not a problem, glad I could help. Can verify the OLED panel supplied in the NanoPi kits are very susceptible to burn in. I found out the hard way after initially not using the script...

Velociraptor85, 2019-08-18:
The instructions in from the forum did not work for dietpi 6.25.3 ... i had to skip step 3 
`sudo pip install --upgrade ssd1306`
everything else worked