MichaIng/DietPi #2534: DietPi-Software | pyLoad: One-click hoster downloader

MichaIng, 2019-02-10:
### Creating a software request

#### Formal software information
- Software title | pyLoad
- Short description | Downloader for one-clock hoster with web UI
- Official URL (if available) | https://pyload.net/
- GitHub URL (if available) | https://github.com/pyload/pyload
- (Official) install documentation | https://github.com/pyload/pyload/wiki/Step-by-Step-Installation-(RaspberryPi)

#### Are there similar/alternative software titles available with DietPi-Software?

- None, at least not for on-click hosters

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Automate Python + modules installs, add systemd unit, configure web UI with proper port (not default 8000)

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
- https://github.com/pyload/pyload/wiki/Step-by-Step-Installation-(RaspberryPi)

Velociraptor85, 2019-09-29:
https://github.com/pyload/pyload/issues/3402

lutfor-diu, 2023-10-16:
please add this. very useful software.