MichaIng/DietPi #2535: PINEA64+ - dt-overlay not working

lwqcz, 2019-02-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.21.1/master/Fourdee
- Distro version | Stretch
- Kernel version | 3.10.105-bsp-1.2-ayufan-77
- SBC device | Pine A64 (aarch64)
- Power supply used | 5V/5A DCDC PSU
- SDcard used | 8GB SanDisk Ultra

#### The issue:
- I have had spend a few days looking for solution how to enable SPI, I2C and 1wire on PineA64 platform. But no luck. I have read that this kind of support was quite a problem when this SBC was new. But this problems are three years old (or so) so I hope that there were some changes.
And I want to ask, if you can add support for these features and also update whole system, as it runs on kernel 3.10.105-bsp-1.2-ayufan-77 from July 2017.

MichaIng, 2019-02-10:
@lwqcz 
How did you try to enable it? `/boot/uEnv.txt` should be where to add the `overlays=spi-spidev` etc lines.

I just checked the most current image and indeed we switched from Ayufan to ARMbian base where the very current 4.19 kernel is pre-installed: https://dietpi.com/downloads/images/

Sadly this cannot be patched/updated on the running system, you requires to flash. The boot/kernel config file is changed there to: `armbianEnv.txt`

lwqcz, 2019-02-11:
Hi @MichaIng !

I have found that these settings probably should be in the /boot/config.txt file. I did noticed /boot/uEnv.txt but as far as I did not find any clue how to use it, I did not.

I have the image downloaded just about two weeks ago, so is it obsolete now? Nice catch :-) I will upgrade tonight.

Can you provide some URL where usage of /boot/armbianEnv.txt is documented? Or one can just assume configuration based on RPi docs?

Thank you.

MichaIng, 2019-02-11:
@Iwqcz
You image is not obsolete and will work fine + software updates. Only the kernel is an older one :wink:.

Raspberry Pi uns Pine64 boards use a different bootloader which reads different config files. RPi uses config.txt while Pine64 boards use uBoot bootloader and read uEnv.txt, respectively armbianEnv.txt on ARMbian based images.
Also the overlay directives are different. Check Pine64 or ARMbian instructions in how to configure them.

I agree we should add those settings to dietpi-config similar how we do on RPi. Perhaps we can find out together the required settings to enable I2C and SPI as a start.

MichaIng, 2019-02-11:
For SPI it seems:
overlays=spi-spidev and
param_spidev_spi_bus=0 or 1

Source: https://forum.armbian.com/topic/4932-recipe-for-enabling-spi-on-pine64/

lwqcz, 2019-02-17:
Hi @MichaIng - thanks for your messages.

I've upgraded system to the latest available on DietPi website.
As I have no SPI device to check if it is working, I am mostly focused to get working my I2C Melexis IR thermometer and a quite large number of DS18B20 1wire sensors.
I have had exactly this setup up & running for nearly two years on RPi model B+, but its computing power was became insufficient to process everything including RTL SW radio using dump1090.
Therefore I am quite sure that I have no hardware issue ...

I have checked all available Allwinner's & PINE64 documentation but I have no luck yet and I also have not another clue what to try 😞 

- content of /boot/armbianEnv.txt:
```
overlays=w1-gpio i2c1 spi-spidev
param_w1_pin=PL10 #I have tried: PA20 PD4 PL10 but no luck
param_spidev_spi_bus=0
```

Any suggestions please? 😉

lwqcz, 2019-02-24:
Hi there,

with help of my friend who is also using Pine64 and DS18B20 1wire sensors, I discovered that the "correct" pin would be PC18 instead of PL10. But IMHO the 1wire bus can be set up nearly on every free GPIO, am I correct?

My current content of `/boot/armbianEnv.txt`:
```
overlays=w1-gpio i2c1 spi-spidev
param_w1_pin=PC8 #PC8 PD4 PL10
param_spidev_spi_bus=0
```

Is anyone sure that kernel (and/or DTD overlays) is compiled "correctly"? Command of `dmesg` is not showing anything unusual AFAIK. Thanks!

lwqcz, 2019-03-28:
Hi,
I just want to inform there that even upgrade to newest DietPi v6.22.3 give no luck for this issue. Sadly. :-(

MichaIng, 2019-03-28:
@lwqcz 
#### General doc: https://docs.armbian.com/User-Guide_Allwinner_overlays/#kernel-provided-vs-user-provided-overlays
#### Board specific overlay instructions can be found at: `/boot/dtb/allwinner/overlay/README.sun50i-a64-overlays`
- 1-Wire GPIO default pin is `PD4`, which can be switches as above but I would keep it as long as there is no specific reason to change.
   - Note: `Requires external pull-up resistor on data pin`
or:
       ```
       param_w1_pin_int_pullup (bool)
            	Enable internal pull-up for the data pin
           	Optional
           	Default: 0
            	Set to 1 to enable the pull-up
            	This option should not be used with multiple devices, parasite power setup
     		or long wires -	please use external pull-up resistor instead
       ```
- `I2C1 pins: (SCL, SDA): PH2, PH3`
- `spi-spidev` overlay requires `param_spidev_spi_bus` but I see you did that already. So pins are:
`SPI 0 pins (MOSI, MISO, SCK, CS): PC0, PC1, PC2, PC3`

This should be all info you need.
Does I2C and SPI work as expected?
And about 1-Wire do you use either an external pull-up resistor or otherwise tried to enable the internal one (although seems not recommended)? Then as said I would skip the parameter to revert to default `PD4`.

And generally: Enable only the overlays that you currently really need, for debugging at best only the single that that you want to get working (+ required parameters in case of course). Then add other required overlays one by one, respecting the readme above about possible pin double uses. But the default 1-Wire + I2C + SPI pins do not conflict.

I am absolutely not expert about GPIO usage in general but I hope the above info collection enables you to get things running.

MichaIng, 2019-04-06:
@lwqcz 
I think I found what is missing to load overlays. Please try `G_AGI device-tree-compiler` and reboot.

lwqcz, 2019-06-28:
@MichaIng where should I put `G_AGI device-tree-compiler` you've mentioned?

MichaIng, 2019-06-28:
@lwqcz 
Run this in terminal. It's our error handled+noninteractive wrapper for `apt install device-tree-compiler` which so thus can run as well instead.

MichaIng, 2020-06-01:
Solved with new image. To add SPI or I2C:
1. Edit `/boot/armbianEnv.txt`
2. Create or alter the `overlays=` line and add e.g. `i2c0` or `spi-spidev` to the list of overlays, separated by white space.
3. Check `/boot/dtb/allwinner/overlay/README.sun50i-a64-overlays` for available overlays and details.

closed 2020-06-01 by MichaIng

lwqcz, 2022-03-09:
@MichaIng sorry for my late reply, but you were correct, according [this GPIO pinout](http://files.pine64.org/doc/Pine%20A64%20Schematic/Pine%20A64%20Pin%20Assignment%20160119.pdf) is easy to locate default PIN of `PD4` (it is on EULER header) and make it running (I was lately concern about 1wire bus):
1. Add `overlays=w1-gpio` (or I2C or SPI)
2. Reboot
3. The result for tow sensors:
```
root@Central:/home/dietpi# cat /sys/bus/w1/devices/28-05*/w1_slave 
3f 01 4b 46 7f ff 0c 10 f3 : crc=f3 YES
3f 01 4b 46 7f ff 0c 10 f3 t=19937
3c 01 4b 46 7f ff 0c 10 36 : crc=36 YES
3c 01 4b 46 7f ff 0c 10 36 t=19750
```