MichaIng/DietPi #2536: DietPi Software | Mopidy fresh installation playback but no sound

arkhub, 2019-02-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2019-02-10:
@arkhub 
Thanks for your report.

We will investigate.

Fourdee, 2019-02-11:
Confirmed bug, no ALSA stream active, will investigate:
```
Feb 11 18:09:42 DietPi mopidy[1045]: WARNING  Failed looking up file:///mnt/diet
pi_userdata/Music/%5BRadio%20Streams%5D.m3u8: gst-stream-error-quark: Stream con
tains no data. (4)
Feb 11 18:09:43 DietPi mopidy[1045]: WARNING  Failed looking up file:///mnt/diet
pi_userdata/Music/%5BRadio%20Streams%5D.m3u8: gst-stream-error-quark: Stream con
tains no data. (4)
Feb 11 18:09:44 DietPi mopidy[1045]: WARNING  GStreamer warning: gst-resource-er
ror-quark: Could not open audio device for playback. (6)
```

Fourdee, 2019-02-11:
## Fix:
```
usermod -a -G dietpi,audio -d $G_FP_DIETPI_USERDATA/mopidy mopidy
```

Fourdee, 2019-02-11:
@arkhub 

Many thanks for reporting this issue 👍 

You can apply the fix now with: https://github.com/Fourdee/DietPi/issues/2536#issuecomment-462433840

Also patched for v6.22.

Completed
```
 DietPi-JustBoom
─────────────────────────────────────────────────────
 Mode: ALSA Output Info

access: RW_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 440
buffer_size: 8820
```

closed 2019-02-11 by Fourdee