MichaIng/DietPi #2537: Rkhunter

WTFMaster, 2019-02-11:
Wish like to have Rkhunter in the Software-Archive. An Installer + Setup for Cron/Reporting would be nice.

MichaIng, 2019-02-11:
@WTFMaster 
Thanks for your request.

Great request:
- Anti virus or rootkit solutions are totally missing in our software offer, although Linux system do not seem to be the obvious target 😉.
- So yeah its a rootkit/backdoor/exploit scanner: https://en.wikipedia.org/wiki/Rkhunter
- Official website: http://rkhunter.sourceforge.net/
- Install instructions (own installer): https://sourceforge.net/p/rkhunter/rkh_code/ci/master/tree/files/README
- Install script: https://sourceforge.net/p/rkhunter/rkh_code/ci/master/tree/installer.sh
  - It does not really install something, just creates the config file according to install directory.
  - Yeah adding a cron job for regular check, schedule based on user choice would be nice + logging to permanent log dir.

MichaIng, 2019-02-12:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+41

Gill-Bates, 2019-03-25:
Awesome! I've vote for it.