MichaIng/DietPi #2540: DietPi-Software | WireGuard: Enhanced way to add multiple clients

MichaIng, 2019-02-12:
Ref: https://github.com/Fourdee/DietPi/issues/2491#issuecomment-462419860

This is mainly an online docs task, but also check back our configs and comments.

MichaIng, 2019-02-12:
@Fourdee 
How actually does the client authenticate against the server on OpenVPN? Client cert is at start only present on the client, which again only has the `ca.cert` to verify the server. But how does the server decide which client to allow and which not? Or are by default all clients allowed??

But our install matches/is derived from official docs: https://openvpn.net/community-resources/how-to/#setting-up-your-own-certificate-authority-ca-and-generating-certificates-and-keys-for-an-openvpn-server-and-multiple-clients

MichaIng, 2019-02-12:
Implemented above results into: https://github.com/Fourdee/DietPi/pull/2526
- [ ] Update online docs

Joulinar, 2019-02-12:
Well done guys, nice online docs. Now I'm able to print QR code in smaller size to my screen 👍 . Relay nice, appreciate all your hard work.

jvteleco, 2019-02-15:
Love DietPi and how easy is to install and setup things.
I noticed on the Wireguard notes, I think there is a typo with the IP of client example (instead of 10.9.0.3 should be 10.8.0.3 ?)
I had a little play to try to make this more automated based on your template. I am not an expert, so feel free to improve. I basically created a variable to store the client IP (WireGuard Client_IP) and then made easier adding the info to the wg0.conf:
```

WGC_IP="10.8.0.3"
cp -a wg0-client.conf wg0-client2.conf
G_CONFIG_INJECT 'Address = ' "Address =  $WGC_IP" wg0-client2.conf
G_CONFIG_INJECT 'PrivateKey = ' "PrivateKey = $(<client2_private.key)" wg0-client2.conf

echo -e "\n\n[Peer]"  >> /etc/wireguard/wg0.conf
echo -e "PublicKey = $(<client2_public.key)" >> /etc/wireguard/wg0.conf
echo -e "AllowedIPs = $WGC_IP/32\n"  >> /etc/wireguard/wg0.conf
WGC_IP="0"

```

Joulinar, 2019-02-15:
@jvteleco 10.9.0.x should be correct. Guys changed it to avoid conflicts with NordVPN

jvteleco, 2019-02-16:
OK, I see.
But the wg0.conf of the server still has 10.8.0.1
https://github.com/Fourdee/DietPi/blob/93e717896d194e099e32b4b122a2206e387f2b30/dietpi/dietpi-software#L10016

MichaIng, 2019-02-16:
@jvteleco @Joulinar 
On current v6.21 it is still `10.8.0.X` which is fine if you don't use OpenVPN (which DietPi-NordVPN is based on) concurrently.

To achieve concurrent OpenVPN compatibility we switch to `10.9.0.X` with v6.22. However this does not affect existing installs, as we do not want to touch existing configs, which we assume run as they are and you configured your clients accordingly already. So `10.8.0.3` is correct if the WireGuard server IP is `10.8.0.1`.

Ah yeah online docs have already been updated to new IP scheme, however I guess you guys figure it out, if your existing server/client IPs are `10.8.0.X` 😉.
_______
> I had a little play to try to make this more automated based on your template.

Jep some automation makes sense here.
Actually by times I have the idea to create a simply GUI to add/remove client configs and configure the server and clients to your needs. It would then scan the server config for `10.9.0.$i` (respectively the IP range based on server IP) entries, where `$i` counts up from `2` until no matching entry in `wg0.conf` was found. This first free IP will then be used for a new client.

Joulinar, 2019-02-16:
Ah yeah that's how I meant my comment. You will change it in 6.22. Sorry for being that inaccurate in my comment before.

Your idea with the automation sound good. Will be happy to test it once ready.

MichaIng, 2019-02-21:
Where we could learn a bid from: https://github.com/burghardt/easy-wg-quick

Joulinar, 2019-02-21:
that looks straightforward. Also good idea to give a name to the config files, to have them identified later on :)

MichaIng, 2019-02-21:
@Joulinar 
Jep, I have the idea to add a simple GUI to add/remove/configure client configs. I would have done it just numbers, e.g. wg-client1.conf wg-client2.conf ... but why not giving them individual names.

So far our implementation works fine, but by times I will check out the additional `iptable` rules and stuff from that script.

Joulinar, 2019-02-21:
Yes for me it's working very well as you have implemented it. And quite stable. 
Personally I changed the names of the client configs already, just to know who is who. Like:

`S8_client.conf`
`TabS3_client.conf`

closed 2019-02-24 by MichaIng