MichaIng/DietPi #2542: docker image disappeared after update to dietpi 6.21.1

n0rrin, 2019-02-13:
Hi.
I used a docker image.
After updating dietpi to v6.21.1 there is no more image when i use
sudo docker images
or
sudo docker ps -a

I am not the only one having this issue.
https://dietpi.com/phpbb/viewtopic.php?f=11&t=5556

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee
stretch
4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux
Odroid XU3/XU4/HC1/HC2 (armv7l)

MichaIng, 2019-02-13:
@n0rrin 
Thanks for your report. We will investigate. Could you send a `dietpi-bugreport` and paste the ID here?

n0rrin, 2019-02-13:
Sure!
f58626f5-6e37-4b0e-b458-62706ae11e81
Hope this helps.

MichaIng, 2019-02-18:
@n0rrin 
Sorry for the delay. Which was the DietPi version you updated from?

Can you check it the image is present in `/mnt/dietpi_userdata/docker-data`
and check for correct entry in: `cat /etc/docker/daemon.json`

MichaIng, 2019-03-08:
Ah, just rechecked the bug report and found that your `/etc` directory is nearly empty. It seems you or something accidentally removed this, including docker configuration file, which causes your symptom.

I hope you have a backup to restore the removed files from. Should be sufficient to copy back only missing files while keeping existing ones, at least as a first try.

MichaIng, 2019-03-12:
I mark this as closed now. However feel free to reopen if required.

closed 2019-03-12 by MichaIng