MichaIng/DietPi #2543: Cannot update from v6.19.7 to v6.21.1

Valdhor, 2019-02-13:
#### Details:
- Date           | Wed 13 Feb 16:03:12 GMT 2019
- Bug report     | N/A
- DietPi version | v6.19.7 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | #1 SMP PREEMPT Wed Apr 18 16:59:34 CEST 2018
- Distro         | stretch (index=4)
- Command        | chmod -R +x /DietPi /var/lib/dietpi/services /etc/cron.daily/dietpi /etc/cron.hourly/dietpi /etc/profile.d/dietpi-*.sh /etc/bashrc.d/dietpi-bash_completion.sh /etc/bashrc.d/dietpi-login.sh /etc/bashrc.d/dietpi-posix-set-locale.sh /etc/bashrc.d/dietpi-ps1_kill_animation.sh /etc/bashrc.d/dietpi-unsupported_terminal.sh
- Exit code      | 1
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
chmod: cannot access '/etc/profile.d/dietpi-*.sh': No such file or directory
```

--------------------------------------------------------------------

MichaIng, 2019-02-13:
@Valdhor 
I guess you tried to update to v6.20 before which failed?

Please do the following:
```
> /etc/profile.d/dietpi-unsupported_terminal.sh
> /DietPi/dietpi/.update_stage
echo 1 /DietPi/dietpi/.install_stage
dietpi-update 1
```

The first steps recover the intended v6.19 situation.

Valdhor, 2019-02-13:
That fixed it. Thank you. Now on v6.21.1

closed 2019-02-13 by Valdhor