MichaIng/DietPi #2547: Problem with soundcard i-sable-2km

klasLiesen, 2019-02-14:
I get error 2 and 'all' failed when selecting/installing soundcard i-sabre-2km in Dietpi-config.
Had no problems prior to Dietpi 6.20.

[image]

MichaIng, 2019-02-14:
@klasLiesen 
Thanks for your report.

The headers does not seem to be installed, which is strange according to the code as you should have received an error message about that before.

Please check:
```
uname -r # Your active kernel is indeed 4.14.79, which is current repo version since last November so should be
dpkg -l | grep raspberrypi-kernel # Kernel headers are installed and match the kernel version
```

klasLiesen, 2019-02-14:
Thanks for your response.

Here is the result plus installation progress of the soundcard. 

[image]

[Sparade Terminal-utmatningar.txt](https://github.com/Fourdee/DietPi/files/2864710/Sparade.Terminal-utmatningar.txt)

MichaIng, 2019-02-14:
@klasLiesen 
Everything as intended. Please check:
```
ls -l /lib/modules/4.14.79-v7+/ # Should contain the "source" dir
dpkg -S /lib/modules/4.14.79-v7+/source # Should list "raspberrypi-kernel-headers" as provider of this dir
```

If somehow this is not the case, please try to reinstall the kernel headers:
`apt reinstall raspberrypi-kernel-headers`

Otherwise I am a bid lost, since this worked well during tests. Perhaps the sources dir the Makefile expects needs to be changed, although very unlikely:
`KERNEL_SRC = /lib/modules/$(shell uname -r)/source` => `KERNEL_SRC = /lib/modules/$(shell uname -r)/build`

klasLiesen, 2019-02-14:
I get this when trying the commands.

I have no problems with other soundcards, only the Sabre.

[image]

[image]

If we can't solv this, is it possible to get a download link to the last dist of DietPi prior to 6.20?

Thank for your help.
Hope you will find a way to help me with this.

/Klas

MichaIng, 2019-02-14:
Ah sorry the APT reinstall command is: `apt install --reinstall raspberrypi-kernel-headers`

klasLiesen, 2019-02-14:
Reinstall now worked fine but still no luck with the Sabre card i'm afraid.

/Klas

[image]

MichaIng, 2019-02-14:
Okay, please try the following to switch the source dir, according to most guides this should be `/lib/modules/$(shell uname -r)/build` instead of `/lib/modules/$(shell uname -r)/source`.
Simply create a symlink to redirect to the correct source dir:
`ln -s /lib/modules/$(shell uname -r)/build /lib/modules/$(shell uname -r)/source`

If this works we need to commit to the module dev to switch this in the `Makefile` to match current `raspberrypi-kernel-headers` file structure.

klasLiesen, 2019-02-14:
Didn't work all the way.

/Klas

[image]

MichaIng, 2019-02-14:
@klasLiesen 
`gcc: Command not found`

Please do: `G_AGI gcc`
_____
The symlink build => source btw. is correct. `rpi-source` does the same, respectively links to both from the same /usr/src/linux-headers-* dir: https://github.com/notro/rpi-source/blob/master/rpi-source#L340-L341

MichaIng, 2019-02-14:
PR up to fix and add infos: https://github.com/Fourdee/DietPi/pull/2550

klasLiesen, 2019-02-15:
Good morning Michalng.
This time I get this response.

 ────────────────────────────────────────────────
 DietPi | Fri 15/02/19 - 07:06
 ────────────────────────────────────────────────
 v6.21.1 | RPi 3 Model B (armv7l)
 ────────────────────────────────────────────────
 eth0 | 192.168.1.211
 ────────────────────────────────────────────────
 DietPi Team     : Daniel Knight (founder), MichaIng, K-Plan
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@DietPi:~# G_AGI g++-4.9
[  OK  ] Root access verified.
[  OK  ] APT installation for: g++-4.9, please wait...
Selecting previously unselected package libmpfr4:armhf.
(Reading database ... 49759 files and directories currently installed.)
Preparing to unpack .../00-libmpfr4_3.1.5-1_armhf.deb ...
Unpacking libmpfr4:armhf (3.1.5-1) ...
Selecting previously unselected package libmpc3:armhf.
Preparing to unpack .../01-libmpc3_1.0.3-1_armhf.deb ...
Unpacking libmpc3:armhf (1.0.3-1) ...
Selecting previously unselected package binutils.
Preparing to unpack .../02-binutils_2.28-5_armhf.deb ...
Unpacking binutils (2.28-5) ...
Selecting previously unselected package cpp-4.9.
Preparing to unpack .../03-cpp-4.9_4.9.3-14_armhf.deb ...
Unpacking cpp-4.9 (4.9.3-14) ...
Selecting previously unselected package libatomic1:armhf.
Preparing to unpack .../04-libatomic1_6.3.0-18+rpi1+deb9u1_armhf.deb ...
Unpacking libatomic1:armhf (6.3.0-18+rpi1+deb9u1) ...
Selecting previously unselected package libasan1:armhf.
Preparing to unpack .../05-libasan1_4.9.3-14_armhf.deb ...
Unpacking libasan1:armhf (4.9.3-14) ...
Selecting previously unselected package libubsan0:armhf.
Preparing to unpack .../06-libubsan0_6.3.0-18+rpi1+deb9u1_armhf.deb ...
Unpacking libubsan0:armhf (6.3.0-18+rpi1+deb9u1) ...
Selecting previously unselected package libgcc-4.9-dev:armhf.
Preparing to unpack .../07-libgcc-4.9-dev_4.9.3-14_armhf.deb ...
Unpacking libgcc-4.9-dev:armhf (4.9.3-14) ...
Selecting previously unselected package gcc-4.9.
Preparing to unpack .../08-gcc-4.9_4.9.3-14_armhf.deb ...
Unpacking gcc-4.9 (4.9.3-14) ...
Selecting previously unselected package libc-dev-bin.
Preparing to unpack .../09-libc-dev-bin_2.24-11+deb9u3_armhf.deb ...
Unpacking libc-dev-bin (2.24-11+deb9u3) ...
Selecting previously unselected package linux-libc-dev:armhf.
Preparing to unpack .../10-linux-libc-dev_4.9.82-1+deb9u3+rpi1_armhf.deb ...
Unpacking linux-libc-dev:armhf (4.9.82-1+deb9u3+rpi1) ...
Selecting previously unselected package libc6-dev:armhf.
Preparing to unpack .../11-libc6-dev_2.24-11+deb9u3_armhf.deb ...
Unpacking libc6-dev:armhf (2.24-11+deb9u3) ...
Selecting previously unselected package libstdc++-4.9-dev:armhf.
Preparing to unpack .../12-libstdc++-4.9-dev_4.9.3-14_armhf.deb ...
Unpacking libstdc++-4.9-dev:armhf (4.9.3-14) ...
Selecting previously unselected package g++-4.9.
Preparing to unpack .../13-g++-4.9_4.9.3-14_armhf.deb ...
Unpacking g++-4.9 (4.9.3-14) ...
Setting up libatomic1:armhf (6.3.0-18+rpi1+deb9u1) ...
[... 60 lines cut ...]
[FAILED] DietPi-Set_Hardware | make

─────────────────────────┤ DietPi Error Handler: ├──────────────────────────────
tPi-Set_Hardware: make
Exit code: 2
DietPi version: v6.21.1 (Fourdee/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4
Image creator: DietPi Core Team
Pre-image: Raspbian Lite

 file contents:
e -C /lib/modules/4.14.79-v7+/source
DIRS=/tmp/DietPi-Set_Hardware/i-sabre-k2m_build KBUILD_VERBOSE=0 modules
e[1]: Entering directory '/usr/src/linux-headers-4.14.79-v7+'
cripts/gcc-version.sh: line 26: gcc: command not found
cripts/gcc-version.sh: line 27: gcc: command not found
e[1]: gcc: Command not found
e[1]: gcc: Command not found
e[1]: gcc: Command not found
e[1]: gcc: Command not found

                   Retry : Re-run the last command that failed

                    <Ok>                            <Exit>

────────────────────────────────────────────────────────────────────────────────

MichaIng, 2019-02-15:
Hmm:
```
Selecting previously unselected package gcc-4.9.
Preparing to unpack .../08-gcc-4.9_4.9.3-14_armhf.deb ...
Unpacking gcc-4.9 (4.9.3-14) ...
```
So `gcc` is installed, and the correct version as expected.

But again `make` reports:
```
Entering directory '/usr/src/linux-headers-4.14.79-v7+'
$cripts/gcc-version.sh: line 26: gcc: command not found
```

Works well here (x86 VM):
```
root@VM-Stretch:/lib/modules/4.9.0-8-amd64/source/scripts# ./gcc-version.sh gcc
0603
```
__________
@klasLiesen 
Thanks for testing.

What do those say:
```
command -v gcc
gcc --version
```
_________
@Fourdee 
Looks like we missed testing this and `rpi-source` indeed includes some checks and steps that are missing when installing the `raspberrypi-kernel-headers`. While not required for WireGuard obviously, here it is.

While I definitely don't want to revert to `rpi-source` method (even it's dev renders it obsolete and only useful as companion to rpi-update for testing/dev reasons), but we need to figure out the missing steps here and remember for other module builds as well. At least the gcc/g++ 4.9 install is very common for this kernel and I wonder why it is not a dependency of the header package. Same for the ../source => ../build symlink 🤔. On x86 both is available, pointing to different directories, where on RPi this is merged, but obviously module Makefiles can expect both...

klasLiesen, 2019-02-15:
Hope this is what you need.
/Klas
[image]

MichaIng, 2019-02-15:
Strange, `gcc` package is installed, but no binary (link).

Just checked, indeed `gcc` binary to version is not created automatically .... 🙄:
```
Setting up gcc-6 (6.3.0-18+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u3) ...
root@VM-Stretch:~# which gcc
root@VM-Stretch:~# which gcc-6
/usr/bin/gcc-6
```

On RPi:
```
2019-02-15 14:26:55 root@micha:/tmp# dpkg-deb -c gcc-4.9_4.9.4-2+rpi1+b19_armhf.deb
drwxr-xr-x root/root         0 2018-02-08 01:59 ./
drwxr-xr-x root/root         0 2018-02-08 01:59 ./usr/
drwxr-xr-x root/root         0 2018-02-08 01:59 ./usr/bin/
-rwxr-xr-x root/root    667784 2018-02-08 01:59 ./usr/bin/gcc-4.9
-rwxr-xr-x root/root     26348 2018-02-08 01:59 ./usr/bin/gcc-ar-4.9
-rwxr-xr-x root/root     26348 2018-02-08 01:59 ./usr/bin/gcc-nm-4.9
-rwxr-xr-x root/root     26348 2018-02-08 01:59 ./usr/bin/gcc-ranlib-4.9
-rwxr-xr-x root/root    371284 2018-02-08 01:59 ./usr/bin/gcov-4.9
drwxr-xr-x root/root         0 2018-02-08 01:59 ./usr/lib/
```

So do:
`ln -s /usr/bin/gcc-4.9 /usr/bin/gcc`
_______
To check:
- Is there a way to force a specific gcc binary for the `make` command?
- Because the above would not work or break other `gcc` versions, if installed...

klasLiesen, 2019-02-15:
We have a success.

I used: ln -s /usr/bin/gcc-4.9 /usr/bin/gcc

and then selected/installed soundcard i-sabre-2km in Dietpi-config.

Should I do: ln -s /usr/bin/g++-4.9 /usr/bin/g++     as well?
/Klas

[image]

MichaIng, 2019-02-15:
@klasLiesen 
Great finally 😅.

Nope g++ is not required then. All this was only to build the kernel module and this has obviously finished successfully.

Hmm even the official docs state to install g++ as well, it indeed seems not to be required. However this is something we can test ourself to fine tune the install/build script.

klasLiesen, 2019-02-15:
I tried: ln -s /usr/bin/g++-4.9 /usr/bin/g++  and selected/installed soundcard i-sabre-2km in Dietpi-config. I got no complains but the soundcard did not show in Roon.

I then: apt install --reinstall raspberrypi-kernel-headers
G_AGI g++-4.9
ln -s /usr/bin/gcc-4.9 /usr/bin/gcc
and selected/installed soundcard i-sabre-2km in Dietpi-config.
(Don't know if that's the right way to do it)
Soundcard was back in Roon and music is playing.

Thanks for all help and have a nice weekend.

Klas

MichaIng, 2019-02-15:
@klasLiesen 
Jep `ln -s /usr/bin/gcc-4.9 /usr/bin/gcc` or giving this information inside the code to the `make` command is definitely required. The `g++` binary obviously doesn't play a role here, also I couldn't find any `g++` calls within the scripts in the kernel source so far...

closed 2019-03-23 by MichaIng