MichaIng/DietPi #2549: Removing Sickrage service

Roxas72, 2019-02-14:
Hello,
I'm trying to switch from Sickrage to Medusa, but I'm having trouble removing the dietpi-service (or at least what remains of it). I've manually removed every sickrage from the userdata folder but when typing the dietpi-services status command the following shows up. Do you have any suggestion to solve this? 

[FAILED] DietPi-Services | sickrage	failed (Result: exit-code) since Thu 2019-02-14 16:58:55 GMT; 2min 26s ago
● sickrage.service - SickRage
   Loaded: loaded (/etc/systemd/system/sickrage.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-02-14 16:58:55 GMT; 2min 26s ago
  Process: 1003 ExecStart=/usr/bin/python /mnt/dietpi_userdata/sickrage/SickBeard.py -q --daemon --nolaunch --datadir=/mnt/dietpi_userdata/sickrage (code=exited, status=210/CHROOT)

Feb 14 16:58:54 DietPi systemd[1]: sickrage.service: Unit entered failed state.
Feb 14 16:58:54 DietPi systemd[1]: sickrage.service: Failed with result 'exit-code'.
Feb 14 16:58:55 DietPi systemd[1]: sickrage.service: Service hold-off time over, scheduling restart.
Feb 14 16:58:55 DietPi systemd[1]: Stopped SickRage.
Feb 14 16:58:55 DietPi systemd[1]: sickrage.service: Start request repeated too quickly.
Feb 14 16:58:55 DietPi systemd[1]: Failed to start SickRage.
Feb 14 16:58:55 DietPi systemd[1]: sickrage.service: Unit entered failed state.
Feb 14 16:58:55 DietPi systemd[1]: sickrage.service: Failed with result 'exit-code'.

MichaIng, 2019-02-14:
`rm /etc/systemd/system/sickrage.service` :smile:

Roxas72, 2019-02-14:
Easier than I thought I would be lol 
Thank you very much

closed 2019-02-14 by Roxas72