MichaIng/DietPi #2551: Failed to connect

techano, 2019-02-15:
#### Details:
- Date           | Fri 15 Feb 03:13:38 GMT 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (index=1)
- Kernel version | #1159 Sun Nov 4 17:28:08 GMT 2018
- Distro         | stretch (index=4)
- Command        | Connection test: https://github.com/weaved/installer/raw/master/Raspbian%20deb/1.3-07/weavedconnectd_1.3-07z1_armhf.deb
- Exit code      | 8
- Software title | DietPi-Software

#### Steps to reproduce:

1. Go to Dietpi-software
2, try to install Remot3.it (Weaved)

#### Expected behaviour:

- Should be a successful install.

#### Actual behaviour:

- Fails to download the file with the connection to github successful just the file seems to not exist

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-02-15 03:13:31--  https://github.com/weaved/installer/raw/master/Raspbian%20deb/1.3-07/weavedconnectd_1.3-07z1_armhf.deb
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

MichaIng, 2019-02-15:
@techano 
Thanks for your report.

Indeed weaved branch removed from GitHub 😱: https://github.com/weaved/

MichaIng, 2019-02-15:
PR up to switch to new repo: https://github.com/Fourdee/DietPi/pull/2552

Fourdee, 2019-02-19:
Testing passed, will mark as completed: https://github.com/Fourdee/DietPi/pull/2552#issuecomment-464109910

closed 2019-02-19 by Fourdee