MichaIng/DietPi #2553: Plex new server versions, official repository for ARM versions

WolfganP, 2019-02-15:
Just as a heads up, Plex started working a while ago in simplifying their code base and tools (https://forums.plex.tv/t/much-faster-hopefully-smaller-mostly-and-better-pms-builds-preview-1-15-0-573-123756e9a/308294)

One of the consequences is that ARMv7 and ARMv8 images now will be officially maintained by Plex themselves (the ARM community version integrated now in DietPi will be discontinued), and they also plan to support an official apt source (confirmed at https://forums.plex.tv/t/official-pms-for-armv7-what-will-happen-with-this-community-one/372290/4)

I'll keep this issue updated with official information on Plex developments that may impact DietPi, if anyone wants to test, beta versions for all platforms are available already at https://forums.plex.tv/t/much-faster-hopefully-smaller-mostly-and-better-pms-builds-preview-1-15-0-573-123756e9a/308294

MichaIng, 2019-02-15:
@WolfganP 
Many thanks for this information. Great that all binaries can be then downloaded from the same source. An official APT repo of course would make things easier again.

I think we could already add those preview versions with DietPi v6.22 or are you aware of any issues with them reported by users?

WolfganP, 2019-02-17:
@MichaIng I tried the new binary in a separate RasPi2 and it worked OK (but it was just a test with a handful of media files and not my big media list nor the customized environment). I plan to migrate once they ship the production images.

A few users at https://forums.plex.tv/tags/server-linux report unresponsive server after upgrade, but as there's no automated official upgrade path, some seems to be manual errors. Clearly it's a public beta, but worth to try for power users.

MichaIng, 2019-02-17:
@WolfganP 
> I tried the new binary in a separate RasPi2 and it worked OK

Do you have another `locale` active than `en_US.UTF-8` or were you able to switch this afterwards?

Currently when installing Plex on ARM devices (via 3rd party repo), we switch active locale to `en_US.UTF-8` and inform user about that.

WolfganP, 2019-02-17:
@MichaIng my locale is en_US.UTF-8. I deleted my test env for a diff test that will take some time, but if any chance appears I'll retry Plex on another locale, and monitor the plex forums for related reports if any.

MichaIng, 2019-02-17:
@WolfganP 
Jep that is great, very helpful 👍!

MichaIng, 2019-02-25:
@WolfganP 
The v15 download links on the forum thread are not updated anymore, but in announcements I find very regular v15 updates to the beta update channel, fixing several of the issues users mentioned. Can't find download links, so it seems this is available through the internal updater, or do you know where to download those directly?

I am still thinking we could/should add this with DietPi v6.22, but only if a very recent external download link is available.

WolfganP, 2019-02-26:
@MichaIng Yes, up to now the v1.15 files were available to anyone, kind of alpha state distribution. It seems they are now confident on the new version stability to move the files to the official beta distribution channel, but not into production nor any sign of the official apt source yet.
I'll keep an eye on it and comment back in here once news appear.

MichaIng, 2019-03-14:
Quick link to check v1.15 release state: https://forums.plex.tv/t/plex-media-server/30447/last

WolfganP, 2019-03-14:
Just announced, `Plex Media Server 1.15.1.791 is now available to everyone` in download pages https://forums.plex.tv/t/plex-media-server/30447/250 (no indication of an official apt repo yet)

Some issues reported by users at the official forums https://forums.plex.tv/tags/server-raspberry-pi , seems unrelated to the update but possible workaround at https://forums.plex.tv/t/plex-server-suddenly-not-starting-on-raspberry-pi-after-crashed-update/387681/5

```
$ sudo LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libarmmem-v6l.so
$ sudo rm /etc/ld.so.preload
$ sudo apt-get update
$ sudo apt-get upgrade
$ sudo reboot
```

MichaIng, 2019-03-15:
@WolfganP 
😄 less than an hour edited after I had a look.

Getting latest release link automatically is possible via: https://plex.tv/pms/downloads/5

MichaIng, 2019-03-15:
@WolfganP 
Okay I implemented latest version catch as above. With this I am okay with merging this change already into v6.22, since quick hotfix releases Plex side will immediately result in a fixed install our side. Without this I would have not been keen to ship this version as it might not yet be too stable (although pushed to releases?).

If you have an ARM test system or one where dev branch is okay, might you test if indeed with this ARM binaries no en_US locale is required anymore? Although it is actually pretty sure and you already mentioned such above. Would be just a last test to be 100% sure 😉.

WolfganP, 2019-03-15:
@MichaIng as per discussions at the Plex forum, the official repo + documentation and migration strategy may come up after the weekend, so I suggest to wait a few days.
I tried production PMS v.1.15 by means of manual install, and it works well as far as I could test.

MichaIng, 2019-03-15:
@WolfganP 
Okay makes sense then to wait for the official migration strategy then. However I will merge Beta v6.22.1 later with current stage (after currently running stets succeed) to allow testing as well other changes. But before release will add migration for existing ARM Plex installs.

MichaIng, 2019-03-16:
v1.15 has been pushed to the dev2day repo as well, which will be dropped ~1 month later: https://forums.plex.tv/t/official-pms-for-armv7-what-will-happen-with-this-community-one/372290/21

Official migration guide is still missing. Looks like we will skip reinstall for now and do the migration with v6.23  as planned release is before repo drop.

MichaIng, 2019-03-23:
Update to official packages seems to be indeed that easy. However we will do that for ARM devices with v6.23 to allow some testing: https://forums.plex.tv/t/official-pms-for-armv7-what-will-happen-with-this-community-one/372290/37

closed 2019-03-23 by MichaIng

MichaIng, 2019-04-18:
Official migration steps are available. Will implement this now for v6.23: https://github.com/MichaIng/DietPi/issues/2655

WolfganP, 2019-04-18:
Just for reference, this is the official documentation of the new repo https://support.plex.tv/articles/235974187-enable-repository-updating-for-supported-linux-server-distributions/

Caveat: Plex v1.15 series doesn't seem to be compatible/stable with old & 3rd party clients (ie Orca's Sansumg TV), so users may need to stay on v1.14 branch if Plex client is not up to date.

MichaIng, 2019-04-18:
@WolfganP 
> Just for reference, this is the official documentation of the new repo 

Jep, I'm following that.

> Caveat: Plex v1.15 series doesn't seem to be compatible/stable with old & 3rd party clients (ie Orca's Sansumg TV), so users may need to stay on v1.14 branch if Plex client is not up to date.

Good to know, but nothing we can take care of. We will migrate all users to new APT repo and upgrade them to v1.15 by running dietpi-software reinstall. Aside from this, even with our previous install methods all ARM users would face the same issue (dev2day is on v1.15 as well for a month) and fresh installs anyway.

MichaIng, 2019-04-18:
Done: https://github.com/MichaIng/DietPi/pull/2722
Luckily very easy, simply remove dev2day package, remove repo and reinstall on top 😃.

I am very happy with this. APT makes things much easier. The v6.22 method to estimate current version from API URL relied very much on consistent tags and those changed quite often, as could be seen on the 3rd party updater which required/requires changes quite often: https://github.com/mrworf/plexupdate