MichaIng/DietPi #2557: Can't set cpu affinity/ clicking issues on USB

rwortman2, 2019-02-16:
### Creating a bug report/issue

#### Required Information
- DietPi version |  6.21.1
- Distro version |  9.6
- Kernel version |  4.14.79-V7+ #1159
- SBC device | 3 model  (armv71)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | PNY 16GB

#### Additional Information (if applicable)
- Software title | roonbridge
- Was the software title installed freshly.  Installed with dietpi image
- Can this issue be replicated on a fresh installation of DietPi? I would imagine so
- `dietpi-bugreport` ID | ed987da7-8247-4ac6-ac0e-f58a33c89e02

#### Steps to reproduce

My Pi produces random clicks into my V90-DAC dac at 96/24 speed.  44/16 does not produce problem, 96/16 seems to work too.  I did hours of internet searching.  I found the discussion here about changing the cpu affinity for the usb interrupts.   When I attempt to do that I get this error:   "-bash: echo: write error: input output error"
2. ...

#### Expected behaviour
should be able to set cpu affinity
- ...

#### Actual behaviour
write error'   "-bash: echo: write error: input output error"
- ...

#### Extra details

- ...

MichaIng, 2019-02-16:
@rwortman2 
Thanks for your report.

We fixed this on Odroid C2 via: https://github.com/Fourdee/DietPi/issues/2101
Looks like we should apply this to USB DACs on all systems? But we need to derive valid CPUs based on CPU count.

Other related: https://dietpi.com/phpbb/viewtopic.php?f=11&t=4585&start=10

rwortman2, 2019-02-16:
I read that thread on 2101 but as reported, I can't make the recommended changes.  The file or pseudofile seems read only.  I am looking at the Linux tweak thread you linked but I fear that I an not Linus command language experienced enough to do it.

MichaIng, 2019-02-17:
@rwortman2 
Ah I see, hmm different SBC, different pseudo file rules 🤔. Perhaps RPi has some `config.txt` entry that allows to adjust USB affinity.

The linked installer is a community project by one of our users. It should install `/usr/bin/Sound.sh` so make the `Sound.sh` command available to apply non-persistent audio tweaks: https://github.com/dynobot/Linux-Audio-Adjustments/blob/master/basic-install.sh#L13

It adds the script to rc.local to be called on boot. You might need to enable the related service via: `systemctl enable rc-local`

So after doing `curl -sSL https://github.com/dynobot/Linux-Audio-Adjustments/raw/master/basic-install.sh | bash` the command `Sound.sh` is not available?
To test: `which Sound.sh`

rwortman2, 2019-02-18:
I tried that command enabling  rc-local and then ran the installer.  Once again no sound.sh on the system.  The script runs fast and then reboots so I can't really see all the stuff it is printing during the purported install.  Not sure it would fix it anyway but it would be interesting to try.   I have a SPDIF hat on the way to see if using i2s to SPDIF instead of the USB port will result in smoother high bit rate operation.  That may make it work for me but doesn't really solve the existing usb issue.

MichaIng, 2019-02-25:
@rwortman2 
Sorry for late reply.

I think you should remove the reboot line from the script to check first what have been applied and check it's output: `sed -i /^reboot/d' basic-install.sh`

MichaIng, 2019-02-27:
@rwortman2 
One thing about rc.local. The system has changed with a recent systemd update, that prevents rc.local from being executed on boot. Solution: https://github.com/MichaIng/DietPi/issues/2566

MichaIng, 2021-02-19:
I mark this as closed as the RPi kernel does not allow changing the IRQ affinities. If it would have any positive effect is a different question but the RPi devs obviously don't give it a high priority nor are there relevant requests/reports at RPi repos.

closed 2021-02-19 by MichaIng