MichaIng/DietPi #2560: filebrowser

msongz, 2019-02-17:
### Creating a software request

#### Formal software information
- Software title |  filebrowser
- Short description |  Web File Browser which can be used as a middleware or standalone app
- Official URL (if available) |  https://filebrowser.xyz/
- GitHub URL (if available) |  https://github.com/filebrowser/filebrowser
- (Official) install documentation | 
https://docs.filebrowser.xyz/installation
#### Are there similar/alternative software titles available with DietPi-Software?

- nextcloud
- owncloud

#### What makes your requested software better than the above solutions, if available?

- lightweight
- easy to use

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. just follow offical steps

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

MichaIng, 2019-02-17:
@msongz 
Thanks for your request.

This is even closer to Syncthing than to ownCloud/Nextcloud. A very simple web based file access implementation

Related forum request: https://dietpi.com/phpbb/viewtopic.php?f=12&t=4008

CouldBeThis, 2021-04-22:
I would like to motivate this further as I don't think there is anything currently filling this use case. 

The project is a model of restrained development. Does one thing perfectly with just a few essential customizations, and nothing else.

MichaIng, 2021-08-16:
Detailed setup instructions: https://dietpi.com/phpbb/viewtopic.php?p=36827#p36827

Joulinar, 2021-08-16:
@MichaIng 
If you like, I could have a look on implementing this for 7.6

MichaIng, 2021-08-16:
Would be awesome, sounds pretty doable and a great light-end file sharing/browser implementation that was still missing. +1 for systemd unit you suggested. Then probably an own service user, but with `dietpi` as primary group and `UMask=002` so that other members of the `dietpi` group have full access to those files? Could be handy when using it for media stuff, an NFS/Samba drive included etc.

Joulinar, 2021-08-16:
Ok good. Will look into this once 7.5 has been released.

Joulinar, 2021-08-19:
@MichaIng 
Question: do you like to use filebrowser install script or should we adapt the steps accordingly.

Filebrowser install script basically detects os + arch and is going to create GitHub download URL

https://github.com/filebrowser/get/blob/0581887ba29c534a1d661b0df4d2001b779cde79/get.sh#L109-L111

MichaIng, 2021-08-19:
Yes exactly, we can adopt the steps we do in many other cases and skip the overhead of the installer, detecting the OS etc. You should also find the cases where we define the `arch=` variable based on `$G_HW_ARCH` to find the correct URL.

Joulinar, 2021-08-19:
yep. I'm just waiting on 7.5 release and the initial PR for 7.6 😉

closed 2021-09-16 by MichaIng