MichaIng/DietPi #2561: Odroid C1 | Stretch image

MichaIng, 2019-02-17:
Okay, I though open up a separate issue makes sense here, as things turn out to be more complicated.

Related issues:
- https://github.com/Fourdee/DietPi/issues/2026#issuecomment-461925950
- More and more software titles drop Jessie support, e.g.: https://github.com/Fourdee/DietPi/issues/2521
- Aim to drop Jessie support: https://github.com/Fourdee/DietPi/issues/2332

The problem seems to be, that from Meverics repo (and hardkernel sources) the latest kernel version is 3.10, while Debian Stretch usually requires 3.16 at least.

Some successful distro upgrades have been done by users, but they might have unexpected issues and limitations in cases: https://forum.odroid.com/viewtopic.php?f=114&t=17569&start=250

ARMbian offers a Stretch image based on kernel 4.18 but is has certain limitations as well, especially no graphics output, serial console only, no eMMC support: https://www.armbian.com/odroid-c1/#kernels-archive

Both above options don't lead to a reliable and functional Stretch image, although distro upgrading Meverics Jessie image at least does not touch kernel/driver so core functionality should stay?
If we have an Odroid C1, at least it is worth testing, distro upgrading to Stretch and test core + desktop + Kodi functionality.

Since I don't have one and Fourdee can't find his C1 currently, if there is anyone willing to try the above, this would be a great first step to check feasibility.

Fourdee, 2019-05-02:
@MichaIng 
Image live, based on ARMbian.

Some notes:
- CPU clocks appear lower, however, no kernel support to obtain clocks. cpu bench +10 seconds compared to 3.x kernel.
- Rebooting the system does not always "reboot", may hang with blue light on. Simply re-plug power cable
- No GPU support

Fourdee, 2019-05-02:
@MichaIng 

Hmm, alot of negatives for Stretch on the C1. 
Debating if we should drop the device altogether as all dev/support by open source comminuty seems to have stopped?

Fourdee, 2019-05-02:
@MichaIng 

I'll do 1 more try with Meveric 3.x under Stretch, when I can. See if any joy.

MichaIng, 2019-05-02:
Hmm dist-upgrading from Meverics Jessie kernel 3.10 image does not work as well?

Yeah I also believe in case we just offer the final Jessie image which user will have at least DietPi v6.24 support for and then drop it, respectively ship as EOL final image.

MichaIng, 2019-05-02:
Ah as said, I plan to merge DietPi v6.24 into the jessie-support branch as well, den most likely remove Jessie support from code with v6.25. The branch migration is done now already to have most users migrated as soon as possible. This allows to remove the migration code itself midterm earlier as we cannot guarantee DietPi-Globals (loaded from patch_file) Jessie support for old Jessie images forever.

jcw, 2019-05-04:
C1+ user here - I understand and respect your reasons for dropping support. Let me just mention that the board is still sold (even marked "hot" on hardkernel.org - maybe it's their way to clear inventory?). Even without GPU support, it's still a nice server with its 1 Gbit Ethernet, which is how I've used it.

MichaIng, 2019-05-04:
@jcw 
> marked "hot" on hardkernel.org - maybe it's their way to clear inventory?

This is how it must be interpreted. Sadly this is common practice across SBC manufacturers. Never buy an SBC that is older than one year. Kernels major versions are in very most cases never updated, limiting the support for newer distro versions. For manufacturer images its mostly worse. Hardkernel at least ships relatively current Ubuntu images, but the kernel is never updated. We use the minimal Debian images from Meveric but his possibilities are limited as well due to lack of kernel development.
As a result in case of C1 I couldn't fine any open source Debian derivatives that still supports C1 images.

This is the reason I would still never buy anything else than an RPi for personal use. This is the ONLY SBC that is at current stage forever supported with current kernel versions and very regular OS updates. Soon you can run your very day-1 RPi1 with Buster image and kernel 4.19. Outstanding from that point of view. Sadly very limited performance.

jcw, 2019-05-12:
Thanks for keeping the Odroid C1 supported - I'm trying the new 6.23 beta, but seem to be running into some issues. Context: image flashed to eMMC (via µSD adapter) on MacOS using `dd`. The first difference I note is that the result complains about not being mountable on MacOS (it used to come up as `/Volumes/boot` after the `dd` completes).

And then, in the Odroid, on bootup, with console attached, I get this:

```
 QA5:A;SVN:B72;POC:17F;STS:0;BOOT:0;INIT:0;READ:41;READ:41;READ:0;CHECK:0;PASS:0;
-----------------------------------------------------------------------
* Welcome to Hardkernel's ODROID-C... (Built at 19:33:00 Dec  8 2014) *
-----------------------------------------------------------------------
CPU : AMLogic S805
MEM : 1024MB (DDR3@792MHz)
BID : HKC1310001
S/N : HKC11122F37E012E
0x0000009f
Loading U-boot...success.

U-boot-00000-gb7b8dc2-dirty(odroidc@b7b8dc21) (Sep 19 2018 - 12:23:57)

I2C:   clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[20]=0
set output val 0xc1108058[20]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
[... 212 lines cut ...]
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
done.
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT!  UUID=18ef9117-8b19-4fa0-a35b-aa6023dd1e3d does not exist.  Dropping to a shell!
(initramfs)
```

jcw, 2019-05-12:
Aha, a few comments up: "no eMMC support" - that's probably it. This is unfortunate, for server use.

MichaIng, 2019-05-12:
@jcw 
> "no eMMC support" - that's probably it. This is unfortunate, for server use.

Jep that is one limitation of the mainline kernel that allows Debian Stretch, sadly.
___________
@Fourdee 
Hah, ARMbian ships a new image with Linux 5.0 and no hint about missing GPU support anymore: https://www.armbian.com/odroid-c1/
Might it be possible the mainline kernel added/widened support for this chip?

Digging: https://kernelnewbies.org/Linux_5.0#ARM
- Many new SBCs, platforms, Cortex-A5 and two Mali entries in the graphics section. I can't find it explicitly but who knows which little quirk made video break before.

I know you invested much time for the current C1 Stretch image, so don't take this as a request, but as information. I just wanted to verify the "no-eMMC" and found it by chance 😄.

jcw, 2019-05-12:
Could this be of use? - https://forum.odroid.com/viewtopic.php?f=111&p=255093

MichaIng, 2019-05-12:
@jcw 
Thanks for posting this as another source to hopefully get some feedback about C1 feature support of Linux 5.0.

itouch5000, 2019-05-18:
I am currently running `Armbian_5.84_Odroidc1_Debian_stretch_next_5.0.12.img` as tiny backup server. And it's working great even reboots are possible with this new kernel.

MichaIng, 2019-05-18:
@itouch5000 
Many thanks for reporting. Jep we should update our image to use this kernel as well.
But it is also possible to update it via APT package, right?

Okay, so still no HDMI. "serial console only" somehow implies this. "eMMC not supported" stated by ARMbian, so this will definitely not work: https://www.armbian.com/odroid-c1/

itouch5000, 2019-05-18:
@MichaIng 
yes, apt update and upgrade works

MichaIng, 2019-05-18:
@itouch5000 
Also to upgrade the kernel version, right? I.e. `dpkg -l` contains some `linux-image-*` like package?
However new DietPi image still makes sense to avoid issues on first boot, especially since apt-get dist-upgrade (most likely required for Linux upgrade) is not called automatically.

itouch5000, 2019-05-18:
yes I think so:
```
ii    linux-image-next-odroidc1    5.84    armhf    Linux kernel, version 5.0.12-odroidc1
```
full output: https://pastebin.com/raw/uE7hNVDG

MichaIng, 2019-05-18:
@itouch5000 
Perfect thanks.

itouch5000, 2019-05-20:
@MichaIng 
I just noticed that USB devices are not recognized at all. Although the odroid was booted with usb device connected.

itouch5000, 2019-05-20:
@MichaIng 
I recently noticed that usb and even usb hotplug is working under certain conditions:
Boot with a 8GB usb stick -> device not recognised
Boot with a 3TB hdd -> device not recognised
Boot with a 2GB usb stick -> device **recognised**
(usb stick with a GUID partiton table and formatted with [libreelec installer](https://libreelec.tv/downloads_new/))
Boot with this 2GB usb stick and the 8GB usb stick and/or the 3TB hdd -> all devices are **recognised**

Very odd. I'm not shure if that helps.

MichaIng, 2019-06-24:
Okay I mark this as closed. Issues should be reported as separate issues from now on.

closed 2019-06-24 by MichaIng