MichaIng/DietPi #2564: No pvr addons kodi ( kodi-addons-odroid )

nemesis1337, 2019-02-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.8

- Kernel version | `uname -a`
Linux mc 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
Odroid C2 (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
kodi
hts pvr

- Was the software title installed freshly or updated/migrated?
fresh install

- Can this issue be replicated on a fresh installation of DietPi?
yes
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. installed kodi via dietpi-software
2. apt install kodi-addons-odroid

#### Expected behaviour

- install all kodi 17.6 pvr addons

#### Actual behaviour

E: Unable to locate package kodi-addons-odroid

#### Extra details

- ...

MichaIng, 2019-02-17:
@nemesis1337 
Thanks for your report.

Note that the `kodi-addons-odroid` package is only available for Jessie systems, not on Stretch: https://forum.odroid.com/viewtopic.php?p=175190&sid=1f8e701205ff6bb7facd5edadfb6400a#p175203

nemesis1337, 2019-02-17:
Thanks @MichaIng is there any other way to get the pvr addons working in kodi perhaps?
They are working in LibreElec for example, but this distro is not as rock solid as Dietpi on the C2.

MichaIng, 2019-02-17:
@nemesis1337 
Not sure. We might need to address this question directly to Meveric, the maintainer of the Debian Stretch image for Odroids that DietPi is based on: https://forum.odroid.com/viewtopic.php?f=138&t=27449

Note that Kodi support was just enabled not too long ago. Perhaps some features known from the Jessie image are simply not (yet) available.

But I am no specialist for Kodi, since I never really used it, especially no special addons. Others might know more 😉.

nemesis1337, 2019-02-17:
Alright, thanks for your replies! I will wait patiently, let's see if Meveric has some time to look into this.
:+1:

MichaIng, 2019-02-17:
@nemesis1337 
Just read a bid about how this is usually enabled: https://kodi.wiki/view/PVR
- So you need a backend software.
- And in Kodi UI you should be able to install the addon (for Kodi communicating with the PVR backend software): https://www.howtogeek.com/247311/how-to-watch-live-tv-on-your-kodi-media-center-with-nextpvr/
  - This is a guide for Windows, but the Kodi UI part should be the same on Debian.

Not sure if the `kodi-addons-odroid` provided all in one package, but it should be possible to install backend and addon manually as above.

nemesis1337, 2019-02-17:
Actually I have everything running already the backend - tvheadend 4.3 (found all sat.channels).
I can watch tv via browser and VLC, normally you would indeed just enable the tvheadend pvr (hts) in Kodi and it works! But unfortunately in stretch there are no pvr addons to choose from in Kodi.
In Jessie there are also none, but you would be able to install them with "apt install kodi-addons-odroid".
I'm trying to find anything regarding porting it maybe from LibreElec, they seems to be ok there.
Actually there are pvr addons in stretch -> apt install kodi-pvr-hts but they will only be available for "kodi" and Dietpi seems to be using "kodi-odroid" (17.6-5+deb9).

MichaIng, 2019-02-17:
Hmm, just played around a bid on VM. When I select **TV**, it asks to install a backend (did not yet do that). When I go into the Kodi.tv repo to browse addons and type `next` it seems to attempt selecting NextPVR addon but again asks for a backend. But it is hidden in the list. I guess it shows/hides addons based on required backends.

Jep on Odroids, `kodi-odroid` is the package that works. `kodi` is from the Debian repo, working on other devices but obviously not Odroids, especially when it comes to GPU acceleration.

Did `tvheadend` particularly work on other systems with Kodi? Not sure if a specific addon is required for each backend?

nemesis1337, 2019-02-17:
Yeah it does work with LIbreElec, CoreElec and OSMC, they have their own linux OS, not based on debian I think. In kodi all the pvr's are available there. 

But those distro's seem to be kind of unstable regarding Odroids GPU acceleration. Also running other programs independantly from Kodi, like Docker, TVheadend server is not possible.

I did read something about "jessie-backports-sloppy", would something like this be possible to run an old version of the addons? (kodi 16.1). But i think that was mostly used for testing the migration to stretch.

MichaIng, 2019-02-17:
@nemesis1337 
Strange that the addons do not show up in the Kodi.tv repo.

A very careful try would be to install the package manually:
```
cd /tmp
wget http://fuzon.co.uk/meveric/pool/main/k/kodi-addons-odroid/kodi-addons-odroid_16.1-2+deb8_arm64.deb
dpkg -i kodi-addons-odroid_16.1-2+deb8_arm64.deb
rm kodi-addons-odroid_16.1-2+deb8_arm64.deb
```
It is made for Kodi 16.1 and will place the addon files into `/usr/local/lib/kodi/addons/` and `/usr/local/share/kodi/addons/`. No guarantee that this works, perhaps Kodi simply denies to enable the addons due to version mismatch. Do a `dietpi-backup` first, just in case, before trying it out.

nemesis1337, 2019-02-17:
This <almost> worked! I was able to install the addons with also installing libglew1.10

`wget http://ftp.de.debian.org/debian/pool/main/g/glew/libglew1.10_1.10.0-3_arm64.deb
dpkg -i libglew1.10_1.10.0-3_arm64.deb`

All the addons show up now in Kodi, but giving a "couldn't be loaded" error msg

So close haha!

nemesis1337, 2019-02-17:
> PVR - Add-on 'Tvheadend HTSP Client' is using an incompatible API version. XBMC minimum API version = '5.2.1', add-on API version '4.1.0'

nemesis1337, 2019-02-18:
Going to sleep now, but thanks for pointing me in some direction, will try more tomorrow!

MichaIng, 2019-02-18:
@nemesis1337 
Yeah I thought so. I guess it is also not possible to now update these addons from within the GUI right?

So we need to find another source for them that match the Kodi version.

nemesis1337, 2019-02-18:
I have a laptop configured with LibreElec and I'll start copying the .so files from it and see how far it will take me :-)

nemesis1337, 2019-02-21:
Finally got it working!! got myself a 2nd SD card installed LibreElec 8.2.5 with Kodi 17.6
and copied the file pvr.hts-3.4.27.1.zip from /.kodi/addons/packages
Simply went to running Dietpi -> addons and did a install from ZIP file
Configured username/password in the TVHeadend PVR and it worked!!

nemesis1337, 2019-02-21:
Thanks @MichaIng for pointing me in the right direction! Will try a "clean" intall some day to see if the packages I installed at testing are really needed..

MichaIng, 2019-02-21:
@nemesis1337 
Nice that is great. You you provide these zip files? Or perhaps I can check where to get them from official source.

nemesis1337, 2019-02-25:
Hi Michalng, yes I'll send it tomorrow.
It was actually on the SD image of the LibreElec 8.2.5 I installed temporarily.

nemesis1337, 2019-02-26:
Hi Michalng, you should be able to download it from here:
[https://nc.my.to/index.php/s/4BKPMXTSwC9J6qp](url)  (self signed cert)

MichaIng, 2019-02-27:
@nemesis1337 
Thank you very much!

nemesis1337, 2019-03-02:
Small update; I do notice a stutter when watching live TV, nor sure if it's my installation though, I'll try a fresh install some day. 
The stutter is kind of inconsistent sometimes the same channel plays fine sometimes it stutters. Almost like a video/audio re-sync.

MichaIng, 2022-09-17:
Marking this as closed. It has become obsolete with modern Odroid images, where the regular Debian `kodi` package as well as the official addons repository are installed.

closed 2022-09-17 by MichaIng