MichaIng/DietPi #2567: dietpi-update Problem

dgburu, 2019-02-19:
#### Details:
- Date           | Tue 19 Feb 12:42:43 CET 2019
- Bug report     | N/A
- DietPi version | v6.19.7 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 2 Model B (armv7l) (index=2)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | chmod -R +x /DietPi /var/lib/dietpi/services /etc/cron.daily/dietpi /etc/cron.hourly/dietpi /etc/                                                                                    profile.d/dietpi-*.sh /etc/bashrc.d/dietpi-bash_completion.sh /etc/bashrc.d/dietpi-login.sh /etc/bashrc.d/dietpi-pos                                                                                    ix-set-locale.sh /etc/bashrc.d/dietpi-ps1_kill_animation.sh /etc/bashrc.d/dietpi-unsupported_terminal.sh
- Exit code      | 1
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
chmod: cannot access '/etc/profile.d/dietpi-*.sh': No such file or directory
```

MichaIng, 2019-02-19:
@dgburu 
Thanks for your report.

You tried to update to v6.20 once already, right?

To solve and update to v6.21:
```
echo 1 /DietPi/dietpi/.install_stage
> /DietPi/dietpi/.update_stage
> /etc/profile.d/dietpi-unsupported_terminal.sh
dietpi-update 1
```

dgburu, 2019-02-19:
Hi
Yes i think so.
Finally I was able to update to v6.21 
Thanks for your help.

closed 2019-02-19 by dgburu