MichaIng/DietPi #2568: Home Assistant Install Fails

rucknapucknavitz, 2019-02-19:
#### Details:
- Date           | Tue 19 Feb 16:43:49 MST 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | Unknown Device (armv7l) (index=0)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | buster (index=5)
- Command        | G_AGI: cmake daemon gcc nmap net-tools swig uuid-dev libc-ares-dev libgnutls28-dev libgnutlsxx28 libglib2.0-dev libudev-dev libusb-1.0-0 libssl-dev libffi-dev libbz2-dev zlib1g-dev libreadline-dev libsqlite3-dev libncurses5-dev libncursesw5-dev
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Install Home Assistant from dietpi-software

#### Expected behaviour:

- Home Assistant Should Install

#### Actual behaviour:

- Install fails

#### Extra details:

- I tried `dpkg --configure -a` `and apt install -f`

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2019-02-20:
@rucknapucknavitz 
Thanks for your report.

Just tested on VM and worked well here. So either an RPi/Raspbian specific or setup specific issue.

Could you identify on which package(s) it actually fails, or plase copy&paste the whole APT run output, in case redo it:
`apt-get -y install cmake daemon gcc nmap net-tools swig uuid-dev libc-ares-dev libgnutls28-dev libgnutlsxx28 libglib2.0-dev libudev-dev libusb-1.0-0 libssl-dev libffi-dev libbz2-dev zlib1g-dev libreadline-dev libsqlite3-dev libncurses5-dev libncursesw5-dev`

rucknapucknavitz, 2019-02-20:
Thanks for the _very quick_ response.

Here's the terminal output:

> root@DietPi:~# apt-get -y install cmake daemon gcc nmap net-tools swig uuid-dev libc-ares-dev libgnutls28-dev libgnutlsxx28 libglib2.0-dev libudev-dev libusb-1.0-0 libssl-dev libffi-dev libbz2-dev zlib1g-dev libreadline-dev libsqlite3-dev libncurses5-dev libncursesw5-dev
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> gcc is already the newest version (4:6.3.0-4).
> gcc set to manually installed.
> libffi-dev is already the newest version (3.2.1-9).
> libusb-1.0-0 is already the newest version (2:1.0.22-2).
> libusb-1.0-0 set to manually installed.
> net-tools is already the newest version (1.60+git20180626.aebd88e-1).
> libssl-dev is already the newest version (1.1.1a-1).
> zlib1g-dev is already the newest version (1:1.2.11.dfsg-1).
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  libbz2-dev : Depends: libbz2-1.0 (= 1.0.6-8.1) but 1.0.6-9 is to be installed
>  libglib2.0-dev : Depends: libglib2.0-0 (= 2.50.3-2) but 2.58.3-1 is to be installed
>                   Depends: libglib2.0-bin (= 2.50.3-2) but it is not going to be installed
>                   Depends: libpcre3-dev (>= 1:8.31) but it is not going to be installed
>  libgnutls28-dev : Depends: nettle-dev but it is not going to be installed
>                    Depends: libtasn1-6-dev but it is not going to be installed
>                    Depends: libp11-kit-dev but it is not going to be installed
>                    Depends: libidn11-dev (>= 1.31) but it is not going to be installed
>  libsqlite3-dev : Depends: libsqlite3-0 (= 3.16.2-5+deb9u1) but 3.27.1-1 is to be installed
>  uuid-dev : Depends: libuuid1 (= 2.29.2-1+deb9u1) but 2.33.1-0.1 is to be installed
> E: Unable to correct problems, you have held broken packages.

MichaIng, 2019-02-20:
@rucknapucknavitz 
Ah just saw `Distro | buster (index=5)`, okay then is is expected since we install some quite specific library deps here.

I will check tomorrow which versions to pick on Buster. But no guarantee Home Assistant supports it already.

Could you please paste all your APT sources just to be sure nothing mixes up:
`cat /etc/apt/sources.list{,.d/*}`

rucknapucknavitz, 2019-02-20:
Of course. Thanks again for looking into this.

```
root@DietPi:~# cat /etc/apt/sources.list{,.d/*}
deb http://raspbian.raspberrypi.org/raspbian stretch main contrib non-free rpi
deb https://deb.debian.org/debian/ sid main
deb [arch=armhf] https://download.docker.com/linux/raspbian stretch edge
deb https://archive.raspberrypi.org/debian/ stretch main ui
root@DietPi:~#
```

MichaIng, 2019-02-20:
@rucknapucknavitz 
Hum, you have the Debian sid repo in place. Did you install WireGuard or where did this come from? Or did you somehow manually added this for a reason?

Actually the sid repo should only be used for the WireGuard packages in this case, based on preferences. Check:
`cat /etc/apt/preferences.d/*`

It looks like parts of your system are already on Buster/sid while others are not. This is a nasty situation. Check the above first and let me try to find the best solution.

Is this a moreless fresh install? Because such mixed repo system might be hard to resolve...

rucknapucknavitz, 2019-02-20:
Yes, I just transitioned over to WireGuard (from PiVPN). 

Here’s the output:

root@DietPi:~# cat /etc/apt/preferences.d/*
Package: docker-ce
Pin: version 5:18.09.0~3-0~raspbian-stretch
Pin-Priority: -1
Package: *
Pin: release n=sid
Pin-Priority: 99
root@DietPi:~#

This install is probably 4 months old and is basically used just for VPN and PiHole . I wonder if I should use another box for Home Assistant? Or might this become a frequent problem as more folks move towards WireGuard? I’m happy to continue troubleshooting it with you - I do appreciate the help!

MichaIng, 2019-02-20:
@rucknapucknavitz 
Actually it should work fine. The `sid` repo has pin priority `99` which means that from this repo no package should be installed if not requested manually, e.g. `apt install wireguard -t sid`

So I am wondering why other package were and are to be installed from the `sid` repo obviously. Lets check which `sid` were already installed: `apt list --installed | grep -v '/stable'`

rucknapucknavitz, 2019-02-20:
Here you go!
```

root@DietPi:~# apt list --installed | grep -v '/stable'

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Listing...
adduser/unstable,now 3.118 all [installed,automatic]
autoconf/unstable,now 2.69-11 all [installed]
automake/unstable,now 1:1.16.1-4 all [installed]
autotools-dev/unstable,now 20180224.1 all [installed,automatic]
base-files/unstable,now 10.1 armhf [installed,automatic]
base-passwd/unstable,now 3.5.46 armhf [installed,automatic]
bash-completion/unstable,now 1:2.8-5 all [installed]
bsdutils/unstable,now 1:2.33.1-0.1 armhf [installed,automatic]
bzip2/unstable,now 1.0.6-9 armhf [installed]
ca-certificates/unstable,now 20190110 all [installed]
cifs-utils/unstable,now 2:6.8-2 armhf [installed]
console-setup/unstable,now 1.189 all [installed]
console-setup-linux/unstable,now 1.189 all [installed,automatic]
coreutils/unstable,now 8.30-1 armhf [installed,automatic]
cpio/unstable,now 2.12+dfsg-6 armhf [installed,automatic]
cron/unstable,now 3.0pl1-130 armhf [installed]
debconf/unstable,now 1.5.70 all [installed]
debianutils/unstable,now 4.8.6.1 armhf [installed,automatic]
device-tree-compiler/unstable,now 1.4.7-3 armhf [installed,automatic]
dh-python/unstable,now 3.20180927 all [installed,automatic]
diffutils/unstable,now 1:3.7-2 armhf [installed,automatic]
distro-info-data/unstable,now 0.39 all [installed,automatic]
dmsetup/unstable,now 2:1.02.155-2 armhf [installed,automatic]
dns-root-data/unstable,now 2018091102 all [installed]
docker-ce/stretch,now 18.06.2~ce~3-0~raspbian armhf [installed]
dosfstools/unstable,now 4.1-2 armhf [installed]
dpkg-dev/unstable,now 1.19.4 all [installed,automatic]
ethtool/unstable,now 1:4.19-1 armhf [installed]
fail2ban/unstable,now 0.10.2-2.1 all [installed]
ffmpeg/now 3.2.4-1 armhf [installed,upgradable to: 7:3.2.12-1~deb9u1+rpt1]
findutils/unstable,now 4.6.0+git+20190209-2 armhf [installed,automatic]
fontconfig/unstable,now 2.13.1-2 armhf [installed,automatic]
fontconfig-config/unstable,now 2.13.1-2 all [installed,automatic]
gettext-base/unstable,now 0.19.8.1-9 armhf [installed,automatic]
gpgv/unstable,now 2.2.12-1 armhf [installed,automatic]
grep/unstable,now 3.3-1 armhf [installed]
gzip/unstable,now 1.9-3 armhf [installed,automatic]
hdparm/unstable,now 9.58+ds-1 armhf [installed]
hostname/unstable,now 3.21 armhf [installed,automatic]
ifupdown/unstable,now 0.8.35 armhf [installed,automatic]
init/unstable,now 1.56+nmu1 armhf [installed,automatic]
init-system-helpers/unstable,now 1.56+nmu1 all [installed,automatic]
initramfs-tools/unstable,now 0.133 all [installed,automatic]
initramfs-tools-core/unstable,now 0.133 all [installed,automatic]
install-info/unstable,now 6.5.0.dfsg.1-4+b1 armhf [installed,automatic]
iproute2/unstable,now 4.20.0-2 armhf [installed,automatic]
iputils-ping/unstable,now 3:20180629-2 armhf [installed]
iw/unstable,now 5.0.1-1 armhf [installed]
kbd/unstable,now 2.0.4-4 armhf [installed,automatic]
keyboard-configuration/unstable,now 1.189 all [installed,automatic]
keyutils/unstable,now 1.6-2 armhf [installed,automatic]
klibc-utils/unstable,now 2.0.6-1 armhf [installed,automatic]
kmod/unstable,now 26-1 armhf [installed]
libacl1/unstable,now 2.2.52-5 armhf [installed,automatic]
libapparmor1/unstable,now 2.13.2-7 armhf [installed,automatic]
libapt-inst2.0/unstable,now 1.8.0~rc3 armhf [installed,automatic]
libassuan0/unstable,now 2.5.2-1 armhf [installed,automatic]
libattr1/unstable,now 1:2.4.47-4 armhf [installed,automatic]
libaudit-common/unstable,now 1:2.8.4-2 all [installed,automatic]
libaudit1/unstable,now 1:2.8.4-2 armhf [installed,automatic]
libavahi-client3/unstable,now 0.7-4+b1 armhf [installed,automatic]
libavahi-common-data/unstable,now 0.7-4+b1 armhf [installed,automatic]
[... 162 lines cut ...]
rsync/unstable,now 3.1.3-5 armhf [installed]
rsyslog/unstable,now 8.40.0-1+b1 armhf [installed]
sed/unstable,now 4.7-1 armhf [installed,automatic]
sensible-utils/unstable,now 0.0.12 all [installed,automatic]
sqlite3/unstable,now 3.27.1-1 armhf [installed]
sudo/unstable,now 1.8.27-1 armhf [installed]
systemd-sysv/unstable,now 240-6 armhf [installed]
sysvinit-utils/unstable,now 2.93-8 armhf [installed,automatic]
tar/unstable,now 1.30+dfsg-5 armhf [installed]
tdb-tools/unstable,now 1.3.16-2+b1 armhf [installed,automatic]
tzdata/unstable,now 2018i-1 all [installed]
ucf/unstable,now 3.0038+nmu1 all [installed,automatic]
unzip/unstable,now 6.0-22 armhf [installed]
update-inetd/unstable,now 4.49 all [installed,automatic]
whiptail/unstable,now 0.52.20-8 armhf [installed]
wireguard/unstable,now 0.0.20190123-1 all [installed]
wireguard-dkms/unstable,now 0.0.20190123-1 all [installed,automatic]
wireguard-tools/unstable,now 0.0.20190123-1 armhf [installed,automatic]
wireless-tools/unstable,now 30~pre9-13 armhf [installed]
xkb-data/unstable,now 2.26-2 all [installed,automatic]
xz-utils/unstable,now 5.2.4-1 armhf [installed,automatic]
zlib1g/unstable,now 1:1.2.11.dfsg-1 armhf [installed,automatic]
zlib1g-devzlib1g-dev/unstable,now 1:1.2.11.dfsg-1 armhf [installed]

```

MichaIng, 2019-02-20:
I am still wondering how anything could have been installed from Debian `sid` repo on RPi, since the archive keyring is actually missing. I get:
```
2019-02-20 21:14:59 root@micha:/tmp# apt update
Hit:1 https://archive.raspberrypi.org/debian buster InRelease
Get:2 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster InRelease [15.0 kB]
Get:3 https://cdn-aws.deb.debian.org/debian sid InRelease [242 kB]
Err:3 https://cdn-aws.deb.debian.org/debian sid InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC
Get:4 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster/main armhf Packages [13.1 MB]
Reading package lists... Done
W: GPG error: https://cdn-aws.deb.debian.org/debian sid InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC
E: The repository 'https://deb.debian.org/debian sid InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

But we have so much users already reporting success with WireGuard on RPi 🤔.

After installing the keyring, as expected, no packages are pulled automatically. I also rechecked the dependencies and there is nothing required from `sid` repo aside `wireguard`, `wireguard-tools` and `wireguard-dkms`. `dkms` and `qrencode` are available in Raspbian Stretch with sufficient version...
_________
So in your case nearly the whole system is now on unstable... That is not ideal.
Do you have a backup in place from before you installed WireGuard?

MichaIng, 2019-02-20:
@Fourdee 
Do you have an RPi available to recheck WireGuard install, if there is some chance that other packages aside the WireGuard ones are pulled from Debian sid?
And that somehow WireGuard packages install successfully even that no `debian-archive-keyring` keyring is installed?

rucknapucknavitz, 2019-02-20:
> So in your case nearly the whole system is now on unstable... That is not ideal.

Nope! :-)
I wonder how that happened?!

> Do you have a backup in place from before you installed WireGuard?

I do, yes.

Are you interested in seeing something in there?
Or just whether I have the ability to restore to a pre-WireGuard state?
If the latter, I'm tempted to just _nuke it_ and reinstall from scratch, not too burdensome given it's just VPN and PiHole on this box...
Thanks!

rucknapucknavitz, 2019-02-20:
> Do you have an RPi available to recheck WireGuard install, if there is some chance that other packages aside the WireGuard ones are pulled from Debian sid?

Yes, I have another box I can test on - but not until tonight.

I'll load up a fresh image and only install WireGuard, and then report back the output from `apt list --installed | grep -v '/stable'`

Does that work?

MichaIng, 2019-02-20:
@rucknapucknavitz 
IMO restoring the backup or reinstalling from scratch is indeed the best solution then. This unstable + partly Debian system state will only cause more issues later, aside from Home Assistant failing to install.

Please do not install WireGuard. Let's test the install steps manually to check where the additional `sed` packages are installed.

___________

Ah `--allow-unauthenticated` allows to install from repo even without keyring. However this is suboptimal, renders the security idea behind the keyring obsolete. Will patch to install the keyring on Raspbian.

As well to be failsafe, will pin all `sid` packages `-1` and only the mentioned ones 99.

rucknapucknavitz, 2019-02-20:
> Please do not install WireGuard. Let's test the install steps manually to check where the additional `sed` packages are installed.

When I reinstall the image - but do not install WireGuard - what should I be looking for and how should I go about the manual test? Thanks!

MichaIng, 2019-02-20:
@rucknapucknavitz 
Go manually through these steps:
```
G_AGUG
G_AGI raspberrypi-bootloader raspberrypi-kernel libraspberrypi-bin libraspberrypi0 raspberrypi-kernel-headers
echo 'deb https://deb.debian.org/debian/ sid main' > /etc/apt/sources.list.d/dietpi-wireguard.list
echo -e 'Package: *\nPin: release n=sid\nPin-Priority: -1\n
Package: wireguard wireguard-dkms wireguard-tools\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
wget https://dietpi.com/downloads/binaries/rpi/debian-archive-keyring.deb
dpkg -i debian-archive-keyring.deb
rm debian-archive-keyring.deb
apt update
```
- If all this went fine so far, check the output of the last `apt update` command, if it does not complain about unauthenticated repo.
- It should as well report about packages that are to be updated. Also check this explicitly via: `apt list --upgradeable`
- There should be no packages listed for upgrade, since we already did this with the very first command.
- If so, then do `apt install wireguard`
- It should not install more then ~10 packages and most importantly it should not imply any package upgrades. Verify this before proceeding with `y`.

If everything went fine. Also check if the following does not list any upgrade, or only a very small number, e.g. max 5: `apt full-upgrade`

If everything went fine, please manually edit your `/DietPi/dietpi/dietpi-software` file and replace:
```
echo -e 'Package: *\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
```
with
```
echo -e 'Package: *\nPin: release n=sid\nPin-Priority: -1\n
Package: wireguard wireguard-dkms wireguard-tools\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
```

Then you can install WireGuard via `dietpi-software install 172` to do the configuration steps.

rucknapucknavitz, 2019-02-21:
Everything went fine until the `dietpi-software install 172`

Here's the full output:

```
root@DietPi:~# dietpi-software install 172
[  OK  ] DietPi-Software | RootFS R/W access verified.

[  OK  ] DietPi-Software | Initialized database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install

[  OK  ] DietPi-Software | Installing WireGuard: an extremely simple yet fast and modern VPN
[  OK  ] DietPi-Software | Free space check: path=/ | available=28012 MB | required=500 MB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Connection test: http://raspbian.raspberrypi.org/raspbian
[  OK  ] NTPD: time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask all: cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT

[  OK  ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian sid InRelease
Reading package lists...

[  OK  ] DietPi-Software | G_AGUP
[ INFO ] DietPi-Software | Running apt simulation to check for errors, please wait...
[  OK  ] DietPi-Software | APT installation for: bash-doc -s, please wait...
Inst bash-doc (4.4-5 Raspbian:stable [all])
Conf bash-doc (4.4-5 Raspbian:stable [all])

[  OK  ] DietPi-Software | G_AGI: bash-doc -s
[  OK  ] DietPi-Software | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-Software | G_AGUG

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing WireGuard: an extremely simple yet fast and modern VPN

[  OK  ] DietPi-Software | APT installation for: raspberrypi-bootloader raspberrypi-kernel libraspberrypi-bin libraspberrypi0 raspberrypi-kernel-headers, please wait...

[  OK  ] DietPi-Software | G_AGI: raspberrypi-bootloader raspberrypi-kernel libraspberrypi-bin libraspberrypi0 raspberrypi-kernel-headers
[  OK  ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian sid InRelease
Reading package lists...

[  OK  ] DietPi-Software | G_AGUP
[  OK  ] DietPi-Software | APT installation for: wireguard iptables qrencode, please wait...
E: Invalid record in the preferences file /etc/apt/preferences.d/dietpi-wireguard, no Package header

[FAILED] DietPi-Software | G_AGI: wireguard iptables qrencode
[  OK  ] DietPi-Software | APT installation for: wireguard iptables qrencode, please wait...
E: Invalid record in the preferences file /etc/apt/preferences.d/dietpi-wireguard, no Package header

[FAILED] DietPi-Software | G_AGI: wireguard iptables qrencode
[  OK  ] DietPi-Software | APT installation for: wireguard iptables qrencode, please wait...
E: Invalid record in the preferences file /etc/apt/preferences.d/dietpi-wireguard, no Package header

[FAILED] DietPi-Software | G_AGI: wireguard iptables qrencode
[  OK  ] DietPi-Software | APT installation for: wireguard iptables qrencode, please wait...
E: Invalid record in the preferences file /etc/apt/preferences.d/dietpi-wireguard, no Package header

[FAILED] DietPi-Software | G_AGI: wireguard iptables qrencode

--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Thu 21 Feb 01:12:23 GMT 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGI: wireguard iptables qrencode
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...

#### Expected behaviour:
<!-- What SHOULD be happening? -->
- ...

#### Actual behaviour:
<!-- What IS happening? -->
- ...

#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...

#### Additional logs:
```
Log file contents:
E: Invalid record in the preferences file /etc/apt/preferences.d/dietpi-wireguard, no Package header
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

root@DietPi:~#
```

MichaIng, 2019-02-23:
@rucknapucknavitz 
Looks like `/etc/apt/preferences.d/dietpi-wireguard` content has a syntax error. Please paste:
`cat /etc/apt/preferences.d/dietpi-wireguard`

rucknapucknavitz, 2019-02-25:
```
root@DietPi:~# cat /etc/apt/preferences.d/dietpi-wireguard
Package: *
Pin: release n=sid
Pin-Priority: -1

			Package: wireguard wireguard-dkms wireguard-tools
Pin: release n=sid
Pin-Priority: 99
```

MichaIng, 2019-02-25:
@rucknapucknavitz 
Ah copy&paste issue, please remove the free space in front of `Package: wireguard wireguard-dkms wireguard-tools` and try again.

rucknapucknavitz, 2019-02-25:
That's in `/DietPi/dietpi/dietpi-software` right?
I had tabbed that line over to match where `echo -e` above began.
You've probably figured out my linux knowledge is a bit limited... 
Thanks!

MichaIng, 2019-02-25:
@rucknapucknavitz 
Jep, ah right in `/DietPi/dietpi/dietpi-software` the second line need to start really at the beginning of the code line. Or merge both into one, which is properly failsafer:
```
echo -e 'Package: *\nPin: release n=sid\nPin-Priority: -1\n\nPackage: wireguard wireguard-dkms\n\nwireguard-tools\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
```

rucknapucknavitz, 2019-02-25:
I'm still getting the same output after `cat /etc/apt/preferences.d/dietpi-wireguard`

I've pasted a screen shot from my terminal, if that's helpful.

[image]

rucknapucknavitz, 2019-02-25:
I'm not sure where the typo is located, but I manually updated `/etc/apt/preferences.d/dietpi-wireguard` to delete the tabs before `Package: wireguard wireguard-dkms wireguard-tools` so that it aligned with the beginning of the line, and the install seemed to go fine (yay!).

But again, I'm not sure where to fix that above so that it's not an issue in the install.

rucknapucknavitz, 2019-02-25:
ok... I think there may have bee a typo in the original instructions that carried through in my notes. I nuked the install and started over re-copying the notes above.

In short, I followed these steps and it worked great! Thanks again.

```
G_AGUG
G_AGI raspberrypi-bootloader raspberrypi-kernel libraspberrypi-bin libraspberrypi0 raspberrypi-kernel-headers
echo 'deb https://deb.debian.org/debian/ sid main' > /etc/apt/sources.list.d/dietpi-wireguard.list
echo -e 'Package: *\nPin: release n=sid\nPin-Priority: -1\n\nPackage: wireguard wireguard-dkms wireguard-tools\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
wget https://dietpi.com/downloads/binaries/rpi/debian-archive-keyring.deb
dpkg -i debian-archive-keyring.deb
rm debian-archive-keyring.deb
apt update
apt list --upgradeable
apt install wireguard
apt full-upgrade
nano /DietPi/dietpi/dietpi-software
```
replace: `echo -e 'Package: *\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard`
with: `echo -e 'Package: *\nPin: release n=sid\nPin-Priority: -1\n\nPackage: wireguard wireguard-dkms wireguard-tools\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard`

Check: `cat /etc/apt/preferences.d/dietpi-wireguard` <-- now ok!
`dietpi-software install 172`

MichaIng, 2019-03-03:
🈯️ Reinstall test with code-side fix linked in OT

MichaIng, 2019-03-11:
I mark this as closed, Feel free to reopen is any questions/issues remain.

closed 2019-03-11 by MichaIng