MichaIng/DietPi #2569: bluetooth not work with dietpi-config, but work with Retropie setup

msongz, 2019-02-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee`
- Distro version | `stretch`
- Kernel version | `Linux pi3 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux`
- SBC device | `RPi 3 Model B+ (armv7l)`
- Power supply used | `5V 1A RAVpower`
- SDcard used | `SanDisk ultra`

#### Steps to reproduce

1. clean install, turn on bluetooth via **dietpi-config**, the bluetooth would not work
2. clean install, turn on bluetooth via **Retropie**, the bluetooth works!

#### Extra details

- test with a switch joy-con

MichaIng, 2019-02-21:
@msongz 
Thanks for your report.

What do you mean by not working? How did you test and where it fails?

Related code: https://github.com/Fourdee/DietPi/blob/dev/dietpi/func/dietpi-set_hardware#L1059-L1155
- We install the required APT packages, (re)enable the required kernel modules and enable+start the required services.

Do you face any error message when running: `/DietPi/dietpi/func/dietpi-set_hardware bluetooth 1`
I never used it, but read through some guides, e.g.: https://www.element14.com/community/docs/DOC-81266/l/setting-up-bluetooth-on-the-raspberry-pi-3
- Does `bluetoothctl` work?

msongz, 2019-02-23:
sorry , after digging the docs I find dietpi bluetoothctl is work better than retropie setup!
enable bluetooth in dietpi and config auto connect in retropie is best option

closed 2019-02-23 by msongz