MichaIng/DietPi #2573: network display error

msongz, 2019-02-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.21.1`
- Distro version | `9.8`
- Kernel version | `Linux pi3 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux`
- SBC device | `RPi 3 Model B+ (armv7l)`
- Power supply used | `5V 1A RAVpower`
- SDcard used | `SanDisk ultra`

#### Additional Information (if applicable)

[image]
[image]

#### Steps to reproduce

1. network connect both to eth0 and wlan0
2. unplug the eth0 and reboot
3. dietpi-config shows network connect and available and LCD screen shows eth0 ip.

MichaIng, 2019-02-21:
@msongz 
Thanks for your report.

You unplugged the Ethernet cable?? Then it is indeed strange. DietPi-Config checks `ip r` for an active route to the interface. This should not be possible without cable connected.

The DietPi banner uses the information file created by: https://github.com/Fourdee/DietPi/blob/dev/dietpi/func/obtain_network_details
It scans `ip -o a s eth0` for a valid IP address. This should as well not be possible in case of no connection 🤔.

Perhaps you can check the outputs manually:
```
ip a
ip r
ethtool eth0
```

msongz, 2019-02-21:
@MichaIng 
connect to eth0 and wlan0, after eth0 and wlan0 could be ping, I unplug the Ethernet cable, and cut off the power, then power on, twice, both times shows "eth0". 
and ssh via wifi, reboot, then shows wlan0

- ethtool eth0

```
Settings for eth0:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Advertised pause frame use: Symmetric
	Advertised auto-negotiation: Yes
	Speed: 10Mb/s
	Duplex: Half
	Port: MII
	PHYAD: 1
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbag
	Wake-on: g
	Current message level: 0x00000007 (7)
			       drv probe link
	Link detected: no
```

- ip a

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:03:6d:39 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:56:38:6c brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.173/24 brd 192.168.2.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:fe56:386c/64 scope link
       valid_lft forever preferred_lft forever
```

MichaIng, 2019-02-23:
@msongz 
So did I get it right that a second reboot solved it or does it already show up correctly when connecting via SSH?

Are your pasts now after correct interface shows up, or when wrong is shown (in parallel or still afterwards)?

> 2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:03:6d:39 brd ff:ff:ff:ff:ff:ff

With this, the login banner should definitely show the correct interface. The info is updated during boot scripts, so as long as on boot the cable it not connected, this should be correct.

As well it should be not possible that with above `ip a` output `ip r` contains any `eth0` entry.

Can you please verify, when plugging off the eth0 cable and after reboot:
- `ip a` does not show an inet/inet6 line on eth0 interface as above
- `ip r` does not show any line containing eth0
- `cat /DietPi/dietpi/.network` line 3 does not show eth0
- Then the banner as well as `dietpi-config` should show the correct interface and eth0 not connected.

If the `ip` commands above show result as expected, but `/DietPi/dietpi/.network` line 3 still shows eth0, then during boot `/DietPi/dietpi/func/obtain_network_details` must have failed. At least the banner should show wrong IP then. The same is the case when plugging of eth0 cable and checking banner without reboot (e.g. second SSH session). `/DietPi/dietpi/func/obtain_network_details` needs to be called to have the banner info updated. `dietpi-config` should immediately show correct connection state based on `ip r` 🤔.

msongz, 2019-02-24:
> So did I get it right that a second reboot solved it or does it already show up correctly when connecting via SSH?

no. the second reboot is also cut off the power. other things are when I unplug the eth0 cable, not yet reboot, I could not ssh to pi3 via wlan0. so I cut off the power and reboot.and shows eth0 IP, then I could ssh to pi3 via wlan0, but in short time, ssh connection will cut off. ping wlan0 timeout.

> Are your pastes now from after correct interface showed up, or when wrong is still shown?

the correct one

# wrong state output:

```
ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:03:6d:39 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.241/24 brd 192.168.2.255 scope global eth0
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:56:38:6c brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.173/24 brd 192.168.2.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:fe56:386c/64 scope link
       valid_lft forever preferred_lft forever
```

```
ip r
default via 192.168.2.1 dev wlan0
192.168.2.0/24 dev wlan0 proto kernel scope link src 192.168.2.173
192.168.2.0/24 dev eth0 proto kernel scope link src 192.168.2.241 linkdown
```

seems cut off the power will cause an unnormal state until a normal reboot.

MichaIng, 2019-02-24:
@msongz 
> 192.168.2.0/24 dev eth0 proto kernel scope link src 192.168.2.241 linkdown

Okay strange, the route is still active and IP assigned, but `linkdown` in `ip r` and `DOWN` in `ip a` indicates the wire is not connected. Internet access works fine then, thus successfully routed through the WLAN interface?

I will add the above strings checks to show correct connection state.

closed 2019-03-23 by MichaIng