MichaIng/DietPi #2575: hide mouse in Chromium autostart kiosk mode

Zixim, 2019-02-22:
#### Required Information
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee

cat /etc/debian_version
9.8

uname -a
Linux DietPi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
SBC device Rpi3

#### Additional Information (if applicable)
- Software title Chromium Autostart
- installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
The Pi has no attached keyb nor mouse.
as described here : https://github.com/Fourdee/DietPi/issues/2298#issuecomment-443406683
install unclutter
add line to chromium-autostart.sh , before xinit line:
```
sleep 10 && unclutter -idle 0.1 &

xinit $FP_CHROMIUM $CHROMIUM_OPTS

```
#### Expected behaviour
Chromium starts and displays page. 
Cursor disappears after approx 10 seconds.

#### Actual behaviour
Cursor never disappears
#### Extra details
Have also tried 
```
xinit $FP_CHROMIUM -nocursor $CHROMIUM_OPTS
```
No joy.

https://github.com/Fourdee/DietPi/issues/2298#issuecomment-443406683

MichaIng, 2019-02-24:
@Zixim 
Thanks for your report.

I remember this from the other topic, so indeed you tried out both method and non of them worked. I will check this out on x86 machine to verify it's either DietPi or RPi specific.

crawc, 2019-02-28:
This seems to do the trick using unclutter, I tested this on a Raspberry Pi 3B+.

/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
Change the last line to these 2 lines:
xinit $FP_CHROMIUM $CHROMIUM_OPTS &
sleep 1 && /usr/bin/unclutter -display :0 -idle 0.01 &

Zixim, 2019-03-01:
I admit that I never added the `display` parameter when trying out dietpi. I do have unclutter working on Raspbian boards without the parameter.

MichaIng, 2019-03-02:
@crawc 
Okay, so unclutter has to be started after `xinit`, this is good to know. I guess xinit has to fully finish, why the sleep is required? Depending on the system 1 second might be not sufficient.

I still wonder why `-nocursor` does not work which looks like a cleaner integrated attempt which should apply to the correct display directly and work without any random sleep. I will have a closer look into this.

crawc, 2019-03-11:
Sorry for the delayed replay.
Yes, without the `sleep` `unclutter` can't utilize `display :0` until `xinit` creates it. You are correct 1 second might not be enough time for other platforms. Maybe a better way would be have a loop to check if `xinit` is started then execute `unclutter`. I will see what I can come up with and post back.

MichaIng, 2019-03-12:
@crawc @Zixim 
Found the solution with `-nocursor`. It is indeed an `xinit`/`Xorg` option, not a `chromium` option. But it needs to be added after the xinit chromium call that everything is recognized correctly and after closing the chromium args via `--`.

So the command entry needs to be: `xinit $FP_CHROMIUM $CHROMIUM_OPTS -- -nocursor`
The cursor is still available. When moving it I see webpage elements highlighting, but it has no visible graphic.
___________
However the noclutter option has the advantage that the cursor can show up when moved.
Found some guides about how to add it as desktop autostart script:
- https://jackbarber.co.uk/blog/2017-02-16-hide-raspberry-pi-mouse-cursor-in-raspbian-kiosk
- https://www.raspberrypi.org/forums/viewtopic.php?p=405215

But this depends on LXDE desktop which is not always chosen, not must be any desktop chosen for chromium. So we need to find a way to add it as xinit/Xorg autostart script.

MichaIng, 2019-03-12:
Okay found the best two solution based on use case:
- If you still want to be able to use the mouse, only hide it when idle: `echo '/usr/bin/unclutter -idle 0.1 &' > /etc/chromium.d/dietpi-unclutter` (needs validation on RPi, since other Chromium package is used there)
- If you never want to use the mouse: `G_CONFIG_INJECT 'xinit[[:blank:]]' 'xinit $FP_CHROMIUM $CHROMIUM_OPTS -- -nocursor' /var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh`

crawc, 2019-03-12:
I like the idea of having options: No cursor or Hide cursor.

I just tried the `unclutter` option on the RPi3B+ and it did NOT hide the cursor.

MichaIng, 2019-03-12:
@crawc 
Thanks for testing on RPi. Does `/etc/chromium.d` even exist there? And if so are there example drop-ins inside? Perhaps this is handled differently by the `chormium-browser` package on RPi compared to the `chromium` package on Debian.

crawc, 2019-03-13:
Yes, the directory did exist with a single file `custom_flags` before I created the `dietpi-unclutter` file.
`root@DietPi:~# ls -ltrh /etc/chromium.d/
total 8.0K
-rw-r--r-- 1 root root 255 Feb 28 16:10 custom_flags
-rw-r--r-- 1 root root  44 Mar 12 20:10 dietpi-unclutter
`

MichaIng, 2019-03-13:
@crawc 
Okay, can you please paste the content of the file, just to be sure?
`cat /etc/chromium.d/custom_flags`

- The xserver is definitely initiated before it runs Chromium (inside), so `unclutter` should work without sleep. However you could try to add a `sleep 1 && ` to verify.
- You could as well check if `-display :0` is required, but actually when running inside an xserver session it should apply the correct display automatically 🤔.

crawc, 2019-03-13:
I tried with `-display :0` , `sleep 1 &&` , and the combination but its still showing the cursor. 

>root@DietPi:~# cat /etc/chromium.d/custom_flags
export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --test-type --no-sandbox --temp-profile --user-data-dir --disable-smooth-scrolling --disable-low-res-tiling --enable-low-end-device-mode --num-raster-threads=4 --profiler-timing=0 --disable-composited-antialiasing "

MichaIng, 2019-03-13:
@crawc 
Hmm the file is identical with the one from Debian. Pre shell script, so the unclutter command should work as desired.
However perhaps the files are not sourced. Which user do you use to login? If not root, could you try to login as root? Perhaps non-root users read/source only from their home dir.

crawc, 2019-03-14:
I'm using the root user.

MichaIng, 2019-03-14:
@MichaIng 
Ahh just found another config dir for the `chromium-browser` package on RPi: `/etc/chromium-browser/customizations`
Could you try to move the dietpi-unclutter inside there:
`mv /etc/chromium.d/dietpi-unclutter /etc/chromium-browser/customizations/99-dietpi-unclutter`
And to be sure revert to initial state:
`echo '/usr/bin/unclutter -idle 0.1 &' > /etc/chromium-browser/customizations/99-dietpi-unclutter`

crawc, 2019-03-18:
@MichaIng That did it on the RPi3B+, I will test on RPi1 and RPi2 as well! Nice find!

MichaIng, 2019-03-18:
@crawc 
Finally, great! Veery important to know that there are indeed such main differences between RPi and Debian Chromium packages.

Did `xinit $FP_CHROMIUM $CHROMIUM_OPTS -- -nocursor` btw work as well to hide the cursor permanently?

MichaIng, 2019-06-28:
Implementation planned with some additional options: https://github.com/MichaIng/DietPi/issues/2938

mrbluecoat, 2020-01-08:
@MichaIng, `xinit $FP_CHROMIUM $CHROMIUM_OPTS -- -nocursor` worked for me on a RPI-3B+

MichaIng, 2020-01-08:
@mrbluecoat 
Thanks for testing, I guess the following works as well, doesn't it?
`xinit -nocursor $FP_CHROMIUM $CHROMIUM_OPTS`
Its all about having -nocursor as xinit option and not as chromium binary option. -- can be used as syntax to end chromium arguments, but having xinit options in the front should work the same way.

mrbluecoat, 2020-01-08:
With your version I get `bad command line option "-nocursor"` and connection to X server is lost

MichaIng, 2020-01-08:
@mrbluecoat 
Thanks for testing.
Okay strange, then xinit requires some different syntax.

arahasya, 2020-10-19:
Hi,
for auto refresh of chromium every 20 seconds interval should I use 99-dietpi-autorefresh? or add the script code somewhere else?

Joulinar, 2020-10-19:
@arahasya 
I guess you raised some question on the forum https://dietpi.com/phpbb/viewtopic.php?f=9&t=8188

MDAR, 2022-12-22:
Have you found a solution for this?

This command works for me on an Odroid C4 with DietPi 8.12.1

```xinit /usr/bin/chromium --kiosk --window-size=1920,1080 --window-position=0,0  http://127.0.0.1:8080/basicui/app  -- -nocursor &```
(I run this from Node-RED, or versions of it with the URL set dynamically)

Which I guess translates to the autostart.sh file as

```exec "$STARTX" "$FP_CHROMIUM" $CHROMIUM_OPTS "${URL:-https://dietpi.com/}"  -- -nocursor &```

pkoevesdi, 2023-08-29:
I think this is the right syntax for "If you never want to use the mouse":
`exec "$STARTX" "$FP_CHROMIUM" $CHROMIUM_OPTS "${URL:-https://dietpi.com/}" -- -nocursor`
that has to got into `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh`. So just added an `-- -nocursor` _to the end_ of the last line.
Maybe somebody could edit the initial posting's solution to this? Because, what's now there simply doesn't work on the current (DietPi 8.20.1) version of `chromium-autostart.sh`.
Remember: this thread here has become part of the official documentation: https://dietpi.com/docs/software/desktop/#chromium Still took me half an hour to find the real correct solution.

MichaIng, 2023-09-01:
Ah right, many thanks. Indeed the config file location as well as the way the X server is invoked has changed. I'll adjust it this weekend.

anandv85, 2025-02-06:
> I think this is the right syntax for "If you never want to use the mouse": `exec "$STARTX" "$FP_CHROMIUM" $CHROMIUM_OPTS "${URL:-https://dietpi.com/}" -- -nocursor` that has to got into [...]

This solution is detailed enough and it worked fine for me.  I was looking all comments and only this specifies exactly which file to update and which line to update. Thanks!