MichaIng/DietPi #2581: Unable to re-install pihole

1985kasper, 2019-02-23:
#### Details:
- Date           | Sat 23 Feb 20:39:27 GMT 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | Unknown Device (armv7l) (index=0)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | buster (index=5)
- Command        | G_AGI: sqlite3
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Uninstall pi-hole through dietpi-software
2. keep the dependancies when asked/prompted

#### Expected behaviour:

- Should be able to install pihole again through dietpi-software

#### Actual behaviour:

- Error related to sqlite3 but the package itself is not installed.

#### Extra details:

- wasn't able to access pihole web interface. repair didn't work so i decided to re-install pihole. Stuck at this step. Manual installation of pihole fails as well.

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2019-02-23:
@1985kasper 
Thanks for your report.

Somehow several system packages have been installed from Debian "sid" repo. I guess you installed WireGuard?
Check out a very similar issue: https://github.com/Fourdee/DietPi/issues/2568

Do you have a backup from before WireGuard install? At least please to the following to deny any other package installs from sid repo:
```
wget https://dietpi.com/downloads/binaries/rpi/debian-archive-keyring.deb
dpkg -i debian-archive-keyring.deb
rm debian-archive-keyring.deb
echo -e 'Package: *\nPin: release n=sid\nPin-Priority: -1\n
Package: wireguard wireguard-dkms wireguard-tools\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
```

Then check if `apt install sqlite3` works.

1985kasper, 2019-02-23:
I did install wireguard but used the dietpi-software method. Unfortunately no backup from before the wireguard installation. 

Tried the steps you provided just having some issues with the last bit:

@DietPi:/etc/network $ echo -e 'Package: *\nPin: release n=sid\nPin-Priority: -1\n
> Package: wireguard wireguard-dkms wireguard-tools\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
bash: /etc/apt/preferences.d/dietpi-wireguard: Permission denied
@DietPi:/etc/network $

MichaIng, 2019-02-23:
@1985kasper 
Are you logged in as root user? In case run the echo command inside `sudo` shell:
```
sudo -s
echo -e 'Package: *\nPin: release n=sid\nPin-Priority: -1\n\nPackage: wireguard wireguard-dkms wireguard-tools\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
exit
```

1985kasper, 2019-02-23:
Afraid i did something very stupid and apt-get got removed. So starting with a fresh install. But i should hold off on installing wireguard again until you fix this? 
Thanks for the help!

MichaIng, 2019-02-23:
@1985kasper 
Okay, perhaps not the worst thing with a half-sid system.

Jep before installing WireGuard then, please go through these steps to avoid accidental "sid" upgrades this time: https://github.com/Fourdee/DietPi/issues/2568#issuecomment-465801893

1985kasper, 2019-02-24:
@MichaIng Thanks!

I am aborting since it's trying to install more then the ~10 packages you described. 

Let me know if you need more testing to be done etc. I am happy to help since Wireguard is really an excellent way for me to be able to access my home LAN

root@DietPi:~# apt list --upgradable 
Listing... Done
root@DietPi:~# apt install wireguard
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  binutils cpp cpp-6 dkms gcc gcc-6 libasan3 libatomic1 libcc1-0 libgcc-6-dev libgomp1 libisl15 libmpc3 libmpfr4 libubsan0 make patch
  wireguard-dkms wireguard-tools
Suggested packages:
  binutils-doc cpp-doc gcc-6-locales python3-apport menu gcc-multilib manpages-dev autoconf automake libtool flex bison gdb gcc-doc
  gcc-6-doc libgcc1-dbg libgomp1-dbg libitm1-dbg libatomic1-dbg libasan3-dbg liblsan0-dbg libtsan0-dbg libubsan0-dbg libcilkrts5-dbg
  libmpx2-dbg libquadmath0-dbg make-doc ed diffutils-doc
Recommended packages:
  fakeroot lsb-release libc6-dev | libc-dev libc6-dev
The following NEW packages will be installed:
  binutils cpp cpp-6 dkms gcc gcc-6 libasan3 libatomic1 libcc1-0 libgcc-6-dev libgomp1 libisl15 libmpc3 libmpfr4 libubsan0 make patch
  wireguard wireguard-dkms wireguard-tools
0 upgraded, 20 newly installed, 0 to remove and 0 not upgraded.
Need to get 17.0 MB of archives.
After this operation, 68.7 MB of additional disk space will be used.
Do you want to continue? [Y/n]

MichaIng, 2019-02-24:
@1985kasper 
Just checked the list, and actually this is fine, indeed typical build tools and libraries required to build the WireGuard kernel module. Most importantly no packages are updated.

Please accept these installs and go on with the steps.

1985kasper, 2019-02-25:
@MichaIng it worked. I was able to install wireguard and the system didn't move to sid.

However i made a mistake with the wireguard config. I uninstalled it through the dietpi-software. Now i tried to install it again and again the system is trying to upgrade to unstable packages. 
So this is still very much broken despite all the changes made.

I hope there is a way to revert this. If so please help me with this. I did a fresh install the other day but really do not feel eager to go through that again. 

kasper@DietPi:~ $ sudo apt-get upgrade --dry-run
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libustr-1.0-1
Use 'sudo apt autoremove' to remove it.
The following packages have been kept back:
  apt apt-transport-https apt-utils avahi-utils bash bc bind9-host curl dash dialog dirmngr dnsutils dpkg e2fslibs e2fsprogs
  git git-man gnupg gnupg-agent htop idn2 iptables isc-dhcp-client libapt-pkg5.0 libcomerr2 libcurl3-gnutls libdrm2 libedit2
  libgcc1 libgnutls30 libgssapi-krb5-2 libidn2-0 libip4tc0 libip6tc0 libiptc0 libk5crypto3 libkrb5-3 libkrb5support0
  libldap-2.4-2 liblocale-gettext-perl libncurses5 libncursesw5 libpam-google-authenticator libparted2 libpsl5
  libpython3-stdlib libreadline7 libss2 libstdc++6 libsystemd0 libtext-charwidth-perl libtiff5 libtinfo5 libudev1 libxml2
  libxtables12 login mount mtr-tiny nano ncurses-base ncurses-bin openssh-client openssh-server openssh-sftp-server parted perl
  perl-base php-apcu pinentry-curses procps psmisc python3 python3-minimal python3-systemd qrencode systemd udev usbutils
  util-linux wget wpasupplicant
The following packages will be upgraded:
  adduser avahi-daemon base-files base-passwd bash-completion bsdutils bzip2 ca-certificates console-setup console-setup-linux
  coreutils cpio cron dbus debconf debianutils device-tree-compiler dh-python diffutils dmsetup dns-root-data dosfstools
  ethtool fail2ban findutils fontconfig-config gettext-base gpgv grep gzip hdparm hostname ifupdown init init-system-helpers
  initramfs-tools initramfs-tools-core install-info iperf iproute2 iputils-ping iw kbd keyboard-configuration klibc-utils kmod
  libacl1 libapparmor1 libapt-inst2.0 libassuan0 libattr1 libaudit-common libaudit1 libavahi-client3 libavahi-common-data
  libavahi-common3 libavahi-core7 libblkid1 libbsd0 libbz2-1.0 libc-bin libc-l10n libc6 libcap-ng0 libcap2 libcap2-bin
  libdaemon0 libdb5.3 libdbus-1-3 libdebconfclient0 libdevmapper1.02.1 libelf1 liberror-perl libexpat1 libfam0 libfdisk1
  libffi6 libfontconfig1 libfreetype6 libgcrypt20 libgd3 libgeoip1 libgmp10 libgpg-error0 libhogweed4 libidn11 libiw30 libjbig0
  libjpeg62-turbo libkeyutils1 libklibc libkmod2 libldap-common libltdl7 liblz4-1 liblzma5 libmagic-mgc libmagic1 libmount1
  libmpdec2 libnetfilter-conntrack3 libnettle6 libnewt0.52 libnfnetlink0 libnghttp2-14 libnih-dbus1 libnih1 libnl-3-200
  libnl-genl-3-200 libnpth0 libp11-kit0 libpam-modules libpam-modules-bin libpam-runtime libpam0g libpcap0.8 libpcre3
  libpcsclite1 libplymouth4 libpng16-16 libpopt0 librtmp1 libsasl2-2 libsasl2-modules-db libseccomp2 libselinux1
  libsemanage-common libsemanage1 libsepol1 libslang2 libsmartcols1 libsqlite3-0 libssh2-1 libssl1.0.2 libssl1.1 libtasn1-6
  libusb-1.0-0 libuuid1 libwebp6 libwrap0 libx11-6 libx11-data libxau6 libxcb1 libxslt1.1 libzip4 lighttpd locales lsb-base
  lsof makedev mawk mime-support multiarch-support netcat netcat-traditional openssl p7zip p7zip-full passwd php-common
  plymouth readline-common rfkill sed sensible-utils sqlite3 sudo systemd-sysv sysvinit-utils tar tcpdump tzdata ucf ufw unzip
  vlan whiptail wireless-tools xkb-data xz-utils zlib1g
182 upgraded, 0 newly installed, 0 to remove and 82 not upgraded.
Inst base-files [9.9+rpi1+deb9u8] (10.1 Debian:unstable [armhf])
Conf base-files (10.1 Debian:unstable [armhf])
Inst libc-l10n [2.24-11+deb9u4] (2.28-7 Debian:unstable [all])
Inst locales [2.24-11+deb9u4] (2.28-7 Debian:unstable [all]) []
Inst libnih-dbus1 [1.0.3-8] (1.0.3-10+b2 Debian:unstable [armhf]) []
Inst libnih1 [1.0.3-8] (1.0.3-10+b2 Debian:unstable [armhf]) []
Inst libc6 [2.24-11+deb9u4] (2.28-7 Debian:unstable [armhf]) [libc-bin:armhf ]
Conf libc6 (2.28-7 Debian:unstable [armhf]) [libc-bin:armhf ]
Inst libc-bin [2.24-11+deb9u4] (2.28-7 Debian:unstable [armhf])
Conf libc-bin (2.28-7 Debian:unstable [armhf])
Inst debconf [1.5.61] (1.5.70 Debian:unstable [all])
Conf debconf (1.5.70 Debian:unstable [all])
Inst libcap-ng0 [0.7.7-3] (0.7.9-2 Debian:unstable [armhf])
Conf libcap-ng0 (0.7.9-2 Debian:unstable [armhf])
Inst libaudit-common [1:2.6.7-2] (1:2.8.4-2 Debian:unstable [all])
Conf libaudit-common (1:2.8.4-2 Debian:unstable [all])
Inst libaudit1 [1:2.6.7-2] (1:2.8.4-2 Debian:unstable [armhf])
Conf libaudit1 (1:2.8.4-2 Debian:unstable [armhf])
Inst libpam0g [1.1.8-3.6+rpi1] (1.3.1-5 Debian:unstable [armhf])
Conf libpam0g (1.3.1-5 Debian:unstable [armhf])
Inst libpcre3 [2:8.39-3] (2:8.39-11 Debian:unstable [armhf])
Conf libpcre3 (2:8.39-11 Debian:unstable [armhf])
Inst libselinux1 [2.6-3] (2.8-1+b1 Debian:unstable [armhf])
Conf libselinux1 (2.8-1+b1 Debian:unstable [armhf])
Inst libsepol1 [2.6-2] (2.8-1 Debian:unstable [armhf])
Conf libsepol1 (2.8-1 Debian:unstable [armhf])
Inst libsemanage-common [2.6-2] (2.8-2 Debian:unstable [all]) [libsemanage1:armhf ]
[... 312 lines cut ...]
[... cut ...]
Conf p7zip-full (16.02+dfsg-6 Debian:unstable [armhf])
Conf p7zip (16.02+dfsg-6 Debian:unstable [armhf])
Conf php-common (2:69 Debian:unstable [all])
Conf plymouth (0.9.4-1 Debian:unstable [armhf])
Conf rfkill (2.33.1-0.1 Debian:unstable [armhf])
Conf sudo (1.8.27-1 Debian:unstable [armhf])
Conf tcpdump (4.9.2-3 Debian:unstable [armhf])
Conf ufw (0.36-1 Debian:unstable [all])
Conf unzip (6.0-22 Debian:unstable [armhf])
Conf vlan (2.0.5 Debian:unstable [all])
Conf wireless-tools (30~pre9-13 Debian:unstable [armhf])

MichaIng, 2019-02-26:
@1985kasper 
Please go again through the steps summed up well here: https://github.com/MichaIng/DietPi/issues/2568#issuecomment-466868405

MichaIng, 2019-03-05:
I mark this as closed, please use the other open issue for further discussion etc: https://github.com/MichaIng/DietPi/issues/2568

closed 2019-03-05 by MichaIng