MichaIng/DietPi #2583: VLAN support

1985kasper, 2019-02-23:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- I use this raspberry to use airplay over multiple vlans. vlans are a great way to segregate IoT devices for example.

#### Describe the solution you'd like:

- I can mannually add a vlan. However with every update or change of the networking etc. dietpi overrules the vlan. It would be great to have vlan support natively in the network settings.
Something very simple like adding interface eth0.20 would do.

#### Describe alternatives you've considered:

- ...

#### Additional context

- ...

MichaIng, 2019-02-23:
@1985kasper 
I have no idea about VLAN actually, since never used it. Could you give an example about how your interfaces file looks like then?

Would it be possible to simply store them inside `/etc/network/interfaces.d` as drop-in config or do they need to be directly besides/within the `eth0` section, as of your example?

1985kasper, 2019-02-24:
The thing is the way i always did it by installing vlan package like here;
https://blog.sleeplessbeastie.eu/2012/12/23/debian-how-to-create-vlan-interface/
https://wiki.debian.org/NetworkConfiguration#Howto_use_vlan_.28dot1q.2C_802.1q.2C_trunk.29_.28Etch.2C_Lenny.29
is being deprecated by debian (no idea how long it will take) and i am seeing messages about it moving to ip command.
https://wiki.archlinux.org/index.php/VLAN
This is the method i use for my use case which is having a airplay receiver on IoT vlan and spotify devices on another vlan:
https://chrisreinking.com/need-bonjour-across-vlans-set-up-an-avahi-gateway/

BUT multiple vlan support would also be great since pihole is able to "serve" multiple interfaces with a little workaround. 
Vlans are handy for lots of things.

MichaIng, 2019-02-24:
@1985kasper 
Reading: https://manpages.debian.org/stretch/vlan/vlan-interfaces.5.en.html
Yeah an interesting feature definitely that could be added natively. We should add `source /etc/network/interfaces.d/*` to the generated `/etc/network/interfaces` file as well to allow drop-in config config files. This is easy to apply now, so with v6.22 you can add your VLANs as drop-in configs.

> is being deprecated by debian
Jep due to `net-tools` (ifconfig, commands being used, which are deprecated an replaced by `iproute2` command set.

1985kasper, 2019-02-24:
@MichaIng Thanks! that would be awesome!

zxp, 2019-03-09:
please enable these kernel options and release a new version:
CONFIG_VLAN_8021Q=m
CONFIG_VLAN_8021Q_GVRP=m
CONFIG_VLAN_8021Q_MVRP=m

MichaIng, 2019-03-09:
@zxp 
I think in kernel this should be enabled, it is just about the possibility to add those from DietPi-Config menu.

Or do you run a DietPi image where the kernel options are missing? Which device do you use in case?
Generally we do no custom kernel coding/creation, so this would be something to forward to the base image devs.

zxp, 2019-03-11:
I'm sure the NanoPi NEO2, K1 Plus and SoPine A64 kernel image don't enabled 8021q support. @MichaIng

MichaIng, 2019-03-12:
@zxp 
Ah okay thanks for the hint in case. So when adding this feature, we must check for kernel support first on the particular device.
However setting the kernel parameters is out of our tasks currently, so this is something to forward to ARMbian (in case of Pine64 images) respectively FriendlyARM for NanoPi.

zxp, 2019-03-13:
but the official Armbian image for that device indeed has 8021q vlan module support. @MichaIng

MichaIng, 2019-03-17:
This has been added to our FeatHub page: https://feathub.com/MichaIng/DietPi/+42

1985kasper, 2019-03-24:
Any news on this? Just lost all VLANs i had setup again on the dietpi box because of the most recent update. 
It would be at least nice to respect manually setup VLANS when dietpi is updated.

MichaIng, 2019-03-24:
@1985kasper 
Since VLAN implementation itself will take a while (and testing), for now I will implement drop-in config support. So you can place any any adjustments/additions into `/etc/network/interfaces.d/` to survive `dietpi-config` touches.

Hmm, not sure why the update removed your VLANs actually, since it does not touch interfaces... Only on first boot and when running `dietpi-config` manually this is the case (or should be).
Could you paste your update log: `cat /var/tmp/dietpi/logs/dietpi-update.log`

zxp, 2019-03-31:
Because there is no 8021q vlan module support in some hardware platform DietPi kernel...that's why I commented at here to request add some configs into kernel config. see #issuecomment-471163150

MichaIng, 2019-03-31:
@zxp 
Note that there is no "DietPi" kernel as we do not compile own kernels. Pine64 images have been switched to ARMbian so these should have VLAN enabled. In case of NanoPi we anyway have to switch base-image AFAIK since FriendlyARM does not provide Debian images anymore.

MichaIng, 2019-05-06:
@1985kasper 
Btw. I made a start and added `/etc/network/interfaces.d/` drop-in config support: https://github.com/MichaIng/DietPi/commit/aa4a36fd890a0774d68fb9e7118e72af042ac8d4#diff-0e127c4cad3f8401a925d13d05b5ae87
So you can add your custom VLAN interfaces as separate config file to not be overwritten by dietpi-config.

I will slowly migrate dietpi-config to use this for own interface configs as well. This allows a more flexible and less intrusive network adapter/interface handling and is a good basis to add more features like VLAN etc.

Alestrix, 2020-05-19:
This will come in handy for PiHole when serving more than one network (e.g. guest and main).

hllhll, 2024-06-23:
> This will come in handy for PiHole when serving more than one network (e.g. guest and main).

Did you figured out an alternative solution that would not collide with dietpi-networking stuff?