MichaIng/DietPi #2584: Kodi video playback on Odroid XU4

autur, 2019-02-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.21.1 
- Distro version | Stretch
- Kernel version | Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux
- SBC device | Odroid XU4 (armv71)
- Power supply used | 5V/6A Power Supply Unit 
- SDcard used | eMMC 16GB

#### Additional Information (if applicable)
- Software title | Kodi
- Was the software title installed freshly or updated/migrated? fresh
- Can this issue be replicated on a fresh installation of DietPi? yes
- 4e8f4e97-3a3c-4d2f-832c-4d961bb3288e ID 

#### Steps to reproduce
1. Install Kodi
2. Attempt playback of any video file or stream on Kodi

#### Expected behaviour
Video should be playing smoothly with audio. 

#### Actual behaviour
Video starts playing for a few seconds, then stutters and freezes, but the file is still being processed (the timer continues). No audio is heard.

#### Extra details
Tested on Vizio P55-F1 (TV) via HDMI
Tested multiple video file formats and codecs, all with the same result
Tested video stream from CBC News video add-on
Tested files on NTFS and EXT4-formatted drives, both microSD and external USB drives
Uninstalled Kodi and re-installed with no change.

Kodi video playback problems on XU4 have been reported on the forum by multiple users with no resolutions:
https://dietpi.com/phpbb/viewtopic.php?t=3916
https://dietpi.com/phpbb/viewtopic.php?f=11&t=2920
https://dietpi.com/phpbb/viewtopic.php?f=11&t=5539

MichaIng, 2019-02-24:
@autur 
Thanks for your report.

Just to assure: You installed Kodi from `dietpi-software`, right?

When starting Kodi from console and closing again after running into this issue, are there any error outputs that might help?
I will also have a look into your bug report.

autur, 2019-02-24:
Yes, installed Kodi via dietpi-software. And no, I don't get any error outputs, unfortunately (I imagine that's the case because the timer continues, even when the video freezes or stutters, so something is still being processed).

MichaIng, 2019-02-26:
@autur 
Your `/var/log/Xorg.0.log` is full of:
```
[   165.797] (WW) ARMSOC(0): flip queue failed: Device or resource busy
[   165.797] (EE) ARMSOC(0): ERROR: swap 13465 ARMSOC_SWAP_FAIL on swap complete
[   165.797] (EE) ARMSOC(0): [DRI2] DRI2SwapBuffers: driver failed to schedule swap
[   165.803] (WW) ARMSOC(0): WARNING: Flip is called too fast
[   165.803] (WW) ARMSOC(0): WARNING: Flip isn't in order
```

I just see there is a very tiny swap file on your system (~50M), which is expected on fresh DietPi on 2G RAM devices, but might be not sufficient for Kodi + other processes... At least worth to check memory consumption (`free -m`) and in case increase the swap file to e.g. 2G via `dietpi-config` or `dietpi-drive_manager`.

... but actually that errors are definitely not about the swap file 🤣.

What I stumbled over:
```
[    56.694] (EE) AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/armsoc_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/armsoc_dri.so: cannot open shared object file: No such file or directory)
[    56.694] (EE) AIGLX: reverting to software rendering
[    56.755] (II) IGLX: enabled GLX_MESA_copy_sub_buffer
```
- Searched for this missing driver file: https://forum.odroid.com/viewtopic.php?f=79&t=5078
- Remember that Odroid actually runs the mali driver...
- Could you please go through the linked thread and check if this doubled libraries apply in your case as well? And in case try to move the mesa libs out of place + `ldconfig` as stated in the thread?
- I am not sure, have to check, but I think we install mesa libs with X.org, Properly this should be skipped on Odroids which come with mali. On the other hand I find many guides that install both...

autur, 2019-02-26:
Thanks for investigating. I did find duplicates of the libEGL files in /usr/lib/arm-linux-gnueabihf/ and moved them to a different directory, then ran `ldconfig` and restarted. However, video playback on Kodi has not changed--videos still freeze.

MichaIng, 2019-02-26:
@autur 
Do you actually start Kodi from desktop, via `dietpi-autostart` choice or from command line? If from command line, with which command?

autur, 2019-02-26:
From the command line using `startkodi`

MichaIng, 2019-02-26:
@autur 
Okay it must be something with the Xorg/mali/mesa...
Can you please paste: `cat /etc/X11/xorg.conf`
Run:
```
apt-get dist-upgrade # Assure everything is up-to-date
apt-get install --reinstall xf86-video-armsoc-odroid
```

And then again paste: `cat /etc/X11/xorg.conf`

autur, 2019-02-26:
Here is the before and after output requested:

**BEFORE REINSTALL----------------------------------------------------------------------------------**
```
root@DietPi:~# cat /etc/X11/xorg.conf
Section "Device"

        Identifier      "Mali-Fbdev"
        Driver          "armsoc"
        Option          "fbdev"                 "/dev/fb0"
        Option          "Debug"                 "false"
        Option          "DPMS"                  "false"
        Option          "Fimg2DExa"             "true"
        Option          "DRI2"                  "true"
        Option          "DRI2_PAGE_FLIP"        "false"
#       Option          "Fimg2DExaSolid"        "false"
#       Option          "Fimg2DExaCopy"         "false"
        Option          "Fimg2DExaComposite"    "true"
    Option          "SWcursorLCD"           "false"

EndSection

Section "Screen"

        Identifier   "Default Screen"
        Device       "Mali-Fbdev"
        DefaultDepth 24

EndSection

Section "DRI"

        Mode 0666

EndSection
```

**AFTER REINSTALL---------------------------------------------------------------------------------**
```
root@DietPi:~# cat /etc/X11/xorg.conf
Section "Device"

        Identifier      "Mali-Fbdev"
        Driver          "armsoc"
        Option          "NoG2D"                 "false"
        Option          "NoHardwareMouse"       "true"
        Option          "NoFlip"                "true"
        Option          "fbdev"                 "/dev/fb0"
        Option          "Debug"                 "false"
        Option          "DPMS"                  "false"
        Option          "Fimg2DExa"             "true"
        Option          "DRI2"                  "true"
        Option          "DRI2_PAGE_FLIP"        "false"
#       Option          "Fimg2DExaSolid"        "false"
#       Option          "Fimg2DExaCopy"         "false"
        Option          "Fimg2DExaComposite"    "true"
    Option          "SWcursorLCD"           "false"

EndSection

Section "Screen"

        Identifier   "Default Screen"
        Device       "Mali-Fbdev"
        DefaultDepth 24

EndSection

Section "DRI"

        Mode 0666

EndSection
```

MichaIng, 2019-02-26:
@autur 
Okay nice that matches to what I read on Meverics repo/image thread what should it be with the new driver: https://forum.odroid.com/viewtopic.php?f=96&t=30552&start=50

`Option          "NoFlip"                "true"` looks like it should solve the error outputs I found in your X.org log.

`Option          "NoHardwareMouse"       "true"` not sure if this is wanted or what it means according to the mouse.

Please test Kodi now with this new xorg.conf. If it runs smoother now, we need to update the `xorg.conf` which `dietpi-software` places accordingly: https://github.com/MichaIng/DietPi/blob/dev/.conf/dps_6/xorg_xu4.conf
- [ ] Also requires testing then with Desktop, at best all desktops we offer.

MichaIng, 2019-03-02:
Possibly related: https://github.com/MichaIng/DietPi/issues/2351

Fourdee, 2019-03-03:
Commit to resolve: https://github.com/MichaIng/DietPi/commit/8633f623320e1f32c03e924ecf96811a8c6770be
Requires testing to verify.

Fourdee, 2019-03-03:
Testing:
- Kodi launches 🈯️ 
- Video playback

MichaIng, 2019-03-03:
@Fourdee 
Thanks for this. How does the fresh xorg.conf looks like?
With existing xorg.conf the package obviously only adds:
```
        Option          "NoG2D"                 "false"
        Option          "NoHardwareMouse"       "true"
        Option          "NoFlip"                "true"
```

Fourdee, 2019-03-04:
@MichaIng 

Looks good 👍 
```
root@DietPi:~# cat /etc/X11/xorg.conf
# X.Org X server configuration file for xf86-video-armsoc-odroid

Section "Device"
        Identifier      "Mali-Fbdev"
        Driver          "armsoc"
        Option          "fbdev"                 "/dev/fb0"
        Option          "Debug"                 "false"
        Option          "DPMS"                  "false"
        Option          "NoFlip"                "true"
        Option          "NoHardwareMouse"       "true"
        Option          "NoG2D"                 "false"
EndSection

Section "Screen"
        Identifier      "Default Screen"
        Device          "Mali-Fbdev"
        DefaultDepth    24
EndSection

Section "DRI"
        Mode 0666
EndSection

Section "Monitor"
        Identifier      "Generic Monitor"
        Option          "DPMS"                  "Off"
EndSection

## disable automatic HDMI turn off
Section "ServerLayout"
        Identifier      "DefaultLayout"
        Option          "BlankTime"             "0"
        Option          "StandbyTime"           "0"
        Option          "SuspendTime"           "0"
        Option          "OffTime"               "0"
EndSection
```

autur, 2019-03-05:
@MichaIng
Apologies for the late reply! I wanted to try a different monitor+cable to be absolutely sure those weren't the issue and had to wait for a shipment. 

To pick up, where I left off a week ago, after reinstalling (`--reinstall xf86-video-armsoc-odroid`) playback still freezes--there are no changed at all. I hope we can continue to investigate, and I'll be sure to reply within hours.

MichaIng, 2019-03-05:
@autur 
Can you please try to use the very default xorg.conf, just in case:
`wget https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_6/xorg_xu4.conf -O /etc/X11/xorg.conf`

And before starting Kodi, please clear the X.org log:
`> /var/log/Xorg.0.log`

When it freezes, paste
`cat /var/log/Xorg.0.log`

autur, 2019-03-05:
Here is the output. Note that `cat /var/log/Xorg.0.log` was run after exiting kodi.

```
root@DietPi:~# cat /var/log/Xorg.0.log
[   168.097]
X.Org X Server 1.19.2
Release Date: 2017-03-02
[   168.097] X Protocol Version 11, Revision 0
[   168.097] Build Operating System: Linux 4.9.0-8-armmp-lpae armv7l Debian
[   168.097] Current Operating System: Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l
[   168.097] Kernel command line: console=tty1 console=ttySAC2,115200n8 root=UUID=3032ee2a-4efc-40b6-9c1f-2839a840144d rootwait ro fsck.repair=yes net.ifnames=0 drm_kms_helper.edid_firmware=edid/1920x1080.bin smsc95xx.macaddr=00:1e:06:61:7a:72 governor=performance hdmi_tx_amp_lvl=31 hdmi_tx_lvl_ch0=3 hdmi_tx_lvl_ch1=3 hdmi_tx_lvl_ch2=3 hdmi_tx_emp_lvl=6 hdmi_clk_amp_lvl=31 hdmi_tx_res=0 HPD=true vout=hdmi  s5p_mfc.mem=16M
[   168.097] Build Date: 03 November 2018  05:53:43AM
[   168.097] xorg-server 2:1.19.2-1+deb9u5 (https://www.debian.org/support)
[   168.097] Current version of pixman: 0.34.0
[   168.097]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   168.097] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   168.098] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Mar  5 15:42:58 2019
[   168.100] (==) Using config file: "/etc/X11/xorg.conf"
[   168.100] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   168.100] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   168.101] (==) ServerLayout "ServerLayout0"
[   168.101] (**) |-->Screen "Default Screen" (0)
[   168.101] (**) |   |-->Monitor "<default monitor>"
[   168.102] (**) |   |-->Device "Mali-Fbdev"
[   168.102] (==) No monitor specified for screen "Default Screen".
        Using a default monitor configuration.
[   168.102] (**) Option "BlankTime" "0"
[   168.102] (**) Option "StandbyTime" "0"
[   168.102] (**) Option "SuspendTime" "0"
[   168.102] (**) Option "OffTime" "0"
[   168.102] (==) Automatically adding devices
[   168.102] (==) Automatically enabling devices
[   168.102] (==) Automatically adding GPU devices
[   168.103] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   168.103] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   168.103]    Entry deleted from font path.
[   168.103] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   168.103]    Entry deleted from font path.
[   168.103] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   168.103]    Entry deleted from font path.
[   168.103] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   168.103]    Entry deleted from font path.
[   168.103] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   168.103]    Entry deleted from font path.
[   168.103] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   168.103]    Entry deleted from font path.
[   168.103] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[   168.103] (==) ModulePath set to "/usr/lib/xorg/modules"
[   168.103] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[   168.103] (II) Loader magic: 0x65ff60
[   168.103] (II) Module ABI versions:
[   168.103]    X.Org ANSI C Emulation: 0.4
[   168.104]    X.Org Video Driver: 23.0
[   168.104]    X.Org XInput driver : 24.1
[   168.104]    X.Org Server Extension : 10.0
[   168.105] (--) using VT number 2

[   168.105] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[   168.106] (II) xfree86: Adding drm device (/dev/dri/card0)
[   168.119] (II) no primary bus or device found
[   168.119]    falling back to /sys/devices/platform/exynos-drm/drm/card0
[   168.119] (II) LoadModule: "glx"
[   168.121] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   168.127] (II) Module glx: vendor="X.Org Foundation"
[... 221 lines cut ...]
[   216.708] (**) Microsoft Microsoft® 2.4GHz Transceiver v9.0: (accel) selected scheme none/0
[   216.708] (**) Microsoft Microsoft® 2.4GHz Transceiver v9.0: (accel) acceleration factor: 2.000
[   216.708] (**) Microsoft Microsoft® 2.4GHz Transceiver v9.0: (accel) acceleration threshold: 4
[   216.769] (II) input device 'Microsoft Microsoft® 2.4GHz Transceiver v9.0', /dev/input/event2 is tagged by udev as: Mouse
[   216.769] (II) Device 'Microsoft Microsoft® 2.4GHz Transceiver v9.0' set to 1000 DPI
[   216.769] (II) input device 'Microsoft Microsoft® 2.4GHz Transceiver v9.0', /dev/input/event2 is a pointer caps
[   533.068] (II) ARMSOC(0): EDID vendor "LNX", prod id 0
[   533.068] (II) ARMSOC(0): Using hsync ranges from config file
[   533.068] (II) ARMSOC(0): Using vrefresh ranges from config file
[   533.068] (II) ARMSOC(0): Printing DDC gathered Modelines:
[   533.068] (II) ARMSOC(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz eP)
[   533.068] (II) ARMSOC(0): Modeline "1920x1080"x60.0  172.80  1920 2040 2248 2576  1080 1081 1084 1118 -hsync +vsync (67.1 kHz e)
[   533.069] (II) ARMSOC(0): EDID vendor "LNX", prod id 0
[   533.069] (II) ARMSOC(0): Using hsync ranges from config file
[... cut ...]
[   533.069] (II) ARMSOC(0): Printing DDC gathered Modelines:
[   533.069] (II) ARMSOC(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz eP)
[   533.069] (II) ARMSOC(0): Modeline "1920x1080"x60.0  172.80  1920 2040 2248 2576  1080 1081 1084 1118 -hsync +vsync (67.1 kHz e)
[   533.521] (II) UnloadModule: "libinput"
[   533.522] (II) UnloadModule: "libinput"
[   533.522] (II) UnloadModule: "libinput"
[   533.522] (II) UnloadModule: "libinput"
[   533.522] (II) UnloadModule: "libinput"

```

MichaIng, 2019-03-05:
@autur 
Okay the log looks much better now. Still:
```
[   168.293] (EE) AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/armsoc_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/armsoc_dri.so: cannot open shared object file: No such file or directory)
[   168.293] (EE) AIGLX: reverting to software rendering
```
Please paste:
`dpkg -S $(find / -name armsoc_dri.so)`

Kodi freezes still after `wget https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_6/xorg_xu4.conf -O /etc/X11/xorg.conf`, right?

autur, 2019-03-05:
Yes, I ran the wget code in your last reply before starting kodi and the videos still freeze.

```
root@DietPi:~# dpkg -S $(find / -name armsoc_dri.so)
find: ‘/mnt/427C6FF87C6FE55F’: No such device
find: ‘/proc/1137’: No such file or directory
find: ‘/proc/1156’: No such file or directory
find: ‘/proc/1612’: No such file or directory
find: ‘/proc/1687’: No such file or directory
find: ‘/proc/1756’: No such file or directory
dpkg-query: error: --search needs at least one file name pattern argument

Use --help for help about querying packages.
```

MichaIng, 2019-03-05:
Ah lol sorry, wrong lib:
```
dpkg -S $(find /usr/lib libEGL*)
dpkg -S $(find /usr/lib libGLES*)
```

autur, 2019-03-06:
I'm not sure those commands are doing what they should. They output thousands of lines with `dpkg-query: no path found matching pattern /usr/lib/`

See this [output.txt](https://github.com/MichaIng/DietPi/files/2933804/output.txt)

MichaIng, 2019-03-06:
Omg, I need to go to sleep, forgot the `-name` option...
```
dpkg -S $(find /usr/lib -name libEGL*)
dpkg -S $(find /usr/lib -name libGLES*)
```

autur, 2019-03-06:
No worries. I really don't mind testing, but if you like it better I can give remote access to the droid for further testing (I realize there may be rules against it... just saying). 

```
root@DietPi:~# dpkg -S $(find /usr/lib -iname libEGL*)
malit628-odroid: /usr/lib/libEGL.so.1.4
malit628-odroid: /usr/lib/libEGL.so
malit628-odroid: /usr/lib/libEGL.so.1

root@DietPi:~# dpkg -S $(find /usr/lib -iname libGLES*)
malit628-odroid: /usr/lib/libGLESv1_CM.so.1
malit628-odroid: /usr/lib/libGLESv2.so.2.0
malit628-odroid: /usr/lib/libGLESv2.so.2
malit628-odroid: /usr/lib/libGLESv1_CM.so.1.1
malit628-odroid: /usr/lib/libGLESv1_CM.so
malit628-odroid: /usr/lib/libGLESv2.so
malit628-odroid: /usr/lib/arm-linux-gnueabihf/libGLESv1_CM.so.1.1.0
malit628-odroid: /usr/lib/arm-linux-gnueabihf/libGLESv1_CM.so.1
malit628-odroid: /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.0
malit628-odroid: /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2
malit628-odroid: /usr/lib/arm-linux-gnueabihf/libGLESv1_CM.so.1.1
malit628-odroid: /usr/lib/arm-linux-gnueabihf/libGLESv1_CM.so
malit628-odroid: /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.0.0
malit628-odroid: /usr/lib/arm-linux-gnueabihf/libGLESv2.so
```

MichaIng, 2019-03-06:
@autur 
Okay all is installed from exactly the same odroid package, so this should be correct. I even believe that the files in /usr/lib/arm-linux-gnueabihf are symlinks to the /usr/lib versions or the other way round.

Lol actually all those finally link to one single lib: `/usr/lib/libMali.so`
[image]

Okay so this should be all fine. Okay have to sleep now, perhaps indeed remote control is the easier way to investigate.

autur, 2019-03-06:
Ok, I'd be happy to give remote access. I tried installing Remot3.it but it gives an [error](https://github.com/MichaIng/DietPi/files/2934165/dietpi.error.txt) attached) related to a broken link and it fails. If there's others ways, let me know.

MichaIng, 2019-03-06:
@autur 
Jep the issue with Remot3.it is already fixed for 6.22: https://github.com/MichaIng/DietPi/issues/2551
They dropped the old weaved repo. You could try to apply the changes manually: https://github.com/MichaIng/DietPi/pull/2552/files#diff-d92a6ee04e02fd2a2dc23d5bec3e6a98R4337

autur, 2019-03-06:
Thanks, but I'm afraid I dont know which changes to apply (what code to run). Still available to test the odroid myself.

MichaIng, 2019-03-07:
It's actually easier to simply do this manually:
```
wget https://github.com/remoteit/installer/releases/download/v2.2.5/connectd_2.2.5_armhf.deb
dpkg -i connectd_2.2.5_armhf.deb
rm connectd_2.2.5_armhf.deb
```
DietPi-Software does nothing else, besides grabbing the most current version for correct architecture automatically.

autur, 2019-03-07:
Ok, I set up remote access with remot3.it
Let me know where I can send you the details

MichaIng, 2019-03-10:
@autur 
I hope I find time tomorrow, some other things to do.

Btw good news for ARM Mali GPUs, the community creates a reverse-engineering FOSS driver that can be implemented into the Linux kernel natively when ready: https://www.golem.de/news/panfrost-arms-mali-gpus-bekommen-kernel-treiber-der-community-1903-139885.html (German article)
This should make it easier for distros to serve proper GPU support and acceleration for ARMs.

autur, 2019-03-17:
Any chance of moving this up the list of priorities? :)

autur, 2019-04-28:
Any chance of this being resolved?

MichaIng, 2019-04-28:
Could you please try:
```
cp /usr/local/share/setup-odroid/xorg/exynos/xorg.conf /etc/X11/
```

autur, 2019-04-29:
I don't have any folders "setup-odroid" on the system.

MichaIng, 2019-04-29:
@autur 
Ah seems to be part of a APT package. Got it:
```
mv /etc/X11/xorg.conf /etc/X11/xorg.conf.bak
cat << _EOF_ > /etc/X11/xorg.conf
# X.Org X server configuration file for xfree86-video-mali

Section "Device"
	Identifier 	"Mali-Fbdev"
	Driver		"armsoc"
	Option		"fbdev"           	"/dev/fb0"
	Option  	"Debug" 		"false"
	Option		"DPMS"			"false"
	Option		"Fimg2DExa"		"true"
	Option		"DRI2"			"true"
	Option		"DRI2_PAGE_FLIP"	"false"
#	Option		"Fimg2DExaSolid"	"false"
#	Option		"Fimg2DExaCopy"		"false"
	Option		"Fimg2DExaComposite"	"true"
        Option          "SWcursorLCD"           "false"
EndSection

Section "Screen"
	Identifier   "Default Screen"
	Device       "Mali-Fbdev"
	DefaultDepth 24
EndSection

Section "DRI"
	Mode 0666
EndSection
_EOF_
```

This is different (very much older, 2015) than the `xorg.conf` provided by the `xf86-video-armsoc-odroid` driver package that is required as well 🤔.

But worth to give it a try since this is what Meveric suggested (link above).

autur, 2019-04-29:
This configuration is the exact same one that I saved as backup in February. See the Feb 26 post. 
It doesn't change Kodi's behavior, I'm afraid.

MichaIng, 2019-04-29:
@autur 
Ah okay, so then we know where we got our old version from 😉. Okay then this is indeed the outdated one and the one from `xf86-video-armsoc-odroid` should be used instead.

Can you please try to install fbset: `G_AGI fbset`
It was actually not required before but for Kodi 18 (available on RPi) when screen resolution is changed. Just worth to give it a try if in some circumstances also Kodi 17 uses it, especially since `kodi-odroid` is a custom package.

autur, 2019-04-29:
I reverted back to `xf86-video-armsoc-odroid` and installed `G_AGI fbset` but no luck.
If interested I can give remote access for any further testing.

MichaIng, 2019-04-30:
@autur 
Okay just to sum up:
- Your kernel version is up to date, right? `apt update && apt full-upgrade`
- The required video drivers are installed: `apt install malit628-odroid xf86-video-armsoc-odroid`
- Your xorg.conf matches the one shipped from the latter armsoc package: https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_6/xorg_xu4.conf
- You tried the old one above as well.
- libGLES/libEGL libraries are all correct, shipped from `malit628-odroid` package.
- Starting Kodi from deskop or via `xinit` works, but freezes shortly after starting video playback.
- Last log entries before freeze:
```
[   533.068] (II) ARMSOC(0): EDID vendor "LNX", prod id 0
[   533.068] (II) ARMSOC(0): Using hsync ranges from config file
[   533.068] (II) ARMSOC(0): Using vrefresh ranges from config file
[   533.068] (II) ARMSOC(0): Printing DDC gathered Modelines:
[   533.068] (II) ARMSOC(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz eP)
[   533.068] (II) ARMSOC(0): Modeline "1920x1080"x60.0  172.80  1920 2040 2248 2576  1080 1081 1084 1118 -hsync +vsync (67.1 kHz e)
[   533.069] (II) ARMSOC(0): EDID vendor "LNX", prod id 0
[   533.069] (II) ARMSOC(0): Using hsync ranges from config file
[   533.069] (II) ARMSOC(0): Using vrefresh ranges from config file
[   533.069] (II) ARMSOC(0): Printing DDC gathered Modelines:
[   533.069] (II) ARMSOC(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz eP)
[   533.069] (II) ARMSOC(0): Modeline "1920x1080"x60.0  172.80  1920 2040 2248 2576  1080 1081 1084 1118 -hsync +vsync (67.1 kHz e)
[   533.521] (II) UnloadModule: "libinput"
[   533.522] (II) UnloadModule: "libinput"
[   533.522] (II) UnloadModule: "libinput"
[   533.522] (II) UnloadModule: "libinput"
[   533.522] (II) UnloadModule: "libinput"
```
- Missing `fbset` is not the reason here.
- And since @Fourdee verified this, it is not an issue of your particular device.

I will forward this to the Odroid forum and hope that Meveric has some idea.

GulyFMG, 2019-05-02:
@autur 
 Hey guys, maybe this could help, in Dieti-Config --> Display --> GPU / RAM what is the value you have? 
I found that my sweet spot is 192 / under this value i have black/green screen's sometimes no sound etc, the new default value of 320 for me its to much.

autur, 2019-05-02:
Message: "This option is not available for Odroid XU3/XU4/HC1/HC2 (armv7l)"

MichaIng, 2019-05-02:
Correct, GPU memory share is an RPi only thing. Actually not sure how Odroids handle this, allocate GPU memory dynamically or if they have their own fixed memory size?

MichaIng, 2019-05-02:
@autur 
We totally forgot to check the Kodi logs, didn't we? Please check/paste `/root/.kodi/kodi.log` (or similar).

Also Meveric mentioned that pulseaudio might break things. At least worth to give it a try:
```
G_AGP pulseaudio
G_AGA
G_AGI --reinstall alsa-utils
```

And Meveric mentions as well that fbset causes issues in his experience. For Kodi 18 it is required but for Kodi 17 it never was. So to avoid interferences please remove it again: `G_AGP fbset`

autur, 2019-05-08:
Sorry, forgot about this. I uninstalled and reinstalled the G_AG* files as requested--no changes. Kodi log is attached.
[kodi.log](https://github.com/MichaIng/DietPi/files/3155166/kodi.log)

MichaIng, 2019-05-26:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=18177#p18177

@Fourdee 
If you find time 😅.

Speeedfire, 2019-05-30:
[https://dietpi.com/phpbb/viewtopic.php?p=18177#p18177](url)
My sound issue is solved this asound.conf

`pcm.dmixed {
        type asym
        playback.pcm {
                type dmix
                ipc_key 5678293
                ipc_perm 0660
                ipc_gid audio

                slave {
                        channels 2 # make 6 or 5.1 channel
                        pcm {
                                format S16_LE # S32_LE
                                rate 44100 # can also be 44100
                                type hw
                                card 0 # your card
                                device 0 # your device
                                subdevice 0 #important?
                        }

                        period_size 1024
                        buffer_size 2048
                }

                bindings {
                        0 0
                        1 1
# Uncomment below if using 6 channel
#                       2 2
#                       3 3
#                       4 4
#                       5 5
                }
        }
        capture.pcm "hw:0"
}

pcm.!default {
        type plug
        slave.pcm "dmixed"
}`

But i cant solve the freeze issue. Kodi used hw acceleration, but i think this is codec issue or hw lib.

MichaIng, 2019-05-30:
@Speeedfire 
Many thanks for reporting. Yeah so I think first we need to get rid of pulseaudio. It doesn't seem to be used anyway. Stick with ALSA and find a working asound.conf. It's configuration is a bid outside my knowledge, will compare with ours, so we find out which directives are relevant here.

About the freeze, yeah definitely some driver/codec issue, although we install the packages and ship xorg.conf (actually the packages do) that should work. Not sure which is missing. I hope Fourdee finds some time soon to contact Meveric. It's just easier to fix it when you have the SBC beside you.

autur, 2019-06-05:
@MichaIng 
` It's just easier to fix it when you have the SBC beside you.`
Do you think remote access to the Odroid would help? Or would you like a volunteer to send in their Odroid?

autur, 2019-11-09:
@MichaIng Is there any chance that playback issues will be resolved in the near future? Without a local media player, I've basically given up on the Odroid+Dietpi :/

S10t, 2020-01-02:
Dietpi V6.27 has had made a small change in performance.
Kodi video playback comes to halt in 20 seconds instead of 5.
Switching between audio passthrough and direct decoding temperarily bootst playback to normal, to come back to an halt after 10 seconds.

I've found this tutorial for kodi on an odroid xu4 which describes the installation of a specifc video driver. Maybe this could be any help. I dont know how to change videodriver on dietpi installation.

MichaIng, 2020-01-03:
@S10t 
Strange, actually v6.27 introduced a bug to not correctly apply CPU governor, hence it should be performance governor now, as defined in our default boot.ini: https://github.com/MichaIng/DietPi/blob/master/boot_xu4.ini#L31

The GPU driver is installed via malit628-odroid APT package, which should be correctly used by the kodi-ordoid package.

MichaIng, 2020-02-04:
@S10t @autur @Speeedfire 
A new Buster-based image has been uploaded for testing. You might want to try if this solves the freeze issue: #2979

MichaIng, 2020-06-24:
I mark this as closed due to new Buster-based image, Kodi version and several other changes. Feel free to reopen if issue persists.

autur, 2020-11-06:
@MichaIng I have just done a clean install of the latest Dietpi version (Buster) on the Odroid XU4 and can confirm that we still have the same issue with Kodi video playback freezing.

MichaIng, 2021-12-06:
We provide now Bullseye images for Odroid XU4 where no dedicated `kodi-odroid` packages are available, but the regular `kodi` package from Debian is used. I'm not sure how performance compares, but the open source Mesa drivers have greatly improves with v20 provided on Debian Bullseye + Kodi has native GBM/DRM support, hence does not make use of the X server anymore, unless starting from within a desktop session. So this may be tested. Also upgrading to Bullseye can be done: https://dietpi.com/blog/?p=811

I'll organise an XU4 the next weeks and get the shipped kernel updated to v5.4 and then do intense tests with Kodi as well.

As this issue does not apply to the current provided setup anymore, I'll mark it as closed.

closed 2021-12-06 by MichaIng