MichaIng/DietPi #2585: Hostapd.service missing

K2AAE, 2019-02-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version` 9.8
- Kernel version | `uname -a` Linux DietPi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3) RPi 3 Model B+ (armv7l)
- Power supply used | (EG: 5V 1A RAVpower) 5V 3amp
- SDcard used | (EG: SanDisk ultra) SanDisk 16GB Extreme

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) Stratux
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi? don't know
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ... dietpi-update  after downloading and installing the kitchen sink tried to enable hotspot and got:
Failed to start hostapd,service: Unit hostapd,service not found

#### Expected behaviour
 enabling a hotspot I think
- ...

#### Actual behaviour
 can't find a service
- ...

#### Extra details

- ...

MichaIng, 2019-02-24:
@K2AAE 
Thanks for your report.

Did you copy&paste the error? Because there is a type inside:
`hostapd,service` comma instead of dot, should be: `hostapd.service`

MichaIng, 2019-03-02:
I will mark this as closed, as I assume this being indeed a typo. The error message clearly is copy&paste and contains a `,` instead or required `.`.

Feel free to reopen if my assumption is wrong and this requires further investigation.

closed 2019-03-02 by MichaIng