MichaIng/DietPi #2587: root password not saved after change

ICWiener-1, 2019-02-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.17.0
- Distro version | stretch 9.8
- Kernel version | Linux DietPi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | RPi Model B+ (armv71)
- Power supply used | 5V 2,5A
- SDcard used | SanDisk EDGE 16Gb A1 U1 Class 10 Micro SD SDHC 

#### Steps to reproduce
1. Connect via SSH to Rpi to user dietpi (root login disabled)
2. Enter passphrase
3. enter "su" and use default root password
4. enter "passwd"
5. change the password to something new and confirm a second time
6. reboot

#### Expected behaviour
- using su a second time after reboot should work with the newly created passwrod

#### Actual behaviour
- newly created password is refused
- default password is accepted

#### Extra details
- Same behavior when reproducing the steps directly on the Rpi (not via SSH)
- didn't try on "fresh" install
- new to linux so i might miss something obvious ?

MichaIng, 2019-02-24:
@ICWiener-1 
Thanks for your report.

I tried to replicate on Stretch VM but it works fine here.

First I though it might be expected, so non-root users can't change the root login. Perhaps this can be also configured within the sudoers entries.

Can you compare/paste: `cat /etc/sudoers{,.d/*}`
```
root@VM-Stretch:~# cat /etc/sudoers{,.d/*}
#
# This file MUST be edited with the 'visudo' command as root.
#
# Please consider adding local content in /etc/sudoers.d/ instead of
# directly modifying this file.
#
# See the man page for details on how to write a sudoers file.
#
Defaults        env_reset
Defaults        mail_badpass
Defaults        secure_path="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"

# Host alias specification

# User alias specification

# Cmnd alias specification

# User privilege specification
root    ALL=(ALL:ALL) ALL

# Allow members of group sudo to execute any command
%sudo   ALL=(ALL:ALL) ALL

# See sudoers(5) for more information on "#include" directives:

#includedir /etc/sudoers.d
dietpi ALL=NOPASSWD: ALL
# Pi-hole: A black hole for Internet advertisements
# (c) 2017 Pi-hole, LLC (https://pi-hole.net)
# Network-wide ad blocking via your own hardware.
#
# Allows the WebUI to use Pi-hole commands
#
# This file is copyright under the latest version of the EUPL.
# Please see LICENSE file for your rights under this license.
#
www-data ALL=NOPASSWD: /usr/local/bin/pihole
#
# As of Debian version 1.7.2p1-1, the default /etc/sudoers file created on
# installation of the package now includes the directive:
#
#       #includedir /etc/sudoers.d
#
# This will cause sudo to read and parse any files in the /etc/sudoers.d
# directory that do not end in '~' or contain a '.' character.
#
# Note that there must be at least one file in the sudoers.d directory (this
# one will do), and all files in this directory should be mode 0440.
#
# Note also, that because sudoers contents can vary widely, no attempt is
# made to add this directive to existing sudoers files on upgrade.  Feel free
# to add the above directive to the end of your /etc/sudoers file to enable
# this functionality for existing installations if you wish!
#
# Finally, please note that using the visudo command is the recommended way
# to update sudoers content, since it protects against many failure modes.
# See the man page for visudo for more information.
#
```

ICWiener-1, 2019-02-25:
Hello,
Answers below : 

> Did you try it when logging in directly with root user (from local terminal, if blocked on SSH)?

Yes, exact same behavior
> And is the password info only lost after reboot, or is it the same when redoing `su` directly after changed password?

Only after reboot. I can successfully use the **new password** when I do `su - dietpi` followed by `su` to log back as root

> Can you compare/paste: `cat /etc/sudoers{,.d/*}`

My sudoers is different then yours, there are added lines after `#includedir /etc/sudoers.d` : 

```
Defaults env_keep += "NO_AT_BRIDGE"
pi ALL=(ALL) NOPASSWD: ALL
```

otherwise I didn't see any differences. full text below :

```
dietpi@DietPi:~$ sudo cat /etc/sudoers{,.d/*}
#
# This file MUST be edited with the 'visudo' command as root.
#
# Please consider adding local content in /etc/sudoers.d/ instead of
# directly modifying this file.
#
# See the man page for details on how to write a sudoers file.
#
Defaults        env_reset
Defaults        mail_badpass
Defaults        secure_path="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"

# Host alias specification

# User alias specification

# Cmnd alias specification

# User privilege specification
root    ALL=(ALL:ALL) ALL

# Allow members of group sudo to execute any command
%sudo   ALL=(ALL:ALL) ALL

# See sudoers(5) for more information on "#include" directives:

#includedir /etc/sudoers.d
Defaults env_keep += "NO_AT_BRIDGE"
pi ALL=(ALL) NOPASSWD: ALL
dietpi ALL=NOPASSWD: ALL
#
# As of Debian version 1.7.2p1-1, the default /etc/sudoers file created on
# installation of the package now includes the directive:
#
#       #includedir /etc/sudoers.d
#
# This will cause sudo to read and parse any files in the /etc/sudoers.d
# directory that do not end in '~' or contain a '.' character.
#
# Note that there must be at least one file in the sudoers.d directory (this
# one will do), and all files in this directory should be mode 0440.
#
# Note also, that because sudoers contents can vary widely, no attempt is
# made to add this directive to existing sudoers files on upgrade.  Feel free
# to add the above directive to the end of your /etc/sudoers file to enable
# this functionality for existing installations if you wish!
#
# Finally, please note that using the visudo command is the recommended way
# to update sudoers content, since it protects against many failure modes.
# See the man page for visudo for more information.
#
dietpi@DietPi:~$
```

Additionnal notes :
- When I run `passwd`, it doesn't prompt for the old password, I don't know if this is expected ? 
```
root@DietPi:~# passwd
Enter new UNIX password:
```
- dietpi-software runs at every boot, same I don't know if this is expected or a sign that something is returning as default (that's how good I am at linux right now)

MichaIng, 2019-02-25:
@ICWiener-1 
Okay, if it is the same when logging in directly with root user, then `sudoers` has nothing to do with it. Also the additions you have are expected. `pi ALL=(ALL) NOPASSWD: ALL` could be removed, since the `pi` user does not exist on DietPi systems (`dietpi` instead). The `www-data` comes from Pi-hole to allow executing the `pihole` from within the web UI.

Very strange indeed, not sure how this cannot be saved. Free space on your SDcard is sufficient, right?
`df`

You can also compare the following before and after changing the password with `passwd`:
`grep ^root /etc/shadow`
- The line contains the password in a hashed way. When changing, the long cryptic hash should change as well.

Try to do force writing changes to disk (force sync) before rebooting:
`sync; sleep 5`

You could also try the other non-interactive command, although this should make no difference:
`chpasswd <<< 'root:youPassword'`

Please also check the following for error messages, e.g. SDcard I/O errors, voltage issues and such (red lines):
`dmesg`

ICWiener-1, 2019-02-25:
> Very strange indeed, not sure how this cannot be saved. Free space on your SDcard is sufficient, right?
`df`

Yes free space is enough, at least for a password... `df` shows /dev/root with 25% used and 57% of /dev/mmcblk used (my sd card if i'm not mistaken)

> You can also compare the following before and after changing the password with `passwd`:
> `grep ^root /etc/shadow`

About the hash, yes it changes after changing the password. However, it also change after rebooting when the default password is back (it is a new hash, different from the first one, i thought the purpose of a hash was to produce the same result with the same input - the default password in this case - probably more complicated then that)

> Try to do force writing changes to disk (force sync) before rebooting:
> `sync; sleep 5`

When doing this as root :
```
passwd
sync; sleep 5
reboot
```
Same behavior, new password not working, default password working

> You could also try the other non-interactive command, although this should make no difference:
> `chpasswd <<< 'root:yourPassword'`

Same behavior

> Please also check the following for error messages, e.g. SDcard I/O errors, voltage issues and such (red lines):
> `dmesg`

Is it safe to past all the output here ? Otherwise here's the 'analysis' :
I have two consecutive **red lines** (the two of the middle)
```
[    0.810541] [vc_sm_connected_init]: start
[    0.814093] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.814101] [vc_sm_connected_init]: failed to initialize shared memory service
[    0.819510] [vc_sm_connected_init]: end - returning -1
```
other warnings but not in red :
```
[    0.684653] WARN::dwc_otg_hcd_init:1046: FIQ DMA bounce buffers: virt = 0xbdd14000 dma = 0xfdd14000 len=9024
```

```
[    0.701268] WARN::hcd_init_fiq:459: FIQ on core 1 at 0x805ed94c
[    0.704061] WARN::hcd_init_fiq:460: FIQ ASM at 0x805edcb4 length 36
[    0.706815] WARN::hcd_init_fiq:486: MPHI regs_base at 0xf0006000
```

```
[    3.572985] systemd[1]: apt-daily.timer: Cannot add dependency job, ignoring: Unit apt-daily.timer is masked.
[    3.586404] systemd[1]: apt-daily-upgrade.timer: Cannot add dependency job, ignoring: Unit apt-daily-upgrade.timer is masked.
[    3.601575] systemd[1]: serial-getty@ttyS0.service: Cannot add dependency job, ignoring: Unit serial-getty@ttyS0.service is masked.
[    3.617128] systemd[1]: getty-static.service: Cannot add dependency job, ignoring: Unit getty-static.service is masked.
[    3.631649] systemd[1]: systemd-logind.service: Cannot add dependency job, ignoring: Unit systemd-logind.service is masked.
```
and this that says that it's setting default values somewhere (i have no clue):
```
[    0.462349] Core Release: 2.80a
[    0.465041] Setting default values for core params
[    0.467744] Finished setting default values for core params
```

MichaIng, 2019-02-25:
@ICWiener-1 
`/dev/root` is the root partition of the SDcard, which is the relevant here.
`/dev/mmcblk0p1` is the boot partition from the SDcard, holding kernel and bootloader.

`dmesg` output is fine as well:
- `vc_...` are expected if you have less than 32M RAM for GPU (default on DietPi is 16M for server setup).
- `dwc_...` expected as well due to limited USB features on RPi.
- `WARN::hcd_init_fiq` I have the same here, not sure what it's about.
- The systemd dependency job warnings are expected as well, on DietPi we disabled some default services that are not mostly not required. Those are enabled on demand, when required by some software installs or config choices.
- The last output I have as well. It is early boot kernel stage, so not related to any user system stuff. Relevant things need to occur after the first `systemd` line.

As you are on v6.17, did you run into some update issue, e.g. when attempting to update to v6.20? I just have the idea that the system might think it's on first run setup and resets the password to the one in `/DietPi/dietpi.txt`. Although on RPi1 this would lead to a quite long lasting reboot.
To check: `cat /boot/dietpi/.install_stage` should be `1` on v6.19 or earlier and `2` on v6.20 or later.

ICWiener-1, 2019-02-26:
Hello,
I edited my first comment, i'm on Rpi 3 and not 1 as written before, sorry made a copy paste mistake.
Will check the install_stage tonight. What you say seems relevant to me as dietpi-software keeps runing at boot and when i exit it, it says that install is not complete.

MichaIng, 2019-02-26:
@ICWiener-1 
Okay yeah so this is definitely related to the v6.20 update issue. Please do the following to resolve:
```
> /etc/profile.d/dietpi-unsupported_terminal.sh
> /DietPi/dietpi/.update_stage
echo 1 > /DietPi/dietpi/.install_stage
dietpi-update 1 # Do not reboot when asked
sed -i '/^After=/s/[[:blank:]]multi-user.target//' /etc/systemd/system/dietpi-postboot.service
systemctl daemon-reload
```

Then try to change your password and reboot.

ICWiener-1, 2019-02-26:
So `cat /boot/dietpi/.install_stage` shows -1

Tried to execute your commands (as root), I have this message at dietpi-update :

```
[FAILED] File does not exist or cannot be written to by current user

Please verify the existence of the file $3
     /DietPi/dietpi/.installed

Retry with proper permissions or apply the setting manually:
    aSOFTWARE_INSTALL_STATE[116]=0

                      <Ok>
```
The command continues to execute, and prompts for reboot > I said no.
The command ends by 'terminated'
a quick `cat /DietPi/dietpi/.version` shows that i'm now in 6.21.1

I don't want to break anything so i've stopped there. I did a full backup before so it should be fine to experiment, but i have no clue what to experiment > Should i continue with the rest of the commands ?

One innocent question, is it on purpose that you first asked to do this in '/boot' :
`cat /boot/dietpi/.install_stage`
then your command is in '/dietpi':
`echo 1 > /DietPi/dietpi/.install_stage`

MichaIng, 2019-02-26:
@ICWiener-1 
The error prompt occurred during `dietpi-update 1` run?
Was this actually a fresh install of of DietPi or have you already installed software via `dietpi-software` before running into this issue (respectively the v6.20 update issue or first run setup loop)?

If `cat /boot/dietpi/.install_stage` shows `-1` and `/DietPi/dietpi/.installed` is non-existent it looks like indeed it is a fresh install?

Okay little background:
- `/DietPi` is the RAMdisk (tmpfs) location of the DietPi code, which is saved on shutdown to `/boot` and reloaded into `/DietPi` on startup into the RAMdisk.
- The update to v6.20 caused an issue in some cases that prevent correct write of the RAMdisk content `/DietPi` back to disk `/boot`. So e.g. when doing a fresh install at the time when v6.20 was the current DietPi version, first run setup updated code, installed software and such, but the current install state and updated code was not written to disk. So after reboot the untouched code from fresh image was loaded and the system though that it is still a fresh install, rerunning the first run setup step.
- For this reason I asked to you first to check `cat /boot/dietpi/.install_stage` to get the value that was saved to disk, prior to the v6.20 issue.
- `dietpi-update` and other DietPi scripts however read and write all values to RAMdisk location, which is why things need to be changed there to assure a situation that guarantees a correct update.
- However not with v6.21 this issue was solved. The above steps should only assure that the install state is correctly interpreted by the updater, fixing a faulty state due to failed v6.20 update. But I assumed you system was already installed and in use for a while.

ICWiener-1, 2019-02-26:
Thanks a lot for the background info :) No my system is not in use for a while, i've started a few days ago with the goal to run owncloud. but before opening it to the web i really wanted to secure it hence the bug report. Dietpi-software behavior makes sense now if i understood correctly :
- owncloud shows as not installed (not selected from the list)
- I had trouble to drop dropbear for openssh, i had to do it manually because it always showed dropbear in the dietpi-software. 
- Keyboard was always set back to qwerty

so, to answer :

> The error prompt occurred during `dietpi-update 1` run?

Yes.

> Was this actually a fresh install of of DietPi or have you already installed software via `dietpi-software` before running into this issue (respectively the v6.20 update issue or first run setup loop)?

Well, i've never updated dietpi since it's installed (like 2weeks ago). As stated above, tried to go for openssh instead of dropbear via dietpi-software but had to do it manually instead because kept showing me dropbear at reboot. It is still the case. I think that's the only thing i did via dietpi-software.

ICWiener-1, 2019-02-26:
For a test, i've run the last two commands : 
```
sed -i '/^After=/s/[[:blank:]]multi-user.target//' /etc/systemd/system/dietpi-postboot.service
systemctl daemon-reload
```
then changed my password and rebooted > It worked ! 
also : dietpi-software didn't run as if it was first boot.

It looks fixed then. except for the error i had during `dietpi-update 1`, i can keep doing investigations if you need to.

and, thanks again for the walkthrough ! learned some stuff :)

MichaIng, 2019-02-28:
@ICWiener-1
Which was the error on update? Something about missing /etc/profile.d/dietpi-*.sh file?

Please verify the version you are now on:
`cat /DietPi/dietpi/.version`

If it's v6.20.X, please do:
```
> /etc/profile.d/dietpi-unsupported_terminal.sh
> /DietPi/dietpi/.update_stage
ech 1 > /DietPi/dietpi/.install_stage
dietpi-update 1
```
!! Do not reboot when asked!

If it's already v6.21.X, just to be failsafe do the same, just instead with as last:
`dietpi-update -1`

Then repeat:
```
sed -i '/^After=/s/[[:blank:]]multi-user.target//' /etc/systemd/system/dietpi-postboot.service
systemctl daemon-reload
```

This fail-safely brings your system to the desired stage, just to rule out any follow up issues.

Aside that, great that it finally works. I will close this issue now, but feel free to reopen if required.

closed 2019-02-28 by MichaIng

ICWiener-1, 2019-02-28:
Hello,
I was on v6.21.X (already don't remember the X...)
So I did all you asked and all went well even the update this time.  (you have a typo in `
ech 1 > /DietPi/dietpi/.install_stage` it should be ech**o**)
now being on v6.21.1 and all looks nominal :) Thanks again ! (hope this comment will not re-open the topic :/  )

MichaIng, 2019-02-28:
@ICWiener-1 
Perfect and jep sorry, typo ech == ech**o** 😅.