MichaIng/DietPi #2588: Rock64 | Performance tweaks

MichaIng, 2019-02-24:
From forum: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5580
Derived from Ayufan script: https://github.com/ayufan-rock64/linux-package/blob/master/root/usr/local/sbin/rock64_fix_performance.sh

Throws some errors on new ARMbian (much newer kernel version) based image but seems to significantly increase certain performance aspects.

Requires someone with Rock64 (or as well test on RockPro64) board an willingness to test each step, in case find the new/correct kernel API file, and run performance tests.

MichaIng, 2019-03-11:
ARMbian offers a script to enable full GPU acceleration on Rock64: https://forum.armbian.com/topic/9310-rk3328-media-script-rock64-renegade/
Do we use this as well, or should we?

MichaIng, 2019-10-08:
GPU acceleration: https://github.com/MichaIng/DietPi/issues/1794

Pain-Patate, 2020-06-09:
I have a rock64 4Gb. Do you want help to test anything on this hardware ?

MichaIng, 2020-06-11:
If you find time, the following would be interesting to get a picture of the default values that are affected by Ayufans performance script:
- Current GPU DevFreq governors:
```
ls -l /sys/class/devfreq/
cat /sys/class/devfreq/ff9a0000.gpu/governor # should return error as ROCKPro64 only
cat /sys/class/devfreq/ff300000.gpu/governor
cat /sys/class/devfreq/dmc/governor
```
- Current SMP affinities:
```
cat /proc/interrupts
while read -r irq cpu0 cpu1 cpu2 cpu3
do
[[ $icq =~ ahci|echi|ohci|xchi|eth|'0000:01:' ]] && echo -n "$icq = " && cat /proc/irq/${irq%:}/smp_affinity
done < /proc/interrupts
```
- Onboard Ethernet RPS:
```
cat /sys/class/net/eth0/queues/rx-0/rps_cpus
cat /proc/sys/net/core/rps_sock_flow_entries
cat /sys/class/net/eth0/queues/rx-0/rps_flow_cnt
```
- I take out CPU governor since we already cover this with `dietpi-config`/`dietpi-set_cpu`.

clach04, 2020-06-11:
Here is the information on a **non**-dietpi os for comparison on a rock64 1Gb (using an OMV image):

    [root@rock64:~/dietpi] 8s # cat /etc/*elease*
    PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
    NAME="Debian GNU/Linux"
    VERSION_ID="9"
    VERSION="9 (stretch)"
    ID=debian
    HOME_URL="https://www.debian.org/"
    SUPPORT_URL="https://www.debian.org/support"
    BUG_REPORT_URL="https://bugs.debian.org/"
    [root@rock64:~/dietpi] # uname -a
    Linux rock64 4.4.167-1213-rockchip-ayufan-g34ae07687fce #1 SMP Tue Jun 18 20:44:49 UTC 2019 aarch64 GNU/Linux
    [root@rock64:~/dietpi] # cat diag_script_issues_2588.bash
    #/bin/bash

    # for https://github.com/MichaIng/DietPi/issues/2588#issuecomment-642601939

    echo    Current GPU DevFreq governors:

    ls -l /sys/class/devfreq/
    cat /sys/class/devfreq/ff9a0000.gpu/governor # should return error as ROCKPro64 only
    cat /sys/class/devfreq/ff300000.gpu/governor
    cat /sys/class/devfreq/dmc/governor

        echo Current SMP affinities:

    cat /proc/interrupts
    while read -r irq cpu0 cpu1 cpu2 cpu3
    do
    [[ $icq =~ ahci|echi|ohci|xchi|eth|'0000:01:' ]] && echo -n "$icq = " && cat /proc/irq/${irq%:}/smp_affinity
    done < /proc/interrupts

    echo Onboard Ethernet RPS:

    cat /sys/class/net/eth0/queues/rx-0/rps_cpus
    cat /proc/sys/net/core/rps_sock_flow_entries
    cat /sys/class/net/eth0/queues/rx-0/rps_flow_cnt

    echo I take out CPU governor since we already cover this with dietpi-config/dietpi-set_cpu.

    [root@rock64:~/dietpi] # bash diag_script_issues_2588.bash
    Current GPU DevFreq governors:
    total 0
    lrwxrwxrwx 1 root root 0 Jun 11 08:35 dmc -> ../../devices/platform/dmc/devfreq/dmc
    lrwxrwxrwx 1 root root 0 Jun 11 08:35 ff300000.gpu -> ../../devices/platform/ff300000.gpu/devfreq/ff300000.gpu
    lrwxrwxrwx 1 root root 0 Jun 11 08:35 ff360000.rkvdec -> ../../devices/platform/ff360000.rkvdec/devfreq/ff360000.rkvdec
    cat: /sys/class/devfreq/ff9a0000.gpu/governor: No such file or directory
    performance
    performance
    Current SMP affinities:
               CPU0       CPU1       CPU2       CPU3
      1:          0          0          0          0     GICv2  29 Edge      arch_timer
      2:    6948175    6619377    7175214    9746112     GICv2  30 Edge      arch_timer
      3:          0          0          0          0     GICv2  27 Edge      kvm guest timer
     12:          0          0          0          0     GICv2  90 Level     rockchip_thermal
     13:        689          0          0          0     GICv2  89 Level     serial
     14:          0          0          0          0     GICv2  68 Level     ff150000.i2c
     15:     443550          0          0          0     GICv2  69 Level     ff160000.i2c
     17:          0      31599          0          0     GICv2  82 Level     rk_pwm_irq
     18:        432          0          0          0     GICv2  32 Level     ff1f0000.dmac
     19:          0          0          0          0     GICv2  33 Level     ff1f0000.dmac
     20:          0          0          0          0     GICv2 122 Level     Mali_GP
     21:          0          0          0          0     GICv2 119 Level     Mali_GP_MMU
     22:          0          0          0          0     GICv2 125 Level     Mali_PP_Broadcast
     23:          0          0          0          0     GICv2 120 Level     Mali_PP0
     24:          0          0          0          0     GICv2 121 Level     Mali_PP0_MMU
     25:          0          0          0          0     GICv2 123 Level     Mali_PP1
     26:          0          0          0          0     GICv2 124 Level     Mali_PP1_MMU
     27:          0          0          0          0     GICv2  41 Level     ff350000.vpu_service, ff351000.avsd
[... 11 lines cut ...]
     44:          1          0          0    1403930     GICv2  56 Level     eth0
     45:          0          0          0          0     GICv2  55 Level     ff580000.usb, dwc2_hsotg:usb1
     46:          0          0          0          0     GICv2  48 Level     ehci_hcd:usb2
     47:          0          0          0          0     GICv2  49 Level     ohci_hcd:usb3
     48:          0          0          0          0     GICv2  25 Edge      vgic
    123:          0          0          0          0     gpio2   6 Level     rk805
    181:          0          0          0          0     GICv2  94 Level     rockchip_usb2phy
    182:          0          0          0          0     GICv2  93 Level     rockchip_usb2phy
    183:       2519          0      34816          0     GICv2  99 Level     xhci-hcd:usb4
    184:          0          0          0          0     rk805   0 Edge      rk8xx_pwrkey_fall
    189:          0          0          0          0     rk805   5 Edge      RTC alarm
    191:          0          0          0          0     rk805   7 Edge      rk8xx_pwrkey_rise
    IPI0:   1300842    1903175    1560139    1056051       Rescheduling interrupts
    IPI1:    319097     795318      21973       3094       Function call interrupts
    IPI2:         0          0          0          0       CPU stop interrupts
    IPI3:         0          0          0          0       CPU stop (for crash dump) interrupts
    IPI4:         0          0          0          0       Timer broadcast interrupts
    IPI5:         0          0          0          0       IRQ work interrupts
    IPI6:         0          0          0          0       CPU wake-up interrupts
    Err:          0
    Onboard Ethernet RPS:
    f
    32768
    32768
    I take out CPU governor since we already cover this with dietpi-config/dietpi-set_cpu.

clach04, 2020-06-11:
rock64 1Gb, **non**-DietPi Buster image. Using image/process in https://github.com/ayufan-rock64/linux-build/issues/426#issuecomment-626369719

    root@rock64:~# cat /etc/*elease*
    PRETTY_NAME="Debian GNU/Linux 10 (buster)"
    NAME="Debian GNU/Linux"
    VERSION_ID="10"
    VERSION="10 (buster)"
    VERSION_CODENAME=buster
    ID=debian
    HOME_URL="https://www.debian.org/"
    SUPPORT_URL="https://www.debian.org/support"
    BUG_REPORT_URL="https://bugs.debian.org/"
    root@rock64:~# uname -a
    Linux rock64 5.6.0-1137-ayufan-ge57f05e7bf8f #ayufan SMP Wed Apr 15 10:16:02 UTC 2020 aarch64 GNU/Linux

    root@rock64:~# bash diag_script_issues_2588.bash
    Current GPU DevFreq governors:
    total 0
    cat: /sys/class/devfreq/ff9a0000.gpu/governor: No such file or directory
    cat: /sys/class/devfreq/ff300000.gpu/governor: No such file or directory
    cat: /sys/class/devfreq/dmc/governor: No such file or directory
    Current SMP affinities:
               CPU0       CPU1       CPU2       CPU3
      1:          0          0          0          0     GICv2  25 Level     vgic
      3:       5578       6673       6107       9647     GICv2  30 Level     arch_timer
      4:          0          0          0          0     GICv2  27 Level     kvm guest vtimer
      6:        577          0          0          0     GICv2  32 Level     ff1f0000.dmac
      7:          0          0          0          0     GICv2  33 Level     ff1f0000.dmac
      8:          0          0          0          0     GICv2 132 Level     arm-pmu
      9:          0          0          0          0     GICv2 133 Level     arm-pmu
     10:          0          0          0          0     GICv2 134 Level     arm-pmu
     11:          0          0          0          0     GICv2 135 Level     arm-pmu
     15:          0          0          0          0     GICv2  89 Level     ttyS2
     16:       3330          0          0          0     GICv2  69 Level     ff160000.i2c
     17:         18          0          0          0     GICv2  81 Level     ff190000.spi
     19:          0          0          0          0     GICv2  90 Level     rockchip_thermal
     27:          0          0          0          0     GICv2  41 Level     ff350000.video-codec
     28:          0          0          0          0     GICv2  43 Level     ff350800.iommu
     29:       9192          0          0          0     GICv2  64 Level     ff373f00.iommu, ff370000.vop
     30:        787          0          0          0     GICv2  67 Level     ff3c0000.hdmi, dw-hdmi-cec
     32:          0          0          0          0     GICv2 115 Level     ff430000.phy
     33:        101          0          0          0     GICv2  46 Level     dw-mci
     34:     453497          0          0          0     GICv2  44 Level     dw-mci
     35:         21          0          0        557     GICv2  56 Level     eth0
     36:          0          0          0          0     GICv2  55 Level     ff580000.usb, dwc2_hsotg:usb1
     37:          0          0          0          0     GICv2  48 Level     ehci_hcd:usb2
     38:          0          0          0          0     GICv2  49 Level     ohci_hcd:usb3
     80:          0          0          0          0  rockchip_gpio_irq   5 Edge      ff500000.mmc cd
    109:          0          0          0          0  rockchip_gpio_irq   2 Edge      gpio-ir-recv-irq
    113:          0          0          0          0  rockchip_gpio_irq   6 Level     rk805
    171:          0          0          0          0     GICv2  94 Level     rockchip_usb2phy
    172:          0          0          0          0     GICv2  99 Level     xhci-hcd:usb4
    178:          0          0          0          0     rk805   5 Edge      RTC alarm
    IPI0:      7433      50486     173388      89906       Rescheduling interrupts
    IPI1:       615        606        489        476       Function call interrupts
    IPI2:         0          0          0          0       CPU stop interrupts
    IPI3:         0          0          0          0       CPU stop (for crash dump) interrupts
    IPI4:         0          0          0          0       Timer broadcast interrupts
    IPI5:      1706        896       1152       1750       IRQ work interrupts
    IPI6:         0          0          0          0       CPU wake-up interrupts
    Err:          0
    Onboard Ethernet RPS:
    f
    32768
    32768
    I take out CPU governor since we already cover this with dietpi-config/dietpi-set_cpu.

clach04, 2020-06-11:
rock64 1Gb, **non**-DietPi Buster image. libreelec 9.2.0

    LibreELEC:~/dietpi # cat /etc/*elease*
    NAME="LibreELEC"
    VERSION="9.2.0"
    ID="libreelec"
    VERSION_ID="9.2"
    PRETTY_NAME="LibreELEC (official): 9.2.0"
    HOME_URL="https://libreelec.tv"
    BUG_REPORT_URL="https://github.com/LibreELEC/LibreELEC.tv"
    BUILD_ID="5d0330e6ad834a1cc1debbc6b58b3464d027acb4"
    OPENELEC_ARCH="RK3328.arm"
    LIBREELEC_ARCH="RK3328.arm"
    LIBREELEC_BUILD="official"
    LIBREELEC_PROJECT="Rockchip"
    LIBREELEC_DEVICE="RK3328"
    RK3328.arm-9.2.0
    LibreELEC:~/dietpi # uname -a
    Linux LibreELEC 4.4.154 #1 SMP Sat Nov 23 02:13:33 CET 2019 aarch64 GNU/Linux

    LibreELEC:~/dietpi # sh diag_script_issues_2588.bash
    Current GPU DevFreq governors:
    total 0
    lrwxrwxrwx    1 root     root             0 Apr 11  2019 dmc -> ../../devices/platform/dmc/devfreq/dmc
    lrwxrwxrwx    1 root     root             0 Apr 11  2019 ff300000.gpu -> ../../devices/platform/ff300000.gpu/devfreq/ff300000.gpu
    lrwxrwxrwx    1 root     root             0 Apr 11  2019 ff360000.rkvdec -> ../../devices/platform/ff360000.rkvdec/devfreq/ff360000.rkvdec
    cat: can't open '/sys/class/devfreq/ff9a0000.gpu/governor': No such file or directory
    performance
    dmc_ondemand
    Current SMP affinities:
               CPU0       CPU1       CPU2       CPU3
      1:          0          0          0          0     GICv2  29 Edge      arch_timer
      2:      55277      54672      51956      64647     GICv2  30 Edge      arch_timer
     12:          0          0          0          0     GICv2  90 Level     rockchip_thermal
     14:      13041          0          0          0     GICv2  69 Level     ff160000.i2c
     16:       4155          0          0          0     GICv2  32 Level     ff1f0000.dmac
     17:          0          0          0          0     GICv2  33 Level     ff1f0000.dmac
     18:       2133          0          0          0     GICv2 122 Level     Mali_GP
     19:          0          0          0          0     GICv2 119 Level     Mali_GP_MMU
     20:       1067          0          0          0     GICv2 125 Level     Mali_PP_Broadcast
     21:          0          0          0          0     GICv2 120 Level     Mali_PP0
     22:          0          0          0          0     GICv2 121 Level     Mali_PP0_MMU
     23:          0          0          0          0     GICv2 123 Level     Mali_PP1
     24:          0          0          0          0     GICv2 124 Level     Mali_PP1_MMU
     25:          0          0          0          0     GICv2  41 Level     ff350000.vpu_service, ff351000.avsd
     27:          0          0          0          0     GICv2  39 Level     ff360000.rkvdec
     29:          0          0          0          0     GICv2 127 Level     ff330000.h265e
     31:          0          0          0          0     GICv2 129 Level     ff340000.vepu
     33:      10229          0          0          0     GICv2  64 Level     ff370000.vop, ff370000.vop
     34:          0          0          0          0     GICv2  63 Level     ff3a0000.iep
     35:       3226          0          0          0     GICv2  67 Level     ff3c0000.hdmi, dw-hdmi-cec
     37:          0          0          0          0     GICv2 115 Level     ff430000.hdmiphy
     38:          0          0          0          0     GICv2 109 Level     rockchip_u3phy
     39:      13261          0          0          0     GICv2  44 Level     dw-mci
     40:         15          0          0          0     GICv2  46 Level     dw-mci
     41:       1541          0          0          0     GICv2  56 Level     eth0
     42:          0          0          0          0     GICv2  55 Level     ff580000.usb, dwc2_hsotg:usb1
     43:          0          0          0          0     GICv2  48 Level     ehci_hcd:usb2
     44:          0          0          0          0     GICv2  49 Level     ohci_hcd:usb3
    116:       5284          0          0          0     gpio2   2 Edge      gpio-ir-recv-irq
    120:          0          0          0          0     gpio2   6 Level     rk805
    178:          0          0          0          0     GICv2  94 Level     rockchip_usb2phy
    179:          0          0          0          0     GICv2  93 Level     rockchip_usb2phy
    180:          0          0          0          0     GICv2  99 Level     xhci-hcd:usb4
    181:          0          0          0          0     rk805   0 Edge      rk8xx_pwrkey_fall
    186:          0          0          0          0     rk805   5 Edge      RTC alarm
    188:          0          0          0          0     rk805   7 Edge      rk8xx_pwrkey_rise
    IPI0:      3875      20678       8823       6542       Rescheduling interrupts
    IPI1:        19         28         23         18       Function call interrupts
    IPI2:         0          0          0          0       CPU stop interrupts
    IPI3:         0          0          0          0       CPU stop (for crash dump) interrupts
    IPI4:         0          0          0          0       Timer broadcast interrupts
    IPI5:         0          0          0          0       IRQ work interrupts
    IPI6:         0          0          0          0       CPU wake-up interrupts
    Err:          0
    SKIP while loop, no bash available :-(
    Onboard Ethernet RPS:
    0
    0
    0
    I take out CPU governor since we already cover this with dietpi-config/dietpi-set_cpu.

clach04, 2020-06-11:
@MichaIng anything else needed? Do you need DietPi test runs? I need to find a spare SD/USB card so not sure when I could do this but in preparation for that do you have an image/steps that need to be tested?

I suspect @Pain-Patate will get same results with the 4Gb model but be good to get that confirmed.

MichaIng, 2020-06-11:
Many thanks for sharing this info.

What I get from it is that `/sys/class/devfreq/` seems to exist only on legacy kernel (4.4) images, so nothing to do our end on mainline kernel.

Ethernet RPS is disabled by default but manually set on Ayufan and OMV images. Actually interesting one, I have to read a bid into it to find out in which cases it is really a network performance benefit: https://www.suse.com/support/kb/doc/?id=000018430
Probably we can build up a simple test scenario and invite users to test it with different devices, e.g. by copying a large file (or large amount of small files) via NFS and/or SFTP/SCP, then measuring CPU usage on all cores and the time required.

For SMP affinities we could follow a certain logic to separate e.g. local drive I/O, GPU tasks, audio and network interfaces, although e.g. in case of audio and drives they may both be on USB and hence difficult to identify/differentiate on a generic basis. By default SMP affinities seem to allow all CPUs for all IRQs which results (at least on non-x86 systems without SMP balancer) in CPU0 being used for all of them. But first we'd need to check if setting the affinity is actually possible. E.g. I tested on my 4-core RPi where all interrupts are handled by CPU0 but it is not possible to change the affinities:
```
2020-06-11 23:32:40 root@micha:~# l /proc/irq/56/smp_affinity
-rw-r--r-- 1 root root 0 Jun 11 23:32 /proc/irq/56/smp_affinity
2020-06-11 23:32:49 root@micha:~# printf 2 > /proc/irq/56/smp_affinity
bash: printf: write error: Input/output error
```
I was searching the web a bid and indeed this does not seem to be uncommon based on kernel I guess.
____________
Yes basically the output on a board with Armbian 5.X mainline kernel would be great, either our current ROCK64 beta image or: https://www.armbian.com/rock64/

Pain-Patate, 2020-06-13:
Sorry, i'm a noob.
You want tests on [Armbian buster](https://www.armbian.com/rock64/#kernels-archive-all) with this [script](https://forum.armbian.com/topic/9310-rk3328-media-script-rock64-renegade/) installed ?

MichaIng, 2020-06-14:
Either Armbian Buster with Linux 5.X kernel or simply our own image. The media script does not play a role since it deals with GPU acceleration only, which is part of this issue or Ayufans performance script.

GPU acceleration for ROCK64 is discussed here: #1794
The Armbian community media script does not work well with Debian Buster AFAIK due to X11 version incompatibilities, however it could be tested as well. So to get this running we'd need to compile/create an own aarch64 build of libMali.so for ROCK64 since there is none available from any source (I found) compatible with Debian Buster aarch64.

MichaIng, 2022-11-14:
I'll mark this as closed, has become mostly obsolete. SMP IRQ affinities are still a potential enhancement option, but one on all boards where those are writable (not on RPi). I'll have a look at this by times.

closed 2022-11-14 by MichaIng