MichaIng/DietPi #2589: Placeholder / TBA

Fourdee, 2019-02-25:
Placeholder / TBA

WilburWalsh, 2019-03-01:
Thanks @Fourdee for DietPi and your work. Although I only came to know and use DietPi some weeks ago, I am a very happy user :-) I wish you all the best!

@MichaIng Good luck with leading DietPi!

WolfganP, 2019-03-01:
Echo the sentiments above. Thanks a lot @Fourdee for your work and dedication, same for @MichaIng and the best of wishes for the future!

MichaIng, 2019-03-01:
Hey guys,

as well I am proud and very thankful for the trust given by @Fourdee ! To fulfil this I will do my very best to continue the work on DietPi. Although I am stepping in great footsteps, impossible to fully follow in every aspect according to kindness, experience, passion and genius brilliant ideas that we know from @Fourdee 💎.

During my time contributing to DietPi I learned very much, starting from nearly Linux and bash beginner. Slowly I was digging deeper and deeper into the depth of DietPi, Debian/Linux in general and bash coding and I feel now capable enough to take the lead for DietPi.

But I also want to announce related limitations and needs here:
- To limit my free time contribution to DietPi (as this is taking already more than perhaps reasonable when taking into account RL priorities) I never started to use other SBCs than my own RPi2 home server and Virtual Machines + x86_64 PCs to develop and test DietPi code. And I decided to not change something about this now.
- This means that we are seeking especially for new contributors that own one (or more) of our supported SBCs and are willing to help testing (and if possible compiling) new fixes, features and software titles on them. Think of it as kind of a **godparenthood for your own SBC** to keep it working in best harmony with DietPi 😇.
- Fourdee will as well continue with this as required, but to take away responsibility and allow him to focus more on family and health, we greatly appreciate every help as above. We will support new contributors as much as possible to easily find into this project and make DietPi better for their SBC.

LexiconCode, 2019-03-02:
@Fourdee thank you for the awesome community you built up in the wonderful support over the years. I hope your health improves and continue on with your journey in life. 

@MichaIng I cant think of anybody else better suited to take on this role. I've no doubt you will keep up the great work and transition DietPi into the next stage. 

I'm currently busy supporting [Caster](https://github.com/dictation-toolbox/caster/tree/develop) for voice coding and RL. However feel free to reach out if you need access to hardware for testing. Perhaps as time marches on I might be able to contribute.

MichaIng, 2019-03-02:
@LexiconCode 
Thanks for your kind words and suggestion.
_____________
#### About unit testing:
- For now I don't think that DietPi will have much benefit from unit testing. We are a pure bash project and use very simple whiptail-based UI elements, which are easy to test manually. And manual tests allow to find visual quirks, wording and typos, which unit testing can't.
- But the major testing cases are 3rd software installs and usage + device/OS/kernel combination tests, where the important parts that can fail are outside of the DietPi code.
- I mean perhaps one could implement unit tests that even cover software installs and system configurations, but that would need to be individual and on every device we support. And there it comes back to the main issue, that we need contributors who are willing to test DietPi fixes/features/issues on their SBC 😉.
- But of course I am open if someone want's to do a start with a certain DietPi script. At least this can be implemented unit by unit on demand and when anyway touching it.

closed 2019-03-23 by MichaIng