MichaIng/DietPi #2591: Emby Server shutdown when idle

gfrcsd, 2019-02-26:
###Build
- DietPi version
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
```
- Distro version | `stretch`
- Kernel version | `Linux DietPi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux`
- SBC device | `RPi 3 Model B (armv7l)`
- Power supply used | Official Raspberry Pi
- SDcard used | 16GB SanDisk Ultra

#### Additional Information
- Software title | Emby Server
- Fresh install of both DietPi and Emby Server (using DietPi 
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

Happy to help and provide more information if needed. Currently Emby Server seems to shutdown when idle, I had a quick look through the processes and could not see it

MichaIng, 2019-02-26:
@gfrcsd 
Thanks for your report.

I will try to replicate on VM. So the case is very similar to yours, sleep after 10 minutes, only manual service restart fixing it?

gfrcsd, 2019-02-26:
I'm not sure about the 10 minutes but the server was set up yesterday and after not using it for a bit a couple of times, Emby isn't available anymore and needs to be restarted. 

After restarting it last night, I looked at the processes this morning, and it looked like it was running but couldn't connect to it

MichaIng, 2019-02-26:
@gfrcsd 
Okay, I leave my VM with unused Emby and check by once and a whole. Meanwhile, as 10 minutes at least is such a specific time that applies to many default hardware sleep timers, perhaps Emby breaks when the attached media drive is going to sleep? Do you have your dietpi_userdata or media library moved to an external drive?

gfrcsd, 2019-02-26:
Thank you for your prompt replies 

That's an interesting take but the only drive Emby uses is a NFS share in a NAS which is mapped using `dietpi-drive_manager`. Emby itself along with DietPi is installed on the microSD Card (no external hard drives plugged in the Pi either)

I tried to access 192.168.XX.XX:8096 it is unable to connect and Emby Server is not in the processes anymore (I restarted it 3 hours ago and everything was up and running)

MichaIng, 2019-02-26:
@gfrcsd 
Hmm, could you try to temporary make Emby use a local drive/directory instead of the NFS share, so we can verify or exclude NFS being the reason.

gfrcsd, 2019-02-26:
Would unmounting the NFS in `dietpi-drive_manager` be enough for that?

MichaIng, 2019-02-26:
@gfrcsd 
At best remove the NFS from within Emby UI temporarily and use a local drive/dir instead.

gfrcsd, 2019-02-26:
Same behavior

MichaIng, 2019-03-01:
@gfrcsd 
Okay so if it is the some without NFS we can sort this out as a reason.

But just to be sure:
- After Emby crash, the NFS drive is still available from console? I remember from the other issue you said it is empty as well?

What I am thinking off is that your network adapter might be going to sleep after 10 minutes, leading to NFS mount connection loss and Emby crash, since it binds to network as well.
- Do you use an Ethernet adapter or WiFi?

And of course before guessing, please paste, directly after crash:
`dmesg | tail`
`journalctl -u emby-server` or `journalctl -t emby-server`

gfrcsd, 2019-03-05:
Sorry for the late reply @MichaIng 

- After the Emby crash, the NFS drive are still there if I go into ```dietpi-drive_manager```

- I use the ethernet adapter, WiFi is not set up

After inspecting the log using ```journalctl -t emby-server```, I may have found the culprit as this seems to appear before every instance of me restarting the server :

``` 
Mar 01 18:41:44 DietPi emby-server[30296]: *** Error in `/opt/emby-server/system/EmbyServer': double free or corruption (fasttop): 0x6c801cc8 *** 
```

Emby does not log the reason for the shutdown in my case, it just stops logging anything and the start up logs are normal

MichaIng, 2019-03-05:
@gfrcsd 
Do you have DLNA features enabled? I searched for this error message and found several matches where enabled DLNA produces this error, e.g.: https://emby.media/community/index.php?/topic/60703-emby-server-on-raspberry-pi-crashing/page-3

gfrcsd, 2019-03-06:
Apologies for not looking this up I have been pretty busy recently. I have turned off DLNA features, will report back. Thank you again for your invaluable help!

MichaIng, 2019-03-08:
@gfrcsd 
Okay. Jep at least the emby communits/devs might have better insight in what might fails. Feel free to forward infos or questions which might be related to DietPi/system.

gfrcsd, 2019-03-12:
Just some feedback in case someone stumble onto this, DLNA seems to be the culprit. If in the same situation, try disabling it.

closed 2019-03-12 by gfrcsd