MichaIng/DietPi #2593: Jackett activating (auto-restart) (Result: exit-code)

ezekini, 2019-02-26:
### Creating a bug report/issue

#### Required Information
- DietPi version : 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee
- Distro version : 9.6
- Kernel version : Linux DietPi-rpi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
- SBC device : RPi 3 Model B+ (armv7l)
- Power supply used: 5V 1A RAVpower

#### Additional Information (if applicable)
- Software title: Jackett
- Was the software title installed freshly or updated/migrated? Fresh
- Can this issue be replicated on a fresh installation of DietPi? Don't know
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`: 3974a20e-b6ca-4207-bf41-bb0f7c9d4f8f

#### Steps to reproduce

1. ... Update to 6.20.6 or 6.21.1 (I don't know when the problem started exactly)
2. ...Jacket won't start: 
[  OK  ] DietPi-Services | jackett	activating (auto-restart) (Result: exit-code) since Tue 2019-02-26 11:28:36 -03; 4s ago

MichaIng, 2019-02-27:
@ezekini 
Thanks for your report.

I will check your bug report.

MichaIng, 2019-03-02:
@ezekini 
Sorry for late reply.

I didn't find the bug report, could you retry and send a (new) report:
```
systemctl start jackett
dietpi-bugreport
```

Otherwise the interesting outputs are:
`journalctl -u jackett` or `journalctl -t jackett`

ezekini, 2019-03-02:
Hi @MichaIng. Here is the bugreport again: 
Bug report sent, reference code: 3974a20e-b6ca-4207-bf41-bb0f7c9d4f8f.
BTW, `journalctl -u jackett `is very long (I gues you will see it the report),  `journalctl -t jackett` has no entries.

MichaIng, 2019-03-02:
@ezekini 
The last lines of `journalctl -u jackett` after crash/service start failure should be sufficient. However will check the bug report now.

ezekini, 2019-03-03:
@MichaIng, memory seems fine. Output is:
```
            total        used        free      shared  buff/cache   available
Mem:            685         323          71          52         290         259
Swap:          1071           0        1071
```
This behaviour with jackett was not a one time event. If I reboot, the same thing happens. If I restart the service, the same. Right now it is basically not functioning.

MichaIng, 2019-03-08:
@ezekini 
Hmm could you send a bug report right after the error occurred? As said in the bug report I checked there was no error visible.

Also interesting that DietPi-Services shows `[  OK  ]` even that it shows an exit code 🤔:
> [ OK ] DietPi-Services | jackett	activating (auto-restart) (Result: exit-code) since Tue 2019-02-26 11:28:36 -03; 4s ago

ezekini, 2019-03-08:
I have reinstalled using dietpi-software reinstall appID and it is working now. If the error reappears, we can investigate further, for now we can close this so you can dedicate time to more important stuff😉 .

MichaIng, 2019-03-08:
@ezekini 
Ah great, jep perhaps simply a faulty/failed install the first time.

DeathIsUnknown, 2019-05-07:
@MichaIng 
This issue and #2328 could be related to Jackett's built-in updater leaving the service in a broken state. I say that since every time an update is released, Jackett fails to start on my machine. This would also explain why these issues are difficult to reproduce and that reinstalls are fixing them.

This is my error from `journalctl -u jackett.service`, which started today around the time of the release of [Jackett v0.11.295](https://github.com/Jackett/Jackett/releases/tag/v0.11.295). Note that the version in the logs is `v0.11.279.0`.
```
May 07 21:17:33 DietPi mono[7821]: 05-07 21:17:33 Info Starting Jackett v0.11.279.0
May 07 21:17:35 DietPi mono[7821]: 05-07 21:17:35 Error System.MissingMethodException: Method not found: AutoMapper.IMappingExpression`2<!0, !1> AutoMapper.IMappingExpression`2.AfterMap(System.Action`2<!0, !1>)
May 07 21:17:35 DietPi mono[7821]:   at AutoMapper.MapperConfiguration.Build (System.Action`1[T] configure) [0x00006] in <8645827e567f4e9e927a7dd252e7ec3a>:0
May 07 21:17:35 DietPi mono[7821]:   at AutoMapper.MapperConfiguration..ctor (System.Action`1[T] configure) [0x00000] in <8645827e567f4e9e927a7dd252e7ec3a>:0
May 07 21:17:35 DietPi mono[7821]:   at AutoMapper.Mapper.Initialize (System.Action`1[T] config) [0x00000] in <8645827e567f4e9e927a7dd252e7ec3a>:0
May 07 21:17:35 DietPi mono[7821]:   at Jackett.Server.Helper.InitAutomapper () [0x00009] in <9c1b87b5e9e74c5086bc89d8846d979e>:0
May 07 21:17:35 DietPi mono[7821]:   at Jackett.Server.Helper.Initialize () [0x00007] in <9c1b87b5e9e74c5086bc89d8846d979e>:0
May 07 21:17:35 DietPi mono[7821]:   at Jackett.Server.Startup.ConfigureServices (Microsoft.Extensions.DependencyInjection.IServiceCollection services) [0x00171] in <9c1b87b5e9e74c5086bc89d8846d979e>:0
May 07 21:17:35 DietPi mono[7821]:   at (wrapper managed-to-native) System.Reflection.MonoMethod.InternalInvoke(System.Reflection.MonoMethod,object,object[],System.Exception&)
May 07 21:17:35 DietPi mono[7821]:   at System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0006a] in <6649516e5b3542319fb262b421af0adb>:0
```

I've correlated Jackett breaking every time a new version is released, but the error is not always the same. Before today I would usually get an error about a missing file such as in this issue: https://github.com/Jackett/Jackett/issues/3635. Checking `/opt/jackett/` for the file would usually show that the file existed, but with a filesize of 0.

Remember that I was originally running **v0.11.279.0** and the service is still reportedly trying to start v0.11.279.0. `/opt/jackett/.config/Jackett/updater.txt` shows this update attempt from earlier today:
```
2019-05-07 18:40:20.8800 Info Jackett Updater **v0.11.295.0**
2019-05-07 18:40:20.9515 Info Options "--Path" "/opt/jackett" "--Type" "Console" "--Args" " JackettConsole.exe " "--KillPids" "6973"
2019-05-07 18:40:20.9573 Info Jackett variant: Mono
2019-05-07 18:40:21.2065 Info Finding files in: /tmp/JackettUpdate-v0.11.295-636928512143410940/Jackett/
2019-05-07 18:40:21.2455 Info Attempting to copy AngleSharp.dll from source: /tmp/JackettUpdate-v0.11.295-636928512143410940/Jackett/AngleSharp.dll to destination: /opt/jackett/AngleSharp.dll
2019-05-07 18:40:21.2493 Info Copied AngleSharp.dll
...
2019-05-07 18:40:21.4265 Info Attempting to copy classix.yml from source: /tmp/JackettUpdate-v0.11.295-636928512143410940/Jackett/Definitions/classix.yml to destination: /opt/jackett/Definitions/classix.yml
2019-05-07 18:40:21.4265 Info Copied classix.yml
2019-05-07 18:40:21.4265 Info Attempting to copy concen.yml from source: /tmp/JackettUpdate-v0.11.295-636928512143410940/Jackett/Definitions/concen.yml to destination: /opt/jackett/Definitions/concen.yml
2019-05-07 18:40:21.4265 Error System.UnauthorizedAccessException: Access to the path "/tmp/JackettUpdate-v0.11.295-636928512143410940/Jackett/Definitions/concen.yml" or "/opt/jackett/Definitions/concen.yml" is denied.
  at System.IO.File.Copy (System.String sourceFileName, System.String destFileName, System.Boolean overwrite) [0x00192] in <6649516e5b3542319fb262b421af0adb>:0
  at Jackett.Updater.Program.CopyUpdateFile (System.String jackettDestinationDirectory, System.String fullSourceFilePath, System.String updateSourceDirectory, System.Boolean previousAttemptFailed) [0x00162] in <a241fc80b1fb463bb7$
  at System.IO.File.Copy (System.String sourceFileName, System.String destFileName, System.Boolean overwrite) [0x00192] in <6649516e5b3542319fb262b421af0adb>:0
  at Jackett.Updater.Program.CopyUpdateFile (System.String jackettDestinationDirectory, System.String fullSourceFilePath, System.String updateSourceDirectory, System.Boolean previousAttemptFailed) [0x00162] in <a241fc80b1fb463bb7$
2019-05-07 18:40:21.4445 Info Attempting to copy concen.yml from source: /tmp/JackettUpdate-v0.11.295-636928512143410940/Jackett/Definitions/concen.yml to destination: /opt/jackett/Definitions/concen.yml
2019-05-07 18:40:21.4445 Info The first attempt copying file: concen.ymlfailed. Retrying and will delete old file first
2019-05-07 18:40:21.4445 Info /opt/jackett/Definitions/concen.yml was found
```

My assumption here is that perhaps the Jackett service is restarting before the updater has a chance to install the new version. Otherwise, perhaps the updater is just broken and this should be an issue on the Jackett repo.
[... cut ...]
I tried to have a look through the Jackett repo for clues and the only difference I can find is that the recommended service runs Jackett with `/usr/bin/mono --debug /opt/jackett/JackettConsole.exe --NoRestart` ([link](https://github.com/Jackett/Jackett/blob/aa8e10672740ab31ac4986dfc89875a2f1c5e418/install_service_systemd_mono.sh#L53)), versus DietPi's `/usr/bin/mono -O=-aot /opt/jackett/JackettConsole.exe` ([link](https://github.com/MichaIng/DietPi/blob/b293ba874d4497679471c882d0a6eba5e135476a/dietpi/dietpi-software#L12036)). Perhaps --NoRestart is important here for the updater?

MichaIng, 2019-05-07:
@DeathIsUnknown 
Many thanks for report and investigation.

The failing part is
> Error System.UnauthorizedAccessException: Access to the path "/tmp/JackettUpdate-v0.11.295-636928512143410940/Jackett/Definitions/concen.yml" or "/opt/jackett/Definitions/concen.yml" is denied.

which is strange since the other files just work well. Since everything runs as "jackett" user all files should be owned by it including the existing Jackett install dir as well as the downloaded update, which is proven until `concen.yml` 🤔.

However could you please verify permission:
```
ls -al /opt/jackett /tmp/JackettUpdate*
```

After failure it tries to copy concen.yml a second time but then stops. The question is, if the daemon stops and then the systemd unit stops because of this, of indeed the updater restarts the daemon, which is expected to fail, since the systemd unit stops when the main process exits, preventing it from starting up again.
This docs btw contain -NoRestart as well: https://www.htpcguides.com/install-jackett-ubuntu-15-x-for-custom-torrents-in-sonarr/
And here: https://github.com/Jackett/Jackett/wiki/systemd-service

Okay I will do some test install to verify the option and update behaviour.

MichaIng, 2019-05-07:
Hmm another topic:

> Standalone version of Jackett is now available - Mono not required 
> To upgrade to the standalone version of Jackett, click here for install instructions. Upgrading is straight forward, simply install the standalone version and your indexers/configuration will carry over. Benefits include: increased performance, improved stability and no dependency on Mono.

MichaIng, 2019-05-07:
@DeathIsUnknown 
Dumb question: Where do I find the internal updater? Or is it a separate binary?

MichaIng, 2019-05-07:
Okay, the new standalone version (available since Jackett v0.11) will be installed on non-ARMv6 devices. Sadly there is no ARMv6 binary available, so these need to stay with mono: https://github.com/MichaIng/DietPi/pull/2773

DeathIsUnknown, 2019-05-08:
I’d be Interested to know what effect `/usr/bin/mono --optimize=all` (Recommended in your link https://github.com/Jackett/Jackett/wiki/systemd-service) has on performance in comparison to `-O=-aot`.

MichaIng, 2019-05-08:
I just remember we had this once but it caused issues. However with the standalone binary it doesn't matter anymore,  as it runs without mono.

DeathIsUnknown, 2019-05-08:
I'm unable to run `ls -al /opt/jackett /tmp/JackettUpdate*` on my broken instance since I've fixed the issue and `tmp` has been cleared. Sorry! Thanks for looking into this - hopefully we've solved it. 👍

MichaIng, 2019-05-08:
@DeathIsUnknown
I could still not verify that using the internal updater does not break things since I couldn't find a way to use/force it.

I ran one instance for > one hour with auto updater enabled and it finds/logs a new version when checking via web UI but how do I force it to apply the update?

DeathIsUnknown, 2019-05-10:
@MichaIng
"Check for updates" from the web UI seemed to trigger the update for me. Here's journalctl:

```
May 11 00:14:58 DietPi mono[30435]: 05-11 00:14:58 Info Jackett variant: Mono
May 11 00:14:59 DietPi mono[30435]: 05-11 00:14:59 Info New release found.  Current: v0.11.295 New: v0.11.316
May 11 00:15:24 DietPi mono[30435]: 05-11 00:15:24 Info Starting updater: mono /tmp/JackettUpdate-v0.11.316-636931305191255460/Jackett/JackettUpdater.exe --Path "/opt/jackett" --Type "Console" --Args " JackettConsole.exe " --KillPids "30435"
May 11 00:15:24 DietPi mono[30435]: 05-11 00:15:24 Info Updater started process id: 31246
May 11 00:15:24 DietPi mono[30435]: 05-11 00:15:24 Info Exiting Jackett..
May 11 00:15:25 DietPi mono[30435]: Application is shutting down...
May 11 00:15:30 DietPi systemd[1]: jackett.service: Service hold-off time over, scheduling restart.
May 11 00:15:30 DietPi systemd[1]: Stopped Jackett Daemon (DietPi).
May 11 00:15:30 DietPi systemd[1]: Started Jackett Daemon (DietPi).
May 11 00:15:32 DietPi mono[31264]: 05-11 00:15:32 Info Starting Jackett v0.11.295.0
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info Environment version: 4.0.30319.42000 (/usr/lib/mono/4.5)
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info OS version: Unix 5.0.0.1105 (64bit OS) (64bit process)
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info Jackett variant: Mono
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info ThreadPool MaxThreads: 600 workerThreads, 200 completionPortThreads
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info App config/log directory: /opt/jackett/.config/Jackett
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info issue: Debian GNU/Linux 9 \n \l
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info mono version: 5.20.1.19 (tarball Thu Apr 11 09:09:09 UTC 2019)
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info TrustedRootCertificates count: 160
May 11 00:15:36 DietPi mono[31264]: 05-11 00:15:36 Info Using HTTP Client: HttpWebClient
May 11 00:15:38 DietPi mono[31264]: 05-11 00:15:38 Info Loading Cardigann definitions from: /opt/jackett/.config/cardigann/definitions/, /etc/xdg/cardigan/definitions/, /opt/jackett/Definitions
May 11 00:15:38 DietPi mono[31264]: 05-11 00:15:38 Info Loading Cardigann definition /opt/jackett/Definitions/1337x.yml
May 11 00:15:38 DietPi mono[31264]: 05-11 00:15:38 Info Loading Cardigann definition /opt/jackett/Definitions/2fast4you.yml
...
May 11 00:15:48 DietPi mono[31264]: 05-11 00:15:48 Info Adding aggregate indexer
May 11 00:15:48 DietPi mono[31264]: 05-11 00:15:48 Info Deleting JackettUpdate temp files from /tmp/JackettUpdate-v0.11.316-636931305191255460
May 11 00:15:50 DietPi mono[31264]: Hosting environment: Production
May 11 00:15:50 DietPi mono[31264]: Content root path: /opt/jackett/Content
May 11 00:15:50 DietPi mono[31264]: Now listening on: http://[::]:9117
May 11 00:15:50 DietPi mono[31264]: Application started. Press Ctrl+C to shut down.
```
After the updater was triggered, Jackett started back up again with the version number still reporting **0.11.295.0** instead of the downloaded **0.11.316.0**.

MichaIng, 2019-05-10:
@DeathIsUnknown 
Strange, I think my instance didn't trigger the update. I will recheck.

However interesting that the service is still restarted with `--NoRestart`. Was the update at least successful then?

> jackett.service: Service hold-off time over, scheduling restart.

Hmm, some timeout is reached triggering systemd to restart the service. If this makes the update still break, possibly we need to increase some timeouts there and/or report the issue to Jackett to be fixed, respectively their example systemd units to be updated.

DeathIsUnknown, 2019-05-10:
@MichaIng 
The update was successful this time, though I'm not sure if I should attribute it to restarting the service and triggering the update or to adding `--NoRestart`.

DeathIsUnknown, 2019-05-20:
@MichaIng 
Thanks for following this issue, but it seems it still exists in the latest update. I am going to assume that this is on Jackett's side since other users over at the Jackett repo seem to be having the same issue. See https://github.com/Jackett/Jackett/issues/5285.

> However interesting that the service is still restarted with `--NoRestart`. Was the update at least successful then?
>
>
> Hmm, some timeout is reached triggering systemd to restart the service. If this [...]

I would assume these lines in `jackett.service` to be the culprit, but they are also recommended by the Jackett docs.
```
Restart=always
RestartSec=5
```

I'll follow the issue on the other repo and update here if a solution is found.

MichaIng, 2019-05-20:
@DeathIsUnknown 
Damn thing. Yeah actually it makes somehow sense:
- If Jackett restarts itself, then systemd will recognise main process being down, thus forces a restart of the service due to: `Restart=always`
- Probably you have two processes starting up then, breaking each other.
- Removing `Restart=always` does not really solve it, because systemd then still thinks that the service is dead and will not recognise another process being spinned up (most probably). So even the Jackett would be up then, the systemd service will report a failure. This also should be the reason why `--NoRestart` is required and then `Restart=always`. So after update, Jackett kills itself (not triggering a restart itself) and systemd recognises process down and restarts the service.

According to your logs, at least I cannot see anything not working about that
```
May 11 00:15:24 DietPi mono[30435]: 05-11 00:15:24 Info Updater started process id: 31246
May 11 00:15:24 DietPi mono[30435]: 05-11 00:15:24 Info Exiting Jackett..
May 11 00:15:25 DietPi mono[30435]: Application is shutting down...
May 11 00:15:30 DietPi systemd[1]: jackett.service: Service hold-off time over, scheduling restart.
May 11 00:15:30 DietPi systemd[1]: Stopped Jackett Daemon (DietPi).
May 11 00:15:30 DietPi systemd[1]: Started Jackett Daemon (DietPi).
May 11 00:15:32 DietPi mono[31264]: 05-11 00:15:32 Info Starting Jackett v0.11.295.0
```

However the update still results in a failing service? Is there any second mono/jackett process up? `pgrep -f 'jackett'`

MichaIng, 2019-05-27:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=18240#p18240

Dr0bac, 2019-07-01:
Is this fixed? I uninstalled jacket because it restart itself every second, temp were high

MichaIng, 2019-07-01:
@Dr0bac 
As of the linked issue on Jackett repo, nope, not fixed. Please disable the internal updater for now.

MichaIng, 2019-07-26:
Referencing all related issues:
- https://github.com/Jackett/Jackett/issues/5208
- https://github.com/Jackett/Jackett/issues/5495
- https://github.com/Jackett/Jackett/issues/5641

Seems to not being fixed soon. We need to find a way to disable the internal updater code-wise for now and prepare our install code to provide a reliable update method via `dietpi-software reinstall 147`.

ngosang, 2019-10-30:
@MichaIng Test the official fix => https://github.com/Jackett/Jackett/issues/5208#issuecomment-547709864

MichaIng, 2019-10-31:
@ngosang 
Jep already did and suggested some minor enhancements. Will add it with next DietPi release as well which will be very soon as kinda hotfix release, reasonable due to some other issues our side.

MichaIng, 2019-11-01:
Note to self: `mono-xsp4` (webserver) sufficient instead of `mono-complete`.

MichaIng, 2019-11-05:
Implemented workaround: https://github.com/MichaIng/DietPi/commit/9a543c7210dcdeabb5f382f261a6d27efcdc8f6d

EdEstes, 2019-12-02:
I've experienced this bug twice now.  (reinstalled the software the first time)  Not sure how to help, but I sent in debug info - reference code: 9fe2513d-da5f-4b3d-9cdb-1226033186aa

MichaIng, 2019-12-02:
@EdEstes 
Note that the workaround has been applied with v6.27. Apply the `dev` branch for the fix: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md
Otherwise beta will be released most likely tomorrow and release until end of week.

closed 2020-01-01 by MichaIng