MichaIng/DietPi #2597: Write error - ~LZMAFILE (28: No space left on device)

kuerious, 2019-02-27:
[  OK  ] DietPi-Software | APT installation for: bash-doc -s, please wait...
Inst bash-doc (4.4-5 Raspbian:stable [all])
Conf bash-doc (4.4-5 Raspbian:stable [all])
E: Write error - ~LZMAFILE (28: No space left on device)

[FAILED] DietPi-Software | G_AGI: bash-doc -s

--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Wed 27 Feb 12:09:58 MST 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: bash-doc -s
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Inst bash-doc (4.4-5 Raspbian:stable [all])
Conf bash-doc (4.4-5 Raspbian:stable [all])
E: Write error - ~LZMAFILE (28: No space left on device)
```
--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2019-02-27:
@kuerious 
The error message is quite clear? How much space do you have left on your root file system:
`df`

kuerious, 2019-02-27:
RFS Storage:   8GB (7.6GB actual)
Space Used:    5.6GB
Space Rem:     1.9GB

I've cleared locked files, checked log files (RAM log, so not an issue), and tried both Debian and Ubuntu troubleshooting.

It still came up with that error. So I've emplaced an 8GB USB drive, enabled, transferred RootFS to it, etc.

Seems to be happy now. My guess, a bad block. But I thought (figured) DietPi handled that...

MichaIng, 2019-02-27:
@kuerious 
Bad blocks should lead to I/O errors shown in `dmesg` output usually. They cannot be handled in an automated way AFAIK, you need to identify the affected inode(s) and on which drive sectors they are located and fill that sectors manually. If something as done wrong, data would be lost and/or corrupted, so way too dangerous to somehow automate this.

How much space did your initial SDcard (?) have? Possible would be as well that file system expansion has failed, so even that space is left on the drive, the file system does not use all of it. Our images are shrunk to allow smaller download and image size to actual data + 50M AFAIK. This should be expanded to maximum drive space on first boot.

kuerious, 2019-02-27:
Michalng, I pointed out it was an 8GB SD card. Replaced now by an 8GB USB flash drive.

At this point, I'm guessing that my system is functional. I have another issue with removing software that brought this issue up. I can now uninstall, et. al. Not sure if you want logs to help prevent this in the future.

MichaIng, 2019-02-27:
@kuerious 
Ah okay SDcard AND USB flash are 8G, okay.

Logs not required, but thanks for offering. DietPi-Software checks for 500M free space before installing software. Of course if you select many/large titles, this might not be sufficient, but impossible to predict. So if it's about drive space, users have to recognize/take care this themselves as well 😉.

closed 2019-02-28 by MichaIng

Symorian, 2019-03-05:
I have same error, with 64GB SDcard. Please try to fix that.

Symorian, 2019-03-05:
I have alot of space because it new install... first software install and i get this error... @MichaIng

Symorian, 2019-03-05:
Im not here for debate... just for put log DietPi tell me to put on this issues.

MichaIng, 2019-03-05:
@Symorian 
Please paste:
```
df
fdisk -l /dev/mmcblk0p2
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

kuerious, 2019-03-05:
I've decided to give up on SD cards for the time being. High-end USB 3.x 16
& 32GB flash drives are incredibly inexpensive as of late. As I use DietPi,
moving the data over is a snap. So ... f$#! SD cards.

MichaIng, 2019-03-05:
@kuerious 
Yeah, SDcards are quite sensible for data corruption. Hight quality/durability SDcards are the way to go, or, as you did move the system onto USB drive. Sadly SDcards are the common default for SBCs, if they don't have a separate eMMC slot.

kuerious, 2019-03-05:
I've bought SanDisk, Samsung, PNY, and others. At complete random, they
freak out and die. Often, with little to no discernable warning signs. RMA
all you want, but I have a small stack I have pulled to replace with USB.

As for SBCs, well, its rare anymore to not find at least a pair of USB 2.0
ports. And either the original, or a replacement, bootloader. <shrug>

I'm sticking with USB and PXE/network boot for the time being. YMMV

MichaIng, 2019-03-05:
@kuerious 
Another issue with SDcards that need to be taken care of:
- Veeery often, actually in most cases, when buying brand (Samsung, SanDisk, Kingston etc...) SDcards from 3rd party vendors on eBay, Amazon and deals/offers found on deal-seeking websites or via search machine, you get a fake product.
- So it is very important to buy those from trusted vendors with own website (or local store) where you know they are reachable with quality support. They are more expensive then usually. And always do some tests/verification that the card is an original one. Otherwise when sending them for replacement/warranty you are lost 😉.

kuerious, 2019-03-05:
I agree

DeHaanSoftware, 2022-10-02:
/var/log is mounted on /dev/zram1 which seems to be a ram disk? It's only 50 Mb and it's full all the time.

Joulinar, 2022-10-02:
Yes it is a ram disk. But usually it should not be used that heavy. Let's check what is inside

```
du -a /var/log | sort -n -r | head -n 20
```

MichaIng, 2022-10-02:
> /var/log is mounted on /dev/zram1

Then it's not a tmpfs (RAM disk) but a zram space (compressed RAM disk), which isn't default on DietPi but on Armbian. However, in both cases some service should take care to clean up the log files regularly.