MichaIng/DietPi #2598: Error message attempting (re-)install of Remot3.it (now weaved); "404 Not found"

kuerious, 2019-02-27:
#### Details:
- Date           | Wed 27 Feb 13:59:21 MST 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | Connection test: https://github.com/weaved/installer/raw/master/Raspbian%20deb/1.3-07/weavedconnectd_1.3-07z1_armhf.deb
- Exit code      | 8
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-02-27 13:57:17--  https://github.com/weaved/installer/raw/master/Raspbian%20deb/1.3-07/weavedconnectd_1.3-07z1_armhf.deb
Resolving github.com (github.com)... 192.30.255.112, 192.30.255.113
Connecting to github.com (github.com)|192.30.255.112|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

kuerious, 2019-02-27:
It would appear that they've closed their public repos? ...
https://i.imgur.com/bD8Xw9n.png

[image]

MichaIng, 2019-02-27:
@kuerious 
Thanks for reporting. Jep the old repo has been dropped. This is already fixed for v6.22: https://github.com/MichaIng/DietPi/pull/2552

You could install the deb package manually: https://github.com/remoteit/installer/releases
No additional configs steps are done by our installer, so the result is the same.
Or apply the changes from the PR manually to your `/DietPi/dietpi/dietpi-software`.

kuerious, 2019-02-27:
I followed the instructions on their landing page at:

https://i.imgur.com/rBfKfnM.png

> apt-get update
> apt-get install connectd -y
> connectd_installer

MichaIng, 2019-02-27:
@kuerious 
And did this work? I also saw that install instructions on their website, but neither the Debian repo (https://packages.debian.org/search?searchon=names&keywords=connectd) nor the Raspbian repo has the `connectd` package:
```
2019-02-27 21:56:11 root@micha:/tmp# apt search connectd
Sorting... Done
Full Text Search... Done
```

kuerious, 2019-02-27:
Dunno. All I care is to have remote SSH access to my boxes while I struggle with MFA VPN in OPNsense.

MichaIng, 2019-02-28:
@kuerious 
Okay, yeah this is fine. Could you please check the currently shipped version? I am just interested 😉:
`apt show connectd`

However we leave our way of fix inside, since Raspbian Stretch is really the only repo that has this package. Kinda exception and we don't know how well this is maintained. Since installing the package manually from official source is really easy, we simply stick with this method for consistency.

I mark this issue as closed, since fixed with v6.22.

closed 2019-02-28 by MichaIng

kuerious, 2019-03-12:
This is interesting. I have 1 physical DietPi & another virtual (on ESXi, so using "x86_64" version) DietPi, for my PiHole servers. The live one worked fine, but the virtual one is unhappy:

```
root@dns2:~# apt show connectd
N: Unable to locate package connectd
N: Unable to locate package connectd
E: No packages found
```

Curious...

kuerious, 2019-03-12:
And it looks like I have a link to others, like myself, who(m) wish to install on a VM.

[ConnectD on VM using x86_64:](https://github.com/remoteit/installer/releases/download/v2.2.10-beta/connectd_2.2.10_amd64.deb)

MichaIng, 2019-03-12:
@kuerious 
Jep since the package is only available on Raspbian Stretch repo, not on Debian nor Raspbian Buster. Raspbian goes strange ways there, shipping packages via stable repo before adding and testing them in testing branch... As well Kodi v18 is only available on Raspbian Stretch while Buster still ships v17.6 😕.